Koenkk/zigbee2mqtt #18313: Insecure Zigbee Network Encryption Key Generation

0xbb, 2023-07-13:
### What happened?

[`generateNetworkKey`](https://github.com/Koenkk/zigbee2mqtt/blob/dev/lib/zigbee.ts#L166) uses the insecure [`Math.random()`](https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/Math/random) function to generate the network encryption key: 

```ts
    private generateNetworkKey(): number[] {
        const key = Array.from({length: 16}, () => Math.floor(Math.random() * 255));
        settings.set(['advanced', 'network_key'], key);
        return key;
    }
```
Additionally, `Math.floor(Math.random() * 255)` will never return the value 255.

I recommend a drop-in fix via [`randomInt`](https://nodejs.org/api/crypto.html#cryptorandomintmin-max-callback) from `node:crypto`:

```ts
const key = Array.from({length: 16}, () => randomInt(256));
```

I think users should regenerate existing network keys to be secure.

---

Maybe it would also be good to use the same method of generating random numbers for the `generatePanID` function just below.

---

Thank you very much for developing Zigbee2MQTT. It's a great project!

### Zigbee2MQTT version

1.32.1 commit: 4ccfdfc

### Adapter firmware version

-

### Adapter

-

Koenkk, 2023-07-13:
Could you submit a PR to fix this?

0xbb, 2023-07-13:
Sure!

closed 2023-08-20 by github-actions[bot]