Koenkk/zigbee2mqtt #18315: Issue with: SmartThings GP-WOU019BBDWG (ZB-ONOFFPlug-D0005) - stops reporting.

tomtom13, 2023-07-13:
### What happened?

Hi,
I'm experiencing peculiar problem with this power plug. After a initial setup it reports power and energy usage properly, however after few days to few weeks it's stops reporting updates about those values seem to be stuck. The lqi, state & power_on_behavior statuses are reported properly.

Unplugging the power plug from mains, rebooting it, resetting it doesn't seem to solve the problem - however changing values in reporting values tab, does seem to force the reset and plug starts updating the power values properly ... this however again lasts days to weeks.

I've got 6 of those plugs and all are having the same problem, hence I don't think it's and issue with hardware. Powering down the plug makes me thinks that it's not plug software problem (alto I might be wrong). That leads me to think that the problem might be on the zigbee2mqtt. If there is anyway I can help with debugging the issue - please let me know.

(admin - is this should be an issue, rather than discussion - do let me know)

### What did you expect to happen?

To keep reporting power forever without an issue ? 

### How to reproduce it (minimal and precise)

Have that plug running for several days to several weeks, however I can help with testing. 

### Zigbee2MQTT version

1.32.1 commit: unknown

### Adapter firmware version

20210708

### Adapter

zStack3x0

closed 2023-08-20 by github-actions[bot]

tomtom13, 2023-08-20:
Soooo no action ? not even a direction where to dig on how to find a culprit ?

irekzielinski, 2023-08-21:
Have the same issue, reboot, power down, etc does not solve the problem.
But thank you for suggesting of reconfiguration - this works for me as well.
As a workaround I will trigger re-config from Home Assistant when value has not been updated for more than one hour.

tomtom13, 2023-09-01:
Yeah I did reconfigure by hand of ALL the entities, trying to beat it to submission.
But hey, since z2m is not helping in any way, I decided to get my self a samsung smart things hub to see whenever there is a firmware update - NOPE, seems like samsung doesn't care. Also samsung app reported nothing even thou there was a real consumption going through the plug, so I would say it's very typical for samsung to release on great promises and then just not support anything. 
Now I've I purchased 8 plugs from Frient. They seem to be a "white label" type of brand (where they buy unbranded generics from china and slap they brand on it) but it seems to work so far. 5 samsung plug were removed from my HA, and it everything goes well for more than 3 months, then samsung plugs will hit the ebay.

aweaton123, 2025-02-16:
@irekzielinski 
Following update to Zigbee2Mqtt v2.0.0 the previous mqtt topic did not work. I have updated my Home Assistant automation mqtt payload accordingly. Sharing for anyone else using this.

```
- alias: SmartThings Power Reset
  id: "202403171157"
  trigger:
    - platform: template
      value_template: >-
        {{ as_timestamp(now()) - as_timestamp(states.sensor['washing_machine_energy_power'].last_updated)|int(0) > (3600*2) }}
      id: "washing_machine"
    - platform: template
      value_template: >-
        {{ as_timestamp(now()) - as_timestamp(states.sensor['tumble_dryer_energy_power'].last_updated)|int(0) > (3600*2) }}
      id: "tumble_dryer"
  condition: []
  action:
    - choose:
        - conditions:
            - condition: trigger
              id: "washing_machine"
          sequence:
            - service: mqtt.publish
              data:
                topic: zigbee2mqtt/bridge/request/device/configure_reporting
                qos: "2"
                payload: >-
                  {"id":"Washing Machine
                  SmartThings","endpoint":1,"cluster":"haElectricalMeasurement","attribute":"activePower","minimum_report_interval":30,"maximum_report_interval":3600,"reportable_change":1}
            - service: mqtt.publish
              data:
                topic: zigbee2mqtt/bridge/request/device/configure_reporting
                qos: "2"
                payload: >-
                  {"id":"Washing Machine
                  SmartThings","endpoint":1,"cluster":"seMetering","attribute":"currentSummDelivered","minimum_report_interval":30,"maximum_report_interval":3600,"reportable_change":0.01}
            - service: persistent_notification.create
              data:
                message: 'Washing machine power monitoring reset'
                title: 'SmartThings Reset'
        - conditions:
            - condition: trigger
              id: "tumble_dryer"
          sequence:
            - service: mqtt.publish
              data:
                topic: zigbee2mqtt/bridge/request/device/configure_reporting
                qos: "2"
                payload: >-
                  {"id":"Tumble Dryer
                  SmartThings","endpoint":1,"cluster":"haElectricalMeasurement","attribute":"activePower","minimum_report_interval":30,"maximum_report_interval":3600,"reportable_change":1}
            - service: mqtt.publish
              data:
                topic: zigbee2mqtt/bridge/request/device/configure_reporting
                qos: "2"

                payload: >-
                  {"id":"Tumble Dryer
                  SmartThings","endpoint":1,"cluster":"seMetering","attribute":"currentSummDelivered","minimum_report_interval":30,"maximum_report_interval":3600,"reportable_change":0.01}
            - service: persistent_notification.create
              data:
                message: 'Tumble dryer power monitoring reset'
                title: 'SmartThings Reset'
```

tomtom13, 2025-02-16:
Update from my side - I got rid of all samsung plugs as it's not worth my time. I got rid of everything except for Nest plugs - where I want to log the power used and philips plug where I need a reliable connection without power reporting. So to sum up: Samsung, Frient, Tuya are gone due to constant hassle.