Koenkk/zigbee2mqtt #18318: [New device support]:  Tuya Zigbee Human Presence Sensor  5.8GHz millimeter wave technology

labaland, 2023-07-14:
### Link

None but bought here: https://www.aliexpress.com/item/1005005764168560.html

### Database entry

{"id":11,"type":"Router","ieeeAddr":"0xa4c13837a1c64227","nwkAddr":5179,"manufId":4417,"manufName":"_TZE204_sooucan5","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"3\u0000\u0000\u0000i4\u0000\u0000\u0000i5\u0000\u0000\u0000i6\u0000\u0000\u0000i7\u0000\u0000\u0000i8\u0000\u0000\u0000i:\u0000\u0000\u0000i:\u0000\u0000\u0000i","65506":56,"65508":1,"modelId":"TS0601","stackVersion":0,"dateCode":"","appVersion":74}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1689293331655,"defaultSendRequestWhen":"immediate"}

### Comments

Im not good at this stuff sadly, i hope someone can add it !
thx <3

i bought som aliexpress.

### External converter

```shell
Conbee 2
```

labaland, 2023-07-17:
@Koenkk 

i think theese are pretty much the same:

https://www.zigbee2mqtt.io/devices/TS0601_smart_human_presence_sensor_1.html
https://www.zigbee2mqtt.io/devices/TS0601_human_presence_sensor.html

them and mine has in modelId "TS0601"  :)

labaland, 2023-07-19:
@Koenkk 
Device works after adding this line un tuya.js

{ modelID: 'TS0601', manufacturerName: '_TZE204_sooucan5' },

thx!

labaland, 2023-07-19:
However, this is spammed in my log, but i maybe should do that :P 
[image]

Koenkk, 2023-07-19:
Where does the line need to be added? (line number)

Regarding the spamming, it seems the device does this (common with these kind of TuYa sensors)

Koenkk, 2023-07-19:
Will be supported after https://github.com/Koenkk/zigbee-herdsman-converters/pull/5985 is merged

closed 2023-07-19 by Koenkk

labaland, 2023-07-19:
> Where does the line need to be added? (line number)
> 
> Regarding the spamming, it seems the device does this (common with these kind of TuYa sensors)

line 4126 or 4127 :) 
Thank you!

kennethpangky, 2023-08-01:
Hi just to verify if this was merged already into 1.32.1-1? Tried adding the device Tuya TS0601 TZE204_sooucan5 but it's still marked as unsupported. Tried adding external convertor but it only shows the lux value, incorrectly if I might add.