Koenkk/zigbee2mqtt #18319: [New device support]: Moes Zigbee Smoke

copetogo, 2023-07-14:
### Link

https://conceptkart.com/products/moes-zigbee-smart-smoke-fire-alarm?currency=INR&variant=40220845604938&utm_medium=cpc&utm_source=google&utm_campaign=Google%20Shopping&gclid=CjwKCAjwwb6lBhBJEiwAbuVUShb2xiegrgN9F9-oWdiv1vyJgyrMb0kv30JlDO7HrKctmRvMMqDE8xoCanoQAvD_BwE

### Database entry

ieee: 0xa4c13833078da550

### Comments

so i was using tuya converter file TS0601 model:_TZE200_e2bed . it used to work but now that converter file has stopped working . it would be great if support for this sensor can be built

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const fzLocal = {
    tuya_smoke: {
        cluster: 'manuSpecificTuya',
        type: ['commandSetDataResponse', 'commandDataReport', 'commandDataResponse'],
        convert: (model, msg, publish, options, meta) => {
            const result = {};
            for (const dpValue of msg.data.dpValues) {
                const dp = dpValue.dp;
                const value = tuya.getDataValue(dpValue);
                switch (dp) {
                case tuya.dataPoints.state:
                    result.smoke = value === 0;
                    break;
                case 4:
                    result.tamper = value;
                    break;
                case 14:
                    result.battery_low = value === 0;
                    break;
                case 16:
                    result.muffling = value;
                    break;
                default:
                    meta.logger.warn(`zigbee-herdsman-converters:tuya_smoke: Unrecognized DP #${dp} with data ${JSON.stringify(dpValue)}`);
                }
            }
            return result;
        },
    }
}

const definition = {
    zigbeeModel: ['e2bedvo9'],
    fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_e2bedvo9'}],
    model: 'TS0601_smoke2',
    vendor: 'TuYa',
    description: 'Smoke sensor',
    fromZigbee: [fzLocal.tuya_smoke],
    toZigbee: [
        {
            key: ['muffling'],
            convertSet: async (entity, key, value, meta) => {
                await tuya.sendDataPointBool(entity, 16, value);
            }
        }
    ],
    exposes: [e.smoke(), e.tamper(), e.battery_low(), exposes.binary('muffling', ea.STATE_SET, true, false)],
};

module.exports = definition;
```

copetogo, 2023-07-14:
Hi @Koenkk  so just found out this is actually the rebranded  tuya model TuYa ZSS-QY-SSD-A-EN which is supported in zigbee2mqtt. now how do i get it working for this model. Can you share me the converter file for this model?

Koenkk, 2023-07-15:
`_TZE200_e2bedvo9` is supported out-of-the-box in the latest z2m release.

copetogo, 2023-07-15:
> ieee: 0xa4c13833078da550

so i was using a sample converter (attached above) to make the moes zigbee sensor work ,i am not sure this is the correct device ID (TZE200_e2bedvo9).The reason i say this is this device has no muffling factor as output .in the device support list it does not show up. do you want me to delete the converter file and try natively?

copetogo, 2023-07-15:
Ok it is working now closing the issue . thanks

closed 2023-07-15 by copetogo

sergeyshaklein, 2024-12-03:
Hello. 

A similar device came to me and it was not detected automatically. 

Zigbee Module: TS0601
Zigbee Manufacturer: _TZE**284**_e2bedvo9
The topic discusses TZE**200**_e2bedvo9

Please tell me how I can add it to Z2M