Koenkk/zigbee2mqtt #18321: [Feature request]: Custom cluster definitions

pzeiler, 2023-07-14:
### Is your feature request related to a problem? Please describe

Cheap DIY ZigBee devices like ESP32-C6, ESP32-H2 or nrf52840 are becoming widely available for makers. The SDKs are still work in progress, and in case of esp-zigbee for example currently lack support for 0x000c (Analog Input). Future support remains unclear, and custom clusters are only allowed with 0xff×× cluster IDs, further limiting the possibilites due to (I suppose) certification reasons..
Z2M offers all mechanism needed to deal with those limitations (custom converters etc.) besides the definition of custom clusters with attributes. If a maker wants to create a ZigBee device with attributes that do not fit into the sdk-supported official clusters, the only way I found to get it running is to add a cluster definition by patching /zcl/definition/cluster.js as a proof-of-concept. Pull request for individual configurations are not the way to deal with this I guess.

### Describe the solution you'd like

Integrate the possibility of a custom cluster configuration in the external_converters definition, maybe by exposing an utils.addCluster('name', {definition}) function and calling it inside the "configure" callback. 

### Describe alternatives you've considered

Just like with external converters, there could be a way to add a custom cluster definition to the stack by adding external js files to the configuration.yaml.

### Additional context

Freedom for the ZigBee clusters!

closed 2023-08-21 by github-actions[bot]

Greginkansas, 2023-11-29:
Cheap DIY ZigBee devices like ESP32-C6 are going to be a big thing  HA.  I think Z2M need to be a part of this.

pzeiler, 2023-11-29:
Well I think no one really looked into this, it was just auto-staled. It would certainly a great thing to have external_clusters similar to external_converters.

roastcom, 2024-01-11:
@pzeiler , did you manage to find any way to work with custom clusters?

pzeiler, 2024-01-11:
@roastcom 
I patched the clusters.js to accept an arbituary cluster ID which also worked with the ESP-IDF ZigBee clusters. This obviously is not update-safe, and specific to my usecase regarding the cluster attributes etc.

roastcom, 2024-01-11:
@pzeiler
Wow, that might be above my level of skill at this point.
Thank you for letting me know.

MetaImi, 2024-02-07:
@pzeiler Can you describe how to patch the clusters.js? :)

nobodyguy, 2024-02-07:
> @pzeiler Can you describe how to patch the clusters.js? :)

Add your custom cluster definition to https://github.com/Koenkk/zigbee-herdsman/blob/master/src/zcl/definition/cluster.ts

nobodyguy, 2024-02-08:
> @nobodyguy You would have to do it in the transpiled runtime cluster.js, not the typescript dev source.
Yes, that's a good point. I was just pointing to the right file where to start looking.

Generic custom clusters similar to the external converters would be the best solution. Meanwhile, you can either override the zigbee-herdsman dependency with your fork or extract a custom git patch and apply it somewhere in your build tasks.

MetaImi, 2024-02-09:
I just found a solution for it. I'm working on a weather station with ESP32-H2 and I want to report wind speed. So I wrote Arduino code with a custom cluster (0xFFFE) and a custom attribute to represent the wind speed. In Zigbee2MQTT  I made an external exporter with a custom converter. Here is the code: 

```const {} = require( 'zigbee-herdsman-converters/lib/modernExtend');
// Add the lines below
const fz = require( 'zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require ('zigbee-herdsman-converters/lib/exposes');
const reporting = require( 'zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const {} = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require( 'zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const fLocal = {
    windSpeed: {
      cluster: '65534', //This part is important
	  type: ['readResponse', 'attributeReport'],
	  convert: (model, msg, publish, options, meta) →> {
	  if (msg.data.hasOwnProperty('1')) {
			const windSpeed = parseFloat (msg-data['1']); 
			return {I'windSpeed']: windSpeed};
		}
	},
};
	
const definition = {
	zigbeeModel: ['MetaWeather'],
	model: 'WeatherStation', // Update this with the real model of the device (written on the device itself or product page) 
	vendor: 'MetaImi', // Update this with the real vendor of the device (written on the device itself or product page) 
	description: 'Weather station by MetaImi', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
	extend: [],
	fromZigbee: [fLocal.windSpeed, fz.temperature, fz.humidity], 
	toZigbee: [],
	exposes: [e.temperature(), e.humidity(), exposes.enum( 'Wind Direction', ea.STATE, ['north', 'east', 'west']).withDescription( 'Wind direction'), e.numeric ( 'windSpeed',ea.STATE).withLabel( 'Wind speed').withUnit ('km/h').withDescription ('Measured wind speed')],
};

module.exports = definition;```

nobodyguy, 2024-02-09:
> I just found a solution for it. I'm working on a weather station with ESP32-H2 and I want to report wind speed.

Don't forget there are generic attributes like analogInput for this purpose.

MetaImi, 2024-02-09:
> > I just found a solution for it. I'm working on a weather station with ESP32-H2 and I want to report wind speed.
> 
> Don't forget there are generic attributes like analogInput for this purpose.

Where is the fun there? :) I will extend it with wind direction + rain cluster (amount, detected)

pzeiler, 2024-02-09:
> I just found a solution for it. I'm working on a weather station with ESP32-H2 and I want to report wind speed. So I wrote Arduino code with a custom cluster (0xFFFE) and a custom attribute to [...]

So you were able to define the custom cluster in the external_converters file? That didn't work when I tried but maybe I got it wrong...

roastcom, 2024-02-10:
> > > I just found a solution for it. I'm working on a weather station with ESP32-H2 and I want to report wind speed.
> > 
> > 
> > Don't forget there are generic attributes like analogInput for this purpose.
> 
> Where is the fun there? :) I will extend it with wind direction + rain cluster (amount, detected)

You don't have it on github by any chance do you? I'd be very curios to see how exactly you implemented it on Zigbee side.

MetaImi, 2024-02-11:
> You don't have it on github by any chance do you? I'd be very curios to see how exactly you implemented it on Zigbee side.

Not yet. It is still work in progress. I plan to release it once it is done, but I need a few weeks to finish everything. Now working on hardware for the the other sensors (rain, wind etc)