Koenkk/zigbee2mqtt #18322: AWOX/EGLO: ROVITO 900091 Firmware updated - doesn't work

MaurizioITA, 2023-07-14:
### Link

https://www.eglo.com/it/plafoniera-rovito-z-900091.html

### Database entry

 {"id":31,"type":"Router","ieeeAddr":"0xa4c1380c05e6248f","nwkAddr":51292,"manufId":4417,"manufName":"AwoX","powerSource":"Mains (single phase)","modelId":"EBF_RGB_Zm_CLP","epList":[1,3,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,8,768,4096,64599,10],"outClusterList":[6],"clusters":{"genBasic":{"attributes":{"modelId":"EBF_RGB_Zm_CLP","manufacturerName":"AwoX","powerSource":1,"zclVersion":3,"appVersion":2,"stackVersion":1,"hwVersion":60,"swBuildId":"2.5.0_999"}}},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":4751,"epId":3,"devId":269,"inClusterList":[65360,65361,4],"outClusterList":[65360,65361],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":1,"hwVersion":60,"swBuildId":"2.5.0_999","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1689329329552,"defaultSendRequestWhen":"immediate"}

### Comments

Hi,
after a factory reset and a firmware update (by the manufacturer) of this light: EGLO ROVITO - 900091, no longer works
The MODEL has changed in **EBF_RGB_Zm_CLP** from **EBF_RGB_Zm**

I've the same problem with EGLO - PADRODIGANO  900486 that was recognized as a ROVITO - it was good because it works peroperly before the firmware update.

Now I've 6 EGLO'S lights: 3 firmware updated that doesn't work and 3 not updated that works properly - but the lights are the same!

[image]
[Database.txt](https://github.com/Koenkk/zigbee2mqtt/files/12051946/Database.txt)

Tks in advance

Koenkk, 2023-07-15:
Added `EBF_RGB_Zm_CLP`.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-07-15 by Koenkk

MaurizioITA, 2023-07-15:
SUPER!