Koenkk/zigbee2mqtt #18323: Vimar 14595.0

giovannismart, 2023-07-14:
### Link

https://www.vimar.com/en/int/catalog/product/index/code/14595.0

### Database entry

"modelZigbee":"DimmerSwitch_v1.0","type":"Router","ieeeAddr":"0xf4ce362762f02e35

### Comments

Hello!

I am currently trying to integrate in my HomeAssistant+Zigbee2MQTT setup a Vimar 14595.0 dimmer-switch which, unfortunately, is currently not supported by Zigbee2MQTT as indicated by the 'Unsupported' tag on the Zigbee2MQTT dashboard.

More specifically, the module is recognized and it can be added to the Zigbee2MQTT dashboard but I believe that since it lacks a suitable configuration in the .ts file, none of its functionalities are available.

Even following the (https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_2-adding-your-device) guide provided on the Zigbee2MQTT site, I was not able to configure the dimmer properly. I borrowed the .ts configuration of supported devices with dimmer capabilities like i) 'BTicino' model: 'L4411C/N4411C/NT4411C'; ii) 'Enbrighten' model: '43080',43113','43082'; iii) 'GE' model: '45852GE','45857GE') but it did not help as the dimmer still could not control the lights (not switching on/off nor dimming).

Please provide guidance. How do I build the correct configuration for my device?

Thank you!

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const legacy = require('zigbee-herdsman-converters/lib/legacy');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['DimmerSwitch_v1.0'],
    model: '14595.0',
    vendor: 'Vimar',
    description: 'IoT connected dimmer mechanism 220-240V',
    extend: extend.light_onoff_brightness({noConfigure: true, disableEffect: true}),
    configure: async (device, coordinatorEndpoint, logger) => {
        await extend.light_onoff_brightness().configure(device, coordinatorEndpoint, logger);
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
        await reporting.onOff(endpoint);
        await reporting.brightness(endpoint, {min: 1});
    },
};

module.exports = definition;
```

assioa, 2023-08-08:
hello,
bought the same dimmer and I'm having the same difficulties.
could you please help us?

thank you

Koenkk, 2023-08-09:
@giovannismart 

- can you provide the data/database.db entry of this device?
- your ext converter, looks good, what isn't working?

assioa, 2023-08-29:
Hello,
I don't want to barge in but it's been 3 weeks and I don't want to freeze the conversation.
Here the data/database.db entry of the device, it's the 19595 in place of the 14595 but it's just appearance:

{"id":2,"type":"Router","ieeeAddr":"0xf4ce3648cd979809","nwkAddr":21211,"manufId":4660,"manufName":"Vimar","powerSource":"Mains (single phase)","modelId":"DimmerSwitch_v1.0","epList":[10,11,242],"endpoints":{"10":{"profId":260,"epId":10,"devId":260,"inClusterList":[0,3],"outClusterList":[3,6,8],"clusters":{"genBasic":{"attributes":{"swBuildId":"","zclVersion":2}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffe99a3ad","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffe99a3ad","endpointID":1},{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffe99a3ad","endpointID":1}],"configuredReportings":[],"meta":{}},"11":{"profId":260,"epId":11,"devId":257,"inClusterList":[0,3,4,5,6,8],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"swBuildId":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":10,"hwVersion":11,"dateCode":"20200101","swBuildId":"","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1693083756248,"defaultSendRequestWhen":"immediate"}

thank you for your help, again

assioa, 2023-09-26:
hello @Koenkk,
after some trial-and-error, I reached the config below. 
Sadly it fixes the "read" issue but not the "write" since  I receive a 'read_only' issue (i.e. Publish 'set' 'write' to 'luce corridoio' failed: 'Error: Write 0xf4ce36879fd982d7/11 genOnOff({"onOff":1}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'READ_ONLY')')

what am I missing?
thank you,
Alessio  

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const legacy = require('zigbee-herdsman-converters/lib/legacy');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [
        {
            type: 'Router', manufacturerName: 'Vimar', modelID: 'DimmerSwitch_v1.0', endpoints: [
                {ID: 10, profileID: 260, deviceID: 260, inputClusters: [0, 3], outputClusters: [3, 6, 8]},
                {ID: 11, profileID: 260, deviceID: 257, inputClusters: [0, 3, 4, 5, 6, 8], outputClusters: []},
                {ID: 242, profileID: 41440, deviceID: 97, inputClusters: [], outputClusters: [33]}
            ],
        },
    ],
    zigbeeModel: ['DimmerSwitch_v1.0'],
    model: '19595.0',
    vendor: 'Vimar',
    description: 'IoT connected dimmer mechanism 220-240V',
    extend: extend.light_onoff_brightness({noConfigure: true, disableEffect: true}),
    configure: async (device, coordinatorEndpoint, logger) => {
        await extend.light_onoff_brightness().configure(device, coordinatorEndpoint, logger);
        const endpoint = device.getEndpoint(11);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
        await reporting.onOff(endpoint);
        await reporting.brightness(endpoint);
    },
};

module.exports = definition;

Koenkk, 2023-09-26:
You cannot write on/off, it has to be set via a command, publish e.g. to `zigbee2mqtt/0xf4ce362762f02e35/set` `{"state": "ON"}`

giovannismart, 2023-10-27:
Sinceramente non so come andare avanti!!
Cosa mi sfugge?

giovannismart, 2023-10-28:
Continuo ad avere questi errori: Error 2023-10-28 19:59:49Publish 'set' 'device_mode' to '0xf4ce362762f02e35' failed: 'Error: Write 0xf4ce362762f02e35/10 manuSpecificLegrandDevices({"0":{"value":257,"type":9}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4129,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 14050 - 10 - 54 - 64513 - 4 after 10000ms)' . Continuo a non comandare il dispositivo, su l pannello sono presenti le entità. questa è un'altra versione :
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const legacy = require('zigbee-herdsman-converters/lib/legacy');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['DimmerSwitch_v1.0'],
    model: '14595.0',
    vendor: 'Vimar',
    description: 'IoT connected dimmer mechanism 220-240V',
    extend: extend.light_onoff_brightness({noConfigure: true}),
    ota: ota.zigbeeOTA,
    fromZigbee: [fz.brightness, fz.identify, fz.on_off, fz.lighting_ballast_configuration, fz.legrand_cluster_fc01,
        fz.power_on_behavior, fz.legrand_led_in_dark],
    toZigbee: [tz.light_onoff_brightness, tz.legrand_settingEnableLedInDark, tz.legrand_settingEnableLedIfOn, tz.legrand_deviceMode,
        tz.legrand_identify, tz.ballast_config, tz.power_on_behavior],
    exposes: [
        e.light_brightness(),
        e.numeric('ballast_minimum_level', ea.ALL).withValueMin(1).withValueMax(254)
            .withDescription('Specifies the minimum brightness value'),
        e.numeric('ballast_maximum_level', ea.ALL).withValueMin(1).withValueMax(254)
            .withDescription('Specifies the maximum brightness value'),
        e.binary('device_mode', ea.ALL, 'dimmer_on', 'dimmer_off')
            .withDescription('Allow the device to change brightness'),
        e.binary('led_in_dark', ea.ALL, 'ON', 'OFF')
            .withDescription('Enables the LED when the light is turned off, allowing to see the switch in the dark'),
        e.binary('led_if_on', ea.ALL, 'ON', 'OFF')
            .withDescription('Enables the LED when the light is turned on'),
        e.power_on_behavior(),
    ],
    configure: async (device, coordinatorEndpoint, logger) => {
        await extend.light_onoff_brightness().configure(device, coordinatorEndpoint, logger);
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genIdentify', 'genOnOff', 'genLevelCtrl', 'genBinaryInput', 'lightingBallastCfg']);
        await reporting.onOff(endpoint);
        await reporting.brightness(endpoint);
    },
};

module.exports = definition;

Grazie

giovannismart, 2023-11-13:
HI!
I'm still looking for an adequate configuration to integrate the Vimar 14595.0 dimmer-switch device into my HomeAssistant+Zigbee2mqtt. It seems I've arrived at a partial resolution. The device is now recognized and added to the Zigbee2mqtt dashboard through this further configuration in the .ts file, the device from the dashboard does not control the dimmer, but if I act directly from the physical dimmer, recognition of the entity occurs on the dashboard and I can both turn on/ turn off and vary the brightness of the light. I report the new configuration below.
Please provide directions. How do I improve the correct configuration for my device that is now half working?

const definition = {
    fingerprint:[{"id":14,"type":"Router","ieeeAddr":"0xf4ce362762f02e35","nwkAddr":37294,"manufId":4660,"manufName":"Vimar","powerSource":"Mains (single phase)","modelId":"DimmerSwitch_v1.0","epList":[10,11,242],"endpoints":{"10":{"profId":260,"epId":10,"devId":260,"inClusterList":[0,3],"outClusterList":[3,6,8],"clusters":{"genBasic":{"attributes":{"swBuildId":""}}},"binds":[],"configuredReportings":[],"meta":{}},"11":{"profId":260,"epId":11,"devId":257,"inClusterList":[0,3,4,5,6,8],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"swBuildId":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":10,"hwVersion":11,"dateCode":"20200101","swBuildId":"","zclVersion":2,"interviewCompleted":true,"meta":{"configured":88764544},"lastSeen":1699431512504,"defaultSendRequestWhen":"immediate"}],
    zigbeeModel: ['DimmerSwitch_v1.0'],
    model: '14595.0',
    vendor: 'Vimar',
    description: 'IoT connected dimmer mechanism 220-240V',
    fromzigbee: [fz.brightness],
    tozigbee: [tz.light_onoff_brightness],
    extend: extend.light_onoff_brightness({noConfigure: true}),
    exposes: [
        e.light_brightness(),
    ],
    configure: async (device, coordinatorEndpoint, logger) => {
    await extend.light_onoff_brightness().configure(device, coordinatorEndpoint, logger);
    const endpoint = device.getEndpoint(11);
    await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl', 'genIdentify']);
    await reporting.onOff(endpoint);
    await reporting.brightness(endpoint);
        },
};

Thanks for your help.

LeoeLeoeL, 2023-11-15:
Is there any news about the integration of this device?

Koenkk, 2023-11-15:
Try adding the `endpoint` to the converter, e.g.

```js
extend: extend.light_onoff_brightness({noConfigure: true}),
endpoint: (device) => {return {default: 11}}, // <- add this line
exposes: [
e.light_brightness(),
],
```

LeoeLeoeL, 2023-11-15:
What I get with:

const definition = {
fingerprint:[{"id":14,"type":"Router","ieeeAddr":"0xf4ce362762f02e35","nwkAddr":37294,"manufId":4660,"manufName":"Vimar","powerSource":"Mains (single phase)","modelId":"DimmerSwitch_v1.0","epList":[10,11,242],"endpoints":{"10":{"profId":260,"epId":10,"devId":260,"inClusterList":[0,3],"outClusterList":[3,6,8],"clusters":{"genBasic":{"attributes":{"swBuildId":""}}},"binds":[],"configuredReportings":[],"meta":{}},"11":{"profId":260,"epId":11,"devId":257,"inClusterList":[0,3,4,5,6,8],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"swBuildId":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":10,"hwVersion":11,"dateCode":"20200101","swBuildId":"","zclVersion":2,"interviewCompleted":true,"meta":{"configured":88764544},"lastSeen":1699431512504,"defaultSendRequestWhen":"immediate"}],
zigbeeModel: ['DimmerSwitch_v1.0'],
model: '14595.0',
vendor: 'Vimar',
description: 'IoT connected dimmer mechanism 220-240V',
fromzigbee: [fz.brightness],
tozigbee: [tz.light_onoff_brightness],
extend: extend.light_onoff_brightness({noConfigure: true}),
endpoint: (device) => {return {default: 11}}, //
exposes: [
e.light_brightness(),
],
configure: async (device, coordinatorEndpoint, logger) => {
await extend.light_onoff_brightness().configure(device, coordinatorEndpoint, logger);
const endpoint = device.getEndpoint(11);
await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl', 'genIdentify']);
await reporting.onOff(endpoint);
await reporting.brightness(endpoint);
},
};

is
[image]

giovannismart, 2023-11-16:
HI,
Thank you for your prompt response and support you are giving me. I entered the line you indicated. Something has changed compared to before. The power-on command from the dashboard now occurs, but with a 10 second delay. However, executing the command from the dashboard to turn off the light does not happen at all. So to turn it off I have to act directly on the physical device.
I report its database below, I hope it can be useful for diagnostics and understanding how to integrate this new device.
Please provide directions. How do I improve the correct configuration for my device that is now half working?

"id":14,"type":"Router","ieeeAddr":"0xf4ce362762f02e35","nwkAddr":37294,"manufId":4660,"manufName":"Vimar","powerSource":"Mains (single phase)","modelId":"DimmerSwitch_v1.0","epList":[10,11,242],"endpoints":{"10":{"profId":260,"epId":10,"devId":260,"inClusterList":[0,3],"outClusterList":[3,6,8],"clusters":{"genBasic":{"attributes":{"swBuildId":""}}},"binds":[],"configuredReportings":[],"meta":{"onLevelSupported":false}},"11":{"profId":260,"epId":11,"devId":257,"inClusterList":[0,3,4,5,6,8],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"swBuildId":""}},"genOnOff":{"attributes":{"onOff":1}},"genLevelCtrl":{"attributes":{"currentLevel":19}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0026e01f82","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b0026e01f82","endpointID":1},{"cluster":3,"type":"endpoint","deviceIeeeAddress":"0x00124b0026e01f82","endpointID":1}],"configuredReportings":[{"cluster":8,"attrId":0,"minRepIntval":1,"maxRepIntval":3600,"repChange":1,"manufacturerCode":null},{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0,"manufacturerCode":null}],"meta":{"onLevelSupported":false}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":10,"hwVersion":11,"dateCode":"20200101","swBuildId":"","zclVersion":2,"interviewCompleted":true,"meta":{"configured":1445959527},"lastSeen":1700124538926,"defaultSendRequestWhen":"immediate"

Thanks for your help.

Koenkk, 2023-11-16:
Please try with https://gist.github.com/Koenkk/2bff7d220e25b4952487d3dd7daebc68, if on/off doesn't work, provide me the debug log when setting it.

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

giovannismart, 2023-11-17:
Good morning,
I rewrote the script relating to the device that I try to make zigbee2mqtt support as you indicated in the attached link. We returned with this configuration in which the device is not controlled from the dashboard, although it is possible to turn it on/off and change brightness. Conversely, operation is correct if all operations are performed by the physical device. In reference to the second link you inserted, I checked the configuration procedure of the zigbee2mqtt add-on installed on my home assistant superviser. From the looks of it I think I've configured it well. Below I report the separate debugs for turning on/off both from the device and from the dashboard and my configuration.yml of zigbee2mqtt. I hope I understood the instructions you gave me correctly and that I followed them correctly.

Debugging from device
Power on:
2023-11-17 11:07:46Received Zigbee message from 'Dimmer_Living Room', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 11 with groupID 0
2023-11-17 11:07:46Received Zigbee message from 'Dimmer_Living Room', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":255}' from endpoint 11 with groupID 0
Shutdown:
2023-11-17 11:05:24Received Zigbee message from 'Dimmer_Living Room', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 11 with groupID 0
2023-11-17 11:05:25Received Zigbee message from 'Dimmer_Living Room', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":255}' from endpoint 11 with groupID 0

Debugging from dashboard:
Power on:
2023-11-17 11:14:33Received MQTT message on 'zigbee2mqtt/Dimmer_Soggiorno/set' with data '{"state":"ON"}'
2023-11-17 11:14:33Publishing 'set' 'state' to 'Dimmer_Soggiorno'

Shutdown:
2023-11-17 11:14:35Received MQTT message on 'zigbee2mqtt/Dimmer_Soggiorno/set' with data '{"state":"OFF"}'
2023-11-17 11:14:35Publishing 'set' 'state' to 'Dimmer_Soggiorno'

This is the configuration,yaml:
devices:
   '0x00124b0025319c45':
     friendly_name: Movement_1
   '0x00124b0026b7fec5':
     friendly_name: Light_storage
     state_action: true
   '0xf4ce3637936305a3':
     friendly_name: Thermostat_1
     thermostat_unit: celsius
   '0xf4ce364c8b7bd209':
     friendly_name: Presa_Macchina_Caffe
   '0x54ef441000777797':
     friendly_name: Movement_2
     occupancy_timeout: 5
     illuminance_calibration: 2000
     illuminance_lux_calibration: 2000
     device_temperature_calibration: 2000
   '0xf4ce368e11bd02d8':
     friendly_name: Light_Disengagement_Device
     state_action: true
   '0xf4ce366d2c32e3fb':
     friendly_name: Tapparella_Cucina_Device
     invert_cover: false
     availability: true
   '0xf4ce36fb2561fefd':
     friendly_name: Thermostat_2
   '0xf4ce366ac9e3d234':
     friendly_name: Tapparella_Studio_Device
     invert_cover: false
     availability: true
   '0xf4ce36d735589441':
     friendly_name: Tapparella_Soggiorno_Device
     invert_cover: false
     avilability: true
   '0xf4ce368a7ff79b66':
     friendly_name: Shutter_Bathroom_Device
     invert_cover: false
     availability: true
   '0xf4ce3677cc8dd866':
     friendly_name: Tapparella_Camera_Matrimoniale_Device
     invert_cover: false
     avilability: true
   '0xf4ce362762f02e35':
     friendly_name: Dimmer_Living Room
     transition: 5
     state_action: false
   '0x00124b002910d850':
     friendly_name: Bedroom_Thermometer
advanced:
   log_level: debug
  external_converters:
     - 14595.js

Thanks for your help

Koenkk, 2023-11-18:
Updated https://gist.github.com/Koenkk/2bff7d220e25b4952487d3dd7daebc68 , can you try again?

LeoeLeoeL, 2023-11-19:
I don't know if it could be useful but, this is the DDF of this device from Deconz and works.

{
  "schema": "devcap1.schema.json",
  "manufacturername": "Vimar",
  "modelid": "DimmerSwitch_v1.0",
  "product": "DimmerSwitch_v1.0",
  "sleeper": false,
  "status": "Draft",
  "subdevices": [
    {
      "type": "$TYPE_DIMMABLE_LIGHT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x0b"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert",
          "description": "The currently active alert effect.",
          "default": "none"
        },
        {
          "name": "state/bri",
          "description": "The current brightness.",
          "refresh.interval": 5
        },
        {
          "name": "state/on",
          "description": "True when device is on; false when off.",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 11,
      "dst.ep": 1,
      "cl": "0x0006",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x10",
          "min": 1,
          "max": 300
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 11,
      "dst.ep": 1,
      "cl": "0x0008",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x20",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        }
      ]
    }
  ]
}

Koenkk, 2023-11-22:
Could you provide the herdsman debug log when turning OFF and brightness 25% is set instead?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

Koenkk, 2023-11-23:
- Added the device (will be supported in the 1 December release)
- `UNSUPPORTED_ATTRIBUTE` indicates the device does not support it, I've disabled it.

closed 2023-11-23 by Koenkk

LeoeLeoeL, 2023-11-23:
I don't think the led light can be controlled by zigbee.

LeoeLeoeL, 2023-12-01:
Just paired again with the latest release. It works.
Thank to both.

P.S. It has been a hell to pair this device.

delli87, 2025-04-12:
> Try adding the `endpoint` to the converter, e.g.
> 
> extend: extend.light_onoff_brightness({noConfigure: true}),
> endpoint: (device) => {return {default: 11}}, // <- add this line
> exposes: [
> e.light_brightness(),
> ],

[log (2).log](https://github.com/user-attachments/files/19720430/log.2.log)