Koenkk/zigbee2mqtt #18326: [New device support]: Tuya Dual Smart plug with power monitoring

fumantsu, 2023-07-14:
### Link

https://www.immax.eu/immax-neo-smart-internal-double-socket-16a-zigbee-3-0-p15142/?cid=1505

### Database entry

{"id":24,"type":"Router","ieeeAddr":"0xa4c1380d0ce83f21","nwkAddr":19480,"manufId":4417,"manufName":"_TZ3000_jak16dll","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,2,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[3,4,5,6,1794,2820,57344,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65534":0,"modelId":"TS011F","manufacturerName":"_TZ3000_jak16dll","powerSource":1,"zclVersion":3,"appVersion":192,"stackVersion":0,"hwVersion":1,"dateCode":""}},"haElectricalMeasurement":{"attributes":{"acCurrentDivisor":1000,"acCurrentMultiplier":1}},"seMetering":{"attributes":{"divisor":100,"multiplier":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00212effff0854c0","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00212effff0854c0","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00212effff0854c0","endpointID":1}],"configuredReportings":[{"cluster":2820,"attrId":1285,"minRepIntval":5,"maxRepIntval":3600,"repChange":5},{"cluster":2820,"attrId":1288,"minRepIntval":5,"maxRepIntval":3600,"repChange":50},{"cluster":2820,"attrId":1291,"minRepIntval":5,"maxRepIntval":3600,"repChange":10},{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":[1,1]}],"meta":{}},"2":{"profId":260,"epId":2,"devId":266,"inClusterList":[4,5,6,57345],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":192,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-1941818122},"lastSeen":1689365856180,"defaultSendRequestWhen":"immediate"}

### Comments

The particular is actually a dual plug (not single) so it is missing control one of the plugs. I have below the datapoints.
{"1":"Switch","2":"Switch 2","9":"Countdown","10":"Countdown 2","17":"Add Electricity","18":"Current","19":"Power","20":"Voltage","21":"Test Bit","26":"Fault","27":"Relay Status","209":"Cycle Timing","210":"Random Timing"}

Since the zigbee model was exist but it points to a single plug I wasn't sure how to approach it from the prospective of the guide

fumantsu, 2023-08-17:
> same as #17041 <https://github.com/Koenkk/zigbee2mqtt/issues/17041>

True, closed as stale because of not activity which is pity.

Send from my hotdog

thebang2, 2023-08-26:
Any further infos for that?

Koenkk, 2023-08-27:
Could you check if the issue is fixed with the following external converter: https://gist.github.com/Koenkk/e8587976de964e3c76c89c12b13b6266

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

thebang2, 2023-08-27:
First of all, thank you very much for your help! The Converter doesn't work in my area, unfortunately. I´m using Docker with koenkk/zigbee2mqtt:latest.

`
Using '/app/data' as data directory
/app/node_modules/zigbee-herdsman-converters/index.js:69
    definition.toZigbee.push(tz.scene_store, tz.scene_recall, tz.scene_add, tz.scene_remove, tz.scene_remove_all, tz.read, tz.write, tz.command, tz.factory_reset);
                        ^
TypeError: Cannot read properties of undefined (reading 'push')
    at Object.addDefinition [as addDeviceDefinition] (/app/node_modules/zigbee-herdsman-converters/index.js:69:25)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:15:17)
    at new Controller (/app/lib/controller.ts:84:58)
    at start (/app/index.js:106:18)
`

wociscz, 2023-08-27:
Neat, works like a charm - Can control both sockets independently. Power metering is common for both sockets (for device as a whole). Icon/Image is different, but nvm.
Thanks for help

[image]
[image]
[image]
 
I'd say - problem fixed.

Koenkk, 2023-08-28:
Great, picture will be fixed after the next release (please remove the external converter after updating to the next release)

closed 2023-08-28 by Koenkk

wociscz, 2023-08-29:
Just little note, power [W] and current [A] seems to not showing any value. Still at `0` all the time. But not important for me.

thebang2, 2023-10-01:
Up! :)

ReNeGaDe124, 2023-12-13:
> > 
> 
> You are right - power, voltage and current without function:
> 
> ![current](https://private-user-images.githubusercontent.com/39240956/264121736-b54b911f-5279-4260-a37d-ce2bc8e67620.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDI0NzQyMjcsIm5iZiI6MTcwMjQ3MzkyNywicGF0aCI6Ii8zOTI0MDk1Ni8yNjQxMjE3MzYtYjU0YjkxMWYtNTI3OS00MjYwLWEzN2QtY2UyYmM4ZTY3NjIwLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjEzVDEzMjUyN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTljYWE4NjgyODEwYzdlYWJmNjkyYWI5OWQ1MzBjM2I0ZjM3ZWIwZTVmZDEwNGE2YTdjZDQ5ZDUxNjY0OTEyNTQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.rwKYueNEgex30JaBjM_ICyhotjPzdis-SYdNNdBcXwI)

Hello, I just bought this model yesterday and I think I found a workaround to properly show the Power, Current and Voltage sensors.

- On line 40 of `ext_converter.js` change from     `meta: {multiEndpoint: true}`, to     `meta: {multiEndpoint: false}`,
- Restart zigbee2mqtt

[image]

This is working for me. Hope it can help you as well. Maybe the developer can include this change in next release or find a better solution.

thebang2, 2023-12-13:
Can confirm the change from @ReNeGaDe124 is working:

[image]

@Koenkk: Maybe you can add this change in the next update?

Koenkk, 2023-12-13:
`meta: {multiEndpoint: false}` will prevent state updates from the device from working, I've pushed a different fix for this.

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.