Koenkk/zigbee2mqtt #18327: power_precision and voltage_precision settings with Nous A1Z are not working

chris777c, 2023-07-15:
### What happened?

Hello,

I have a plug Nous A1Z with last firmware (version 192), the precision settings seem to be ignored, power and voltage values are always without decimals however energy or current work.

I don't know if it comes from device or z2m and I don't know how to help you for resolved issue ?

### What did you expect to happen?

I could like a precision for power and voltage can be set on my device

### How to reproduce it (minimal and precise)

Settings :
[image]

return :
`{
    "child_lock": "UNLOCK",
    "current": 0.4,
    "energy": 45.7,
    "indicator_mode": "off",
    "linkquality": 123,
    "power": 90,
    "power_outage_memory": "restore",
    "state": "ON",
    "update": {
        "installed_version": 192,
        "latest_version": 192,
        "state": "idle"
    },
    "voltage": 237
}`

### Zigbee2MQTT version

1.32.1

### Adapter firmware version

20220219

### Adapter

zStack3x0

### Debug log

`debug 2023-07-15 08:24:35Received MQTT message on 'zigbee2mqtt/bridge/request/device/options' with data '{"id":"0xa4c138e76f43279c","options":{"current_precision":1},"transaction":"7q2k1-14"}'
info 2023-07-15 08:24:36Changed config for device 0xa4c138e76f43279c
info 2023-07-15 08:24:36MQTT publish: topic 'zigbee2mqtt/bridge/response/device/options', payload '{"data":{"from":{"current_precision":2,"energy_precision":2,"legacy":false,"power_precision":1,"state_action":true,"voltage_precision":1},"id":"0xa4c138e76f43279c","restart_required":false,"to":{"current_precision":1,"energy_precision":2,"legacy":false,"power_precision":1,"state_action":true,"voltage_precision":1}},"status":"ok","transaction":"7q2k1-14"}'
debug 2023-07-15 08:24:41Received MQTT message on 'zigbee2mqtt/bridge/request/device/options' with data '{"id":"0xa4c138e76f43279c","options":{"energy_precision":1},"transaction":"7q2k1-15"}'
info 2023-07-15 08:24:41Changed config for device 0xa4c138e76f43279c
info 2023-07-15 08:24:41MQTT publish: topic 'zigbee2mqtt/bridge/response/device/options', payload '{"data":{"from":{"current_precision":1,"energy_precision":2,"legacy":false,"power_precision":1,"state_action":true,"voltage_precision":1},"id":"0xa4c138e76f43279c","restart_required":false,"to":{"current_precision":1,"energy_precision":1,"legacy":false,"power_precision":1,"state_action":true,"voltage_precision":1}},"status":"ok","transaction":"7q2k1-15"}'
debug 2023-07-15 08:24:41Received MQTT message on 'zigbee2mqtt/bridge/request/device/options' with data '{"id":"0xa4c138e76f43279c","options":{"energy_precision":0},"transaction":"7q2k1-16"}'
info 2023-07-15 08:24:41Changed config for device 0xa4c138e76f43279c
info 2023-07-15 08:24:41MQTT publish: topic 'zigbee2mqtt/bridge/response/device/options', payload '{"data":{"from":{"current_precision":1,"energy_precision":1,"legacy":false,"power_precision":1,"state_action":true,"voltage_precision":1},"id":"0xa4c138e76f43279c","restart_required":false,"to":{"current_precision":1,"energy_precision":0,"legacy":false,"power_precision":1,"state_action":true,"voltage_precision":1}},"status":"ok","transaction":"7q2k1-16"}'
debug 2023-07-15 08:24:51Received Zigbee message from 'Nous-plug-1', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,4574]}' from endpoint 1 with groupID 0
info 2023-07-15 08:24:51MQTT publish: topic 'zigbee2mqtt/Nous-plug-1', payload '{"child_lock":"UNLOCK","current":0.4,"energy":46,"indicator_mode":"off","linkquality":127,"power":96,"power_outage_memory":"restore","state":"ON","update":{"installed_version":192,"latest_version":192,"state":"idle"},"voltage":239}'
debug 2023-07-15 08:24:51Received Zigbee message from 'Nous-plug-1', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,4574]}' from endpoint 1 with groupID 0
debug 2023-07-15 08:24:53Received Zigbee message from 'Xiaomi-motionsensor', type 'attributeReport', cluster 'msIlluminanceMeasurement', data '{"measuredValue":3}' from endpoint 1 with groupID 0
info 2023-07-15 08:24:53MQTT publish: topic 'zigbee2mqtt/Xiaomi-motionsensor', payload '{"battery":100,"device_temperature":29,"illuminance":3,"illuminance_lux":3,"linkquality":94,"occupancy":true,"power_outage_count":48,"voltage":3035}'
debug 2023-07-15 08:24:53Received Zigbee message from 'Xiaomi-motionsensor', type 'attributeReport', cluster 'msOccupancySensing', data '{"occupancy":1}' from endpoint 1 with groupID 0
info 2023-07-15 08:24:53MQTT publish: topic 'zigbee2mqtt/Xiaomi-motionsensor', payload '{"battery":100,"device_temperature":29,"illuminance":3,"illuminance_lux":3,"linkquality":98,"occupancy":true,"power_outage_count":48,"voltage":3035}'
debug 2023-07-15 08:24:53Received Zigbee message from 'Evology-PSE', type 'readResponse', cluster 'genBasic', data '{"zclVersion":1}' from endpoint 1 with groupID 0
debug 2023-07-15 08:24:53Successfully pinged 'Evology-PSE' (attempt 1/2)
debug 2023-07-15 08:24:54Received MQTT message on 'zigbee2mqtt/bridge/request/device/options' with data '{"id":"0xa4c138e76f43279c","options":{"energy_precision":1},"transaction":"7q2k1-17"}'
info 2023-07-15 08:24:54Changed config for device 0xa4c138e76f43279c
info 2023-07-15 08:24:55MQTT publish: topic 'zigbee2mqtt/bridge/response/device/options', payload '{"data":{"from":{"current_precision":1,"energy_precision":0,"legacy":false,"power_precision":1,"state_action":true,"voltage_precision":1},"id":"0xa4c138e76f43279c","restart_required":false,"to":{"current_precision":1,"energy_precision":1,"legacy":false,"power_precision":1,"state_action":true,"voltage_precision":1}},"status":"ok","transaction":"7q2k1-17"}'
debug 2023-07-15 08:25:02Received Zigbee message from 'Nous-plug-1', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":87}' from endpoint 1 with groupID 0
info 2023-07-15 08:25:02MQTT publish: topic 'zigbee2mqtt/Nous-plug-1', payload '{"child_lock":"UNLOCK","current":0.4,"energy":46,"indicator_mode":"off","linkquality":120,"power":87,"power_outage_memory":"restore","state":"ON","update":{"installed_version":192,"latest_version":192,"state":"idle"},"voltage":239}'
debug 2023-07-15 08:25:03Received MQTT message on 'zigbee2mqtt/bridge/request/device/options' with data '{"id":"0xa4c138e76f43279c","options":{"power_precision":2},"transaction":"7q2k1-18"}'
info 2023-07-15 08:25:03Changed config for device 0xa4c138e76f43279c
info 2023-07-15 08:25:04MQTT publish: topic 'zigbee2mqtt/bridge/response/device/options', payload '{"data":{"from":{"current_precision":1,"energy_precision":1,"legacy":false,"power_precision":1,"state_action":true,"voltage_precision":1},"id":"0xa4c138e76f43279c","restart_required":false,"to":{"current_precision":1,"energy_precision":1,"legacy":false,"power_precision":2,"state_action":true,"voltage_precision":1}},"status":"ok","transaction":"7q2k1-18"}'
debug 2023-07-15 08:25:24Saving state to file /opt/zigbee2mqtt/data/state.json
debug 2023-07-15 08:25:33Received Zigbee message from 'Nous-plug-1', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":98}' from endpoint 1 with groupID 0
info 2023-07-15 08:25:34MQTT publish: topic 'zigbee2mqtt/Nous-plug-1', payload '{"child_lock":"UNLOCK","current":0.4,"energy":46,"indicator_mode":"off","linkquality":116,"power":98,"power_outage_memory":"restore","state":"ON","update":{"installed_version":192,"latest_version":192,"state":"idle"},"voltage":239}'
debug 2023-07-15 08:25:47Received MQTT message on 'zigbee2mqtt/bridge/request/device/options' with data '{"id":"0xa4c138e76f43279c","options":{"power_precision":1},"transaction":"7q2k1-19"}'
info 2023-07-15 08:25:47Changed config for device 0xa4c138e76f43279c
info 2023-07-15 08:25:48MQTT publish: topic 'zigbee2mqtt/bridge/response/device/options', payload '{"data":{"from":{"current_precision":1,"energy_precision":1,"legacy":false,"power_precision":2,"state_action":true,"voltage_precision":1},"id":"0xa4c138e76f43279c","restart_required":false,"to":{"current_precision":1,"energy_precision":1,"legacy":false,"power_precision":1,"state_action":true,"voltage_precision":1}},"status":"ok","transaction":"7q2k1-19"}'
debug 2023-07-15 08:25:52Received Zigbee message from 'Nous-plug-2', type 'attributeReport', cluster 'genBasic', data '{"65506":54,"65508":0,"appVersion":192}' from endpoint 1 with groupID 0
debug 2023-07-15 08:25:52Received Zigbee message from 'Nous-plug-2', type 'attributeReport', cluster 'genBasic', data '{"65506":54,"65508":0,"appVersion":192}' from endpoint 1 with groupID 0
[... cut ...]  'attributeReport', cluster 'genBasic', data '{"65506":54,"65508":0,"appVersion":192}' from endpoint 1 with groupID 0
debug 2023-07-15 08:25:52Received Zigbee message from 'Nous-plug-2', type 'attributeReport', cluster 'genBasic', data '{"65506":54,"65508":0,"appVersion":192}' from endpoint 1 with groupID 0
debug 2023-07-15 08:25:53Received Zigbee message from 'Nous-plug-2', type 'attributeReport', cluster 'genBasic', data '{"65506":54,"65508":0,"appVersion":192}' from endpoint 1 with groupID 0`

Nitrouze, 2023-08-10:
I have the exact same problem running a Conbee II adapter. 
The plug is identified by z2m version 1.32.2  as an TS011F /  _TZ3000_2putqrmw model.

chris777c, 2023-09-09:
Hello,

A little up

wherzig, 2023-10-03:
Same problem here. With Firmware 77 = 1.0.13. 

Is there any firmware available to change the precision?

kladderadeng, 2023-10-11:
I have the same issue and I updated from 77 to 192 yesterday.

tlamoureux24, 2023-12-22:
Exact same issue.
Z2M HA Addon 1.34.0-1
A1Z firmware 192

wherzig, 2023-12-23:
After reading some forums, I think that the Nouz A1Z does not support better precesion. zigbee2mqtt does just offer to change it but this does not mean that the device support it.

chris777c, 2023-12-23:
Thank you for you reply, I close ticket

closed 2023-12-23 by chris777c

prahal, 2024-05-01:
It seems Nous A1Z supports better precision as in cluster TuyaZBElectricalMeasurement 0x0b4, the rms_current attribute 0x0508 is in milliamps (read via ZHA in Home Assistant). But the ac_current_divisor is 1000 and the ac_current_multiplier is 1 and cannot be changed at least I failed to in ZHA.
At least in HomeAssistant I get via ZHA a rms_current of 20mA for a Bluetooth speaker, but with the default _decimals of 1 for electricalMeasurement and with a divisor of 1000 rounding only give an output of "0".

I took a look in z2m zigbee-herdman-converters it seems  rmsCurrent is not rounded i z2m https://github.com/Koenkk/zigbee-herdsman-converters/blob/3839ea6df4f0e66d9fc6f01faffd1298d7e508e5/src/converters/fromZigbee.ts#L788. So I don't know why you have the issue in z2m. But you should at least be able to get rmsCurrent 0,02 from what A1Z produces.