Koenkk/zigbee2mqtt #18331: Alecto Smart-Smoke10 doesn't pair

thedoturtle, 2023-07-16:
### What happened?

I recently bought some Alecto Smart-Smoke10 Zigbee smoke detectors because of their compatibility with Zigbee2MQTT (https://www.zigbee2mqtt.io/devices/SMART-SMOKE10.html). When I put the device in Pairing Mode (3x short press) it shows a red light. Zigbee2MQTT is in pairing mode but won't find the smoke detectors.
Pairing other devices works perfectly, so it has something to do with this specific model?

### What did you expect to happen?

The device should show up and pair.

### Zigbee2MQTT version

1.32.1

### Adapter firmware version

20210708

### Adapter

Sonoff Zigbee dongle plus

thedoturtle, 2023-07-19:
Well.. the included batteries were almost dead. Fresh batteries fixed the Problem :)

closed 2023-07-19 by thedoturtle

rogervdh, 2023-08-01:
I have the exact same problem, but two fresh AA's don't change anything. :-( I'm using the latest version of Zigbee2MQTT and the Skyconnect coordinator with the latest firmware. Strange, all other Zigbee devices (including Frient and Heiman smokedetector) show up within 2 seconds during pairing.

thedoturtle, 2023-08-01:
> I have the exact same problem, but two fresh AA's don't change anything. :-( I'm using the latest version of Zigbee2MQTT and the Skyconnect coordinator with the latest firmware. Strange, all other Zigbee devices (including a Frient and Heiman smokedetector) show up within 2 seconds during pairing.

Well.. pretty weird. Better check your batteries again. I use a sonoff Zigbee 3.0 USB Dongle and now it works perfectly:
Insert fresh batteries --> press 3x --> red light shows up --> pairing mode "on" in Zigbee2MQTT --> automatically pairing.
Good luck!

rogervdh, 2023-08-01:
I don't think that the batteries are the problem. I have tried the included GP batteries, some new Kruidvat batteries and also a pair of new Aerocell (Lidl) batteries (all with an expiration date in 2025 and never used before) and after pushing 3 times the red light shows up....but no pairing in Zigbee2MQTT (in pairing mode of course). I have tried paring mode of all my Zigbee devices as well my coordinator (Skyconnect) only, but unfortunately... It looks like that I have to return the item to Bol.com. :-(

RDC-gitty, 2023-08-05:
I had the same issue. But what worked for me was that you insert the batteries and immediately press the button 3 times (or 4). I had 2 detectors and both times it gave a loud beep after the first press, not sure if it matters if it does or doesn't. So if it doesn't work. 

Steps for pairing:

1. Make sure that the coordinator is in joinable state.
2. Remove 1 battery (if they were already inserted) and press the button.
3. Insert the battery and...
4. Quickly press 3 times.

Hope this helps someone later on.

rogervdh, 2023-08-05:
That was the solution! :-) It took several attempts, but then the smoke detector showed up in Zigbee2MQTT. Thanks again!