Koenkk/zigbee2mqtt #18335: MOES switches Zigbee Direct Binding / Clusters

dinde, 2023-07-16:
### What happened?

Hello,

Something is not working as expected as reported on closed issue #9465 

**Context**
Zigbee2mqtt as Home Assistant Module version 1.32.1
Home Assistant 2023.7.2
Coordinator: ZBDongle-P, type: zStack3x0, firmware: 

**Issue 1**
I installed several MOES Switchs reported on z2m as:
TS0011 -> _TZ3000_hhiodade -> Moes ZS-EUB_1gang which is: MOES ZS-EU1-WH-MS => correct
TS0012 -> _TZ3000_18ejxno0 -> TuYa TS0012 which is: MOES ZS-EU2-WH-MS => incorrect
TS0013 -> _TZ3000_qewo8dlz -> TuYa TS0013 which is: MOES ZS-EU3-WH-MS => incorrect

**Issue 2**
On TS0011 & TS0012, there is a "Mode of the backlight: Off/Normal/Inverted" which is correct
On TS0013, there is "Intensity of the backlight: Low/Medium/High" which is wrong.
It should be as TS001[1-2] "Mode of the backlight: Off/Normal/Inverted."

**Issue 3**
Zigbee Direct Binding which is supported by MOES app as advertised.
Z2M does not expose correctly Clusters and there is a bug as @phretor described on #9465 
If you Unbind an Endpoint (let's say 2) and want to bind it again as it was with the coordinator: Clusters are empty.
You need to reconfigure the device to get it back.
If you want to direct bind an Endpoint (let's say 2 to another identical MOES switch) to achieve a wireless 2 ways with Zigbee Direct Binding feature: Clusters are empty (OnOff missing) so you can't bind it.

I also tried to directly publish a payload without success on MQTT.
Tested on dev branch same behaviour

Thanks a lot for your wonderful work @Koenkk !

### What did you expect to happen?

1. Switches to be correctly recognized and named.
2. TS0013 to work as others with Mode of the backlight option
3. Zigbee direct binding payload to work as it's featured by the seller

### How to reproduce it (minimal and precise)

Repeat the steps described on following comment:
https://github.com/Koenkk/zigbee2mqtt/issues/9465#issuecomment-1449354541

### Zigbee2MQTT version

1.32.1

### Adapter firmware version

20230507

### Adapter

Sonoff ZBDongle-P

### Debug log

MQTT publish: topic 'zigbee2mqtt/bridge/response/device/bind', payload '{"data":{"clusters":[],"failed":[],"from":"Inter-3G-One/3","to":"Inter-3G-Two/3"},"error":"Nothing to bind","status":"error"}'
MQTT publish: topic 'zigbee2mqtt/bridge/response/device/bind', payload '{"data":{"clusters":[],"failed":[],"from":"Inter-3G-One/right","to":"Inter-3G-Two/right"},"error":"Nothing to bind","status":"error"}'

dinde, 2023-07-20:
Ok, as seen with manufacturer, you need to have a Neutral connected to enjoy this feature !
I can't test myself since I have no neutral.
HTH

dinde, 2023-07-20:
Closing this as there is no fix needed.

closed 2023-07-20 by dinde

phretor, 2023-07-20:
@dinde I have a neutral connected so I can do tests. I don't understand what relationship there is between ZigBee binding and neutral. Can you explain?

phretor, 2023-07-20:
@dinde I think I the MOES/Zemismart wall touch switches don't support standard ZigBee binding. Maybe they support "some" ZigBee binding if one uses the vendor's app/hub.

Do you know of anyone who succeeded binding those switches?

dinde, 2023-07-20:
Not really, except they are advertised doing something similar.
But it can be as you said the gateway/app from Moes doing a virtual binding instead of direct binding.
When you try to bind it, are the clusters with OnOff etc ?
Mine are empty but I have no neutral.

dinde, 2023-07-21:
If zigbee direct binding is indeed supported you should get clusters genOnOff OnOff as other people reported with Ikea devices supporting it.
Let me know @phretor ;)

dominicusmento, 2023-08-10:
I have Moes TS0601 (1, 2 and 3 gang versions) with neutral connected and there are no clusters available to bind for those switches so I don't think it's due to neutral. 

On the other end, I have some Ikea remotes which I can bind successfully to blitzwolf tuya TS0121 power plugs.
What's also interesting from my perspective is that those switches are not routers even if they are connected with neutral and have constant power supply. (Device type: EndDevice)

We still don't know if that's by design or it is z2m bug so maybe this issue should be reopened ?

dinde, 2023-09-14:
@dominicusmento 
Question is how to know if it's by design ?
I believe people reversed the endpoints with the tuya gateway and seen no direct binding exposed.
BUT this feature is working thru MOES app not sure about tuya regular one.

Sadly I gave away my tuya gateway so I'm not sure how I can double check that.
Maybe we could try to test with a converter to add OnOff ?