Koenkk/zigbee2mqtt #18336: [New device support]: TS0601 _TZE204_wvovwe9h 2 gang switch

MaciejMilinkiewicz, 2023-07-17:
### Link

https://www.aliexpress.com/item/1005004836095177.html

### Database entry

{"id":68,"type":"Router","ieeeAddr":"0xdc8e95fffef7d250","nwkAddr":19084,"manufId":4098,"manufName":"_TZE204_wvovwe9h","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"j5G,ij5G,ij5G,i","65506":31,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE204_wvovwe9h","stackVersion":0,"dateCode":"","appVersion":70}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1689552988792,"defaultSendRequestWhen":"immediate"}

### Comments

YIDI Tuya with no neutral 2 gang switch.

closed 2023-08-24 by github-actions[bot]

BandBxx, 2023-08-31:
+1.  Same device didnt recognized

santo76, 2023-09-09:
+1. Same device didnt recognized

Koenkk, 2023-09-10:
Could you check if the issue is fixed with the following external converter: https://gist.github.com/Koenkk/47a68248f02eae3cc74f54256297a252

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed (it will also expose dimming functionality but ignore this for now)

BandBxx, 2023-09-11:
Hi.   Have this error after reboot:

[22:28:19] INFO: Preparing to start...
[22:28:20] INFO: Socat not enabled
[22:28:21] INFO: Starting Zigbee2MQTT...
/app/data/extension/externally-loaded.js:10
    fingerprint: tuya.fingerprint('TS0601', ['_TZE200_fjjbhx9d', '_TZE200_e3oitdyu', '_TZE200_gwkapsoq', '_TZE204_zenj4lxv', '_TZE204_wvovwe9h']),
                 ^
ReferenceError: tuya is not defined
    at /app/data/extension/externally-loaded.js:10:18
    at Script.runInContext (node:vm:141:12)
    at Script.runInNewContext (node:vm:146:17)
    at Object.runInNewContext (node:vm:306:38)
    at loadModuleFromText (/app/lib/util/utils.ts:152:8)
    at loadModuleFromFile (/app/lib/util/utils.ts:159:12)
    at Object.getExternalConvertersDefinitions (/app/lib/util/utils.ts:169:25)
    at getExternalConvertersDefinitions.next (<anonymous>)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:12:20)
    at new Controller (/app/lib/controller.ts:84:58)

BandBxx, 2023-09-11:
[image]
[image]

Debug 2023-09-11 22:52:34Received Zigbee message from '0xa4c1389327bbad7b', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,0],"type":"Buffer"},"datatype":2,"dp":7}],"seq":8960}' from endpoint 1 with groupID 0
Error 2023-09-11 22:52:34Exception while calling fromZigbee converter: Value '0' is not allowed, expected one of true,false}
Debug 2023-09-11 22:52:34Error: Value '0' is not allowed, expected one of true,false at Object.from (/app/node_modules/zigbee-herdsman-converters/src/lib/tuya.ts:364:35) at Object.convert (/app/node_modules/zigbee-herdsman-converters/src/lib/tuya.ts:1006:57) at Receive.onDeviceMessage (/app/lib/extension/receive.ts:150:51) at EventEmitter.emit (node:events:525:35) at EventBus.emitDeviceMessage (/app/lib/eventBus.ts:102:22) at Controller.<anonymous> (/app/lib/zigbee.ts:108:27) at Controller.emit (node:events:513:28) at Controller.selfAndDeviceEmit (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:527:14) at Controller.onZclOrRawData (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:738:18) at EZSPAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:144:70)
Info 2023-09-11 22:52:34MQTT publish: topic 'zigbee2mqtt/0xa4c1389327bbad7b', payload '{"brightness_l1":0,"brightness_l2":0,"countdown_l1":null,"countdown_l2":null,"last_seen":"2023-09-11T22:52:33+03:00","linkquality":108,"min_brightness_l1":138,"min_brightness_l2":0,"state_l1":"OFF","state_l2":"OFF"}'
Info 2023-09-11 22:52:34MQTT publish: topic 'zigbee2mqtt/0xa4c1389327bbad7b/l1', payload '{"brightness":0,"countdown":null,"min_brightness":138,"state":"OFF"}'
Info 2023-09-11 22:52:34MQTT publish: topic 'zigbee2mqtt/0xa4c1389327bbad7b/l2', payload '{"brightness":0,"countdown":null,"min_brightness":0,"state":"OFF"}'
Debug 2023-09-11 22:52:34Received Zigbee message from '0xa4c1389327bbad7b', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,0],"type":"Buffer"},"datatype":2,"dp":8}],"seq":9216}' from endpoint 1 with groupID 0
Info 2023-09-11 22:52:34MQTT publish: topic 'zigbee2mqtt/0xa4c1389327bbad7b', payload '{"brightness_l1":0,"brightness_l2":0,"countdown_l1":null,"countdown_l2":null,"last_seen":"2023-09-11T22:52:33+03:00","linkquality":108,"min_brightness_l1":138,"min_brightness_l2":0,"state_l1":"OFF","state_l2":"OFF"}'
Info 2023-09-11 22:52:34MQTT publish: topic 'zigbee2mqtt/0xa4c1389327bbad7b/l1', payload '{"brightness":0,"countdown":null,"min_brightness":138,"state":"OFF"}'
Info 2023-09-11 22:52:34MQTT publish: topic 'zigbee2mqtt/0xa4c1389327bbad7b/l2', payload '{"brightness":0,"countdown":null,"min_brightness":0,"state":"OFF"}'

BandBxx, 2023-09-11:
So far, this converter is working successfully for me, but there are a minimum of additional properties and settings. Only direct switching on and off of both channels, but successfully:

const legacy = require('zigbee-herdsman-converters/lib/legacy');
const fz = {...require('zigbee-herdsman-converters/converters/fromZigbee'), legacy: require('zigbee-herdsman-converters/lib/legacy').fromZigbee};
const tz = {...require('zigbee-herdsman-converters/converters/toZigbee'), legacy: require('zigbee-herdsman-converters/lib/legacy').toZigbee};
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;

const definition = {fingerprint: [
{modelID: 'TS0601', manufacturerName: '_TZE204_wvovwe9h'},

],
model: 'TS0601_switch_2_gang',
vendor: 'TuYa',
description: '2 gang switch',
exposes: [e.switch().withEndpoint('l1').setAccess('state', ea.STATE_SET),
          e.switch().withEndpoint('l2').setAccess('state', ea.STATE_SET)],
fromZigbee: [fz.ignore_basic_report, legacy.fromZigbee.tuya_switch],
toZigbee: [legacy.toZigbee.tuya_switch_state],
meta: {multiEndpoint: true},
configure: async (device, coordinatorEndpoint, logger) => {
    await tuya.configureMagicPacket(device, coordinatorEndpoint, logger);
    await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff']);
    if (device.getEndpoint(2)) await reporting.bind(device.getEndpoint(2), coordinatorEndpoint, ['genOnOff']);
},

endpoint: (device) => {
    return {'l1': 1, 'l2': 1};
},

};

module.exports = definition;

MaciejMilinkiewicz, 2023-09-11:
It works.

Koenkk, 2023-09-13:
@BandBxx thanks, added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)