Koenkk/zigbee2mqtt #18337: [New device support]: TS0601 _TZE204_ptaqh9tk YIDI Tuya with no neutral 1 gang switch

MaciejMilinkiewicz, 2023-07-17:
### Link

https://www.aliexpress.com/item/1005004836095177.html

### Database entry

{"id":69,"type":"Router","ieeeAddr":"0xdc8e95fffeea9ce4","nwkAddr":1769,"manufId":4098,"manufName":"_TZE204_ptaqh9tk","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"�DG,i�DG,i","65506":31,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE204_ptaqh9tk","powerSource":1,"zclVersion":3,"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1689552974142,"defaultSendRequestWhen":"immediate"}

### Comments

YIDI Tuya with no neutral 1 gang switch.

closed 2023-08-24 by github-actions[bot]

santo76, 2023-09-09:
+1. Same device didnt recognized

Koenkk, 2023-09-10:
Could you check if the issue is fixed with the following external converter: https://gist.github.com/Koenkk/08bfbe2833a03b9a7e6dbbb54e761315

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

MaciejMilinkiewicz, 2023-09-11:
Works for me:

```
const legacy = require('zigbee-herdsman-converters/lib/legacy');
const fz = {...require('zigbee-herdsman-converters/converters/fromZigbee'), legacy: require('zigbee-herdsman-converters/lib/legacy').fromZigbee};
const tz = {...require('zigbee-herdsman-converters/converters/toZigbee'), legacy: require('zigbee-herdsman-converters/lib/legacy').toZigbee};
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;

const definition = {fingerprint: [
{modelID: 'TS0601', manufacturerName: '_TZE204_ptaqh9tk'},

],
model: 'TS0601_switch_1_gang',
vendor: 'TuYa',
description: '1 gang switch',
exposes: [e.switch().withEndpoint('l1').setAccess('state', ea.STATE_SET)],
fromZigbee: [fz.ignore_basic_report, legacy.fromZigbee.tuya_switch],
toZigbee: [legacy.toZigbee.tuya_switch_state],
meta: {multiEndpoint: true},
configure: async (device, coordinatorEndpoint, logger) => {
await tuya.configureMagicPacket(device, coordinatorEndpoint, logger);
await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff']);
},

endpoint: (device) => {
return {'l1': 1};
},

};

module.exports = definition;
```

based on:
[https://github.com/Koenkk/zigbee2mqtt/issues/18336#issuecomment-1714493253](https://github.com/Koenkk/zigbee2mqtt/issues/18336#issuecomment-1714493253)

santo76, 2023-09-22:
no working for me, legacy not found

ReferenceError: legacy is not defined
    at /app/data/extension/externally-loaded.js:22:42
    at Script.runInContext (node:vm:135:12)
    at Script.runInNewContext (node:vm:140:17)
    at Object.runInNewContext (node:vm:292:38)
    at loadModuleFromText (/app/lib/util/utils.ts:152:8)
    at loadModuleFromFile (/app/lib/util/utils.ts:159:12)
    at Object.getExternalConvertersDefinitions (/app/lib/util/utils.ts:169:25)
    at getExternalConvertersDefinitions.next (<anonymous>)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:12:20)
    at new Controller (/app/lib/controller.ts:84:58)
Using '/app/data' as data directory
/app/data/extension/externally-loaded.js:22
    fromZigbee: [fz.ignore_basic_report, legacy.fromZigbee.tuya_switch],

santo76, 2023-09-22:
i add this line to script and now is working like a charm

`const legacy = require('zigbee-herdsman-converters/lib/legacy');`

thanks you very much

Koenkk, 2023-09-23:
Added the device for out-of-the-box support, you can remove the ext converter after updating to the next release.