Koenkk/zigbee2mqtt #18338: Hue Signify LTU001 E14 buld support

oscfor, 2023-07-17:
### Link

https://www.philips-hue.com/sv-se/p/hue-white-ambiance-klotlampa---e14-smart-ljuskalla/8719514491106

### Database entry

{"id":26,"type":"Router","ieeeAddr":"0x001788010d9f3e80","nwkAddr":189,"manufId":4107,"manufName":"Signify Netherlands B.V.","powerSource":"Mains (single phase)","modelId":"LTU001","epList":[11,242],"endpoints":{"11":{"profId":260,"epId":11,"devId":268,"inClusterList":[0,3,4,5,6,8,4096,64515,768,64516],"outClusterList":[25],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":1,"hwVersion":1,"dateCode":"20230116","swBuildId":"1.101.10","zclVersion":8,"interviewCompleted":true,"meta":{},"lastSeen":1689588294617,"defaultSendRequestWhen":"immediate"}

### Comments

I tried to write a new .js-file but got stuck at exposes: [e.battery(), e.temperature(), e.humidity()]. What should I use for a normal lamp device? Checked the Philips library at https://github.com/Koenkk/zigbee-herdsman-converters/tree/master/src/devices but couldn't get it to work.

### Supported color modes

color temp

Koenkk, 2023-07-17:
Please add the color temperature range as described on https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_3-adding-converter-s-for-your-device , then I can add support for this device.

oscfor, 2023-07-17:
Ah, sorry about that!

{"colorTempPhysicalMin":153}
{"colorTempPhysicalMax":454}

closed 2023-07-18 by Koenkk