Koenkk/zigbee2mqtt #18341: [New device support]: Philips 929003053401

maxk1337, 2023-07-17:
### Link

https://www.philips-hue.com/de-at/p/hue-white---color-ambiance-flourish-tischleuchte/8719514343481

### Database entry

{"id":6,"type":"Router","ieeeAddr":"0x001788010cec4582","nwkAddr":19155,"manufId":4107,"manufName":"Philips","powerSource":"Mains (single phase)","modelId":"929003053401","epList":[11,242],"endpoints":{"11":{"profId":260,"epId":11,"devId":269,"inClusterList":[0,3,4,5,6,8,4096,64514,768,64513],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"929003053401","manufacturerName":"Philips","powerSource":1,"zclVersion":2,"appVersion":2,"stackVersion":1,"hwVersion":1,"dateCode":"20201117","swBuildId":"1.76.11"}},"lightingColorCtrl":{"attributes":{"colorTempPhysicalMin":153,"colorTempPhysicalMax":500}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":1,"hwVersion":1,"dateCode":"20201117","swBuildId":"1.76.11","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1689607785896,"defaultSendRequestWhen":"immediate"}

### Comments

Hi, 
I am quite new to this and I dont want to mess around. 

My new Philips Flourish is not yet supported. It should be similar to Philips 929003052601. So I created a external converter as you can see below. But if I follow the guide and change the configuration.yaml the Plugin wont start...

It would be nice, if you can support me, thank you.

Best
max

### External converter

```shell
const exposes = require('../lib/exposes');
const fz = {...require('../converters/fromZigbee'), legacy: require('../lib/legacy').fromZigbee};
const tz = require('../converters/toZigbee');
const ota = require('../lib/ota');
const reporting = require('../lib/reporting');
const philips = require('../lib/philips');
const e = exposes.presets;
const ea = exposes.access;

const extend = {switch: require('../lib/extend').switch};

const definition = {
    zigbeeModel: ['929003053401'],
    model: '929003053401',
    vendor: 'Philips',
    description: 'Hue Flourish white and color ambiance table light with Bluetooth',
    extend: philips.extend.light_onoff_brightness_colortemp_color({colorTempRange: [153, 500]}),
};

module.exports = definition;
```

### Supported color modes

color temperature, color (rgb)

### Color temperature range

{"colorTempPhysicalMin":153,"colorTempPhysicalMax":500}

Koenkk, 2023-07-17:
Added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-07-17 by Koenkk

maxk1337, 2023-07-17:
Thank you very much!