Koenkk/zigbee2mqtt #18346: [New device support]: TuYa

BlackStar-95, 2023-07-17:
### Link

https://developer.tuya.com/en/docs/iot-device-dev/Zigbee_SDK_sample_plug?id=Kavqomaym367l

### Database entry

{"id":25,"type":"Router","ieeeAddr":"0xa4c138ee2ddc0390","nwkAddr":14616,"manufId":4107,"manufName":"_TZE600_iypcp4py","powerSource":"Mains (single phase)","modelId":"TS0105","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,3,61184,10],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"appVersion":0,"hwVersion":0,"manufacturerName":"_TZE600_iypcp4py","zclVersion":3,"modelId":"TS0105","powerSource":1}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1689629607349,"defaultSendRequestWhen":"immediate"}

### Comments

I have tried to add this External Converter but the Zigbee2MQTT is not starting
Cannot read properties of undefined (reading 'options')

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0105',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE600_iypcp4py',
        },
    ],
    model: 'TS0105_new',
    vendor: 'TuYa',
    description: '3 gang switch',
    extend: extend.switch(),
    exposes: [e.switch().withEndpoint('l1').setAccess('state', ea.STATE_SET),
        e.switch().withEndpoint('l2').setAccess('state', ea.STATE_SET),
        e.switch().withEndpoint('l3').setAccess('state', ea.STATE_SET)],
    fromZigbee: [fz.ignore_basic_report, fz.tuya_switch],
    toZigbee: [tz.tuya_switch_state],
    meta: {multiEndpoint: true},
    endpoint: (device) => {
        // Endpoint selection is made in tuya_switch_state
        return {'l1': 1, 'l2': 2, 'l3': 3};
    },
};

module.exports = definition;
```

BlackStar-95, 2023-07-18:
Here is My External converter, the device is working
`
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0105',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE600_iypcp4py',
        },
    ],
    model: 'TS0105_new',
    vendor: 'TuYa',
    description: 'Fill in a description of the device here',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    configure: tuya.configureMagicPacket,
    exposes: [
        e.switch().withEndpoint('l1').setAccess('state', ea.STATE_SET),
        e.switch().withEndpoint('l2').setAccess('state', ea.STATE_SET),
        e.switch().withEndpoint('l3').setAccess('state', ea.STATE_SET),
    ],
    meta: {
        multiEndpoint: true,
        tuyaDatapoints: [
            [1, 'state_l1', tuya.valueConverter.onOff],
            [2, 'state_l2', tuya.valueConverter.onOff],
            [3, 'state_l3', tuya.valueConverter.onOff],
        ],
    },
    endpoint: (device) => {
        return {'l1': 1, 'l2': 1, 'l3': 1};
    },
};

module.exports = definition;
`

BlackStar-95, 2023-07-18:
Here is My External converter, the device is working
```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
// Since a lot of TuYa devices use the same modelID, but use different datapoints
// it's necessary to provide a fingerprint instead of a zigbeeModel
fingerprint: [
{
// The model ID from: Device with modelID 'TS0601' is not supported
// You may need to add \u0000 at the end of the name in some cases
modelID: 'TS0105',
// The manufacturer name from: Device with modelID 'TS0601' is not supported.
manufacturerName: '_TZE600_iypcp4py',
},
],
model: 'TS0105_new',
vendor: 'TuYa',
description: 'Fill in a description of the device here',
fromZigbee: [tuya.fz.datapoints],
toZigbee: [tuya.tz.datapoints],
configure: tuya.configureMagicPacket,
exposes: [
e.switch().withEndpoint('l1').setAccess('state', ea.STATE_SET),
e.switch().withEndpoint('l2').setAccess('state', ea.STATE_SET),
e.switch().withEndpoint('l3').setAccess('state', ea.STATE_SET),
],
meta: {
multiEndpoint: true,
tuyaDatapoints: [
[1, 'state_l1', tuya.valueConverter.onOff],
[2, 'state_l2', tuya.valueConverter.onOff],
[3, 'state_l3', tuya.valueConverter.onOff],
],
},
endpoint: (device) => {
return {'l1': 1, 'l2': 1, 'l3': 1};
},
};

module.exports = definition;
```

closed 2023-08-25 by github-actions[bot]

FrankGiesecke, 2023-09-18:
Is anyone working on this issue?
How can I get the action state from the switches?

FrankGiesecke, 2023-09-18:
@BlackStar-95 it would be great if you reopen this issue.

BlackStar-95, 2023-11-26:
> @BlackStar-95 it would be great if you reopen this issue.

Did you tired My External converter?