Koenkk/zigbee2mqtt #18347: E2201 RODRET Status Error

derDeno, 2023-07-17:
### What happened?

The status page returns null for actions and battery.

### What did you expect to happen?

Expected to see available actions and battery status

### Zigbee2MQTT version

1.32.1-dev commit: 313865b

### Adapter firmware version

20210708

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus, TI CC2652P + CP2102(N)

Koenkk, 2023-07-19:
That the action is always `null` is expected, see https://www.zigbee2mqtt.io/guide/faq/#why-is-the-action-property-always-empty

Regarding the battery:

Could you check if the issue is fixed with the following external converter: https://gist.github.com/Koenkk/688987a60de3f7bc13fc20b2413b76a1

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed (make sure that configure succeeds, you can manually trigger it via the z2m frontend -> click on device -> refresh button, if it fails make sure to wakeup the device right before triggering it)

kcdc80, 2023-07-21:
I tried your suggestion. Seems that z2m accepts the remote, but no events seems to be fired in HA when I try to push the buttons.

Koenkk, 2023-07-23:
And the battery part?

coolspot18, 2023-08-02:
I updated to the latest version of Z2M [1.32.2] and the E2201 is now recognized but not responding to any button presses.

I also ran the little JSON snippet from the Wiki to unbind the device - still no luck.

Any ideas?

bogd, 2023-08-07:
This is very strange... I just downloaded the most recent version of koenkk/zigbee2mqtt:latest (f7a8b612ea8c). And while the device does show up under the correct name (RODRET Dimmer), it also shows up as "unsupported".

I am not using any custom converters. Based on #18381 , the device should have been supported in the latest version. And it is confusing to me to see people here reporting that it works for them.

Is there anything else I need to do in order to get the device to work?

Shellfishgene, 2023-08-08:
My Rodret paired fine, all values look normal, except the "action" does not change when I press the buttons. This is with version 1.32.2.

bogd, 2023-08-10:
> My Rodret paired fine, all values look normal, except the "action" does not change when I press the buttons. This is with version 1.32.2. Home Assistant also does not get any button presses.

Maybe a silly question, but could it be [this](https://www.zigbee2mqtt.io/guide/faq/#why-is-the-action-property-always-empty)?

Do any actions show up in the Zigbee2MQTT logs when you press buttons? (go to Z2M/Logs, and filter by your device name in the "filter by text" box - then try pressing some buttons)

With my unit, I see the actions in the logs (on/off, brightness_move_up/down). 

Also, I have to ask - how do I figure out my Z2M version?

Shellfishgene, 2023-08-10:
> Maybe a silly question, but could it be [this](https://www.zigbee2mqtt.io/guide/faq/#why-is-the-action-property-always-empty)?
>
> Do any actions show up in the Zigbee2MQTT logs when you press buttons? [...]

No, nothing in the logs. With all my other IKEA remotes I see the action in the z2m interface change, even if it goes back immidiately.

basvdl, 2023-09-12:
I had the same issue. In most of cases the binds were not set correctly after pairing the device. After restarting a couple of times, repairing multiple times, and some manual bind / unbind operations, it finally worked out for me.

[image]

[image]

SpeedBlack, 2023-10-17:
Hello all,
I have the same problem (I tried the external converter without success).
Do you have any idea ? This problem is not solve ?

Thanks & regards,

Shellfishgene, 2023-10-17:
I deleted and rejoined the Rodret with the current version of z2m, and now it works as expected.

bogd, 2023-10-26:
Strangely enough, I am now seeing the same issue @Shellfishgene mentioned. 

I now have 3 Rodret remotes, all successfully paired to Z2M. For one of them, I see no action in the logs when pressing buttons. I see the device_announce message when resetting it, but that is it. Even after reconfiguring, or deleting and re-pairing the remote, I see the same behavior

Even more strangely:
* the other two remotes work just fine, only one of them has the issue
* If I pair the remote to another system (HA with SkyConnect), the actions are seen just fine. So the problem seems to be limited to Z2M, and intermittent

LoQue90, 2023-10-31:
Same for me, 3 Rodret, 2 of the work perfectly fine. The third pairs, but no actions or battery satus.

onimoschta, 2023-11-02:
I'm going to tell you how i did it:

pairing the device to zigbee2mqtt my Coordinator as "target" just had "genPollCtrl" ticked. (No Action reported while pressing a button, no linkquality shown)
I wanted to bind it directly to a ikea tradfri psu.

so while staying in the bind menu i pressed the pairing button 4 times to get it to pair.
while it started blinking i immediately added the "LevelCtrl" checkbox to the Coordinator and pressed on the blue heart box to "bind" it. i also pressed the pair box for my 2nd pairing TARGET (tradfri psu) with the checkbox "OnOff" ticked.
i prepared both target settings before i pressed the blue heart "bind" box. 
you need to press them both fast when you get the rodret remote into pairing mode.

It worked this way, cause the pairing window is like 10 seconds after the rodret led starts lighting up.

tristaoeast, 2023-11-05:
I´m having the same issue, only managed to get the actions to work once, and it seemed pretty random, never managed to get it working again

tristaoeast, 2023-11-07:
> I'm going to tell you how i did it:
>
> pairing the device to zigbee2mqtt my Coordinator as "target" just had "genPollCtrl" ticked. (No Action reported while pressing a button, no linkquality shown) I [...]

I´m not sure I understand the procedure you described there: how can you be in the bind menu if you are pairing it to a different router (the tardfri psu)?

tristaoeast, 2023-11-07:
> Strangely enough, I am now seeing the same issue @Shellfishgene mentioned.
>
> I now have 3 Rodret remotes, all successfully paired to Z2M. For one of them, I see no action in the logs when pressing [...]

I´m feeling tempted to get a SkyConnect

jschlyter, 2023-12-30:
I just tried adding two Rodret (E2201) devices with Z2M v1.34.0 coordinator zStack3x0 revision 20220219. They do pair and interview succeeds, but they do not respond once paired (no events emitted on button presses and bind operations fail).

jschlyter, 2024-01-04:
> I just tried adding two Rodret (E2201) devices with Z2M v1.34.0 coordinator zStack3x0 revision 20220219. They do pair and interview succeeds, but they do not respond once paired (no events emitted on button presses and bind operations fail).

… and with the latest release (1.35.0), it now pairs as well as works (but required a re-interview after first pair)

wiiNinja, 2024-02-21:
The device just seems flaky. The first two times after pairing, I could not get the action to work, but after the third time, the action seems to work reliably now (well at least for the last few hours). Will see if that is now a permanent condition.

Tortur3, 2024-03-14:
Bought a rodret few weeks back and everything went perfectly from the beginning.

Then I bought 2 more and they both pair but like many others, no button activites are being recognized.

I noticed there is some kind of product version (I assume) on the packaging and also on the inside of the rodret, just below the battery.

The one that is working fine got W2344 - the ones that don't work got printed W2342. At my local ikea there are only 2339 and 2342 available, got the 2344 from another store.

Just a wild guess, maybe someone else can evaluate this.

bogd, 2024-03-21:
@Tortur3 - where is this documented?

splizz1st, 2024-04-25:
@Tortur3 it would be very helpful to get answers for the questions from @bogd !

@bogd or did you get the rodret working?

bogd, 2024-04-25:
@splizz1st - I was never able to get the RODRET to work reliably with Z2M, so I just gave up (I still have a large collection of the "old" Ikea on/off switches, which work just fine, so I kept using those). Even when the RODRET was connected and I could see the actions, it would just stop working after a while.

I assume the problems will eventually be fixed, so I just wait until that happens :)

Please keep in mind that I haven't tested the device in a couple of months by now, so the problems might have already been fixed.

splizz1st, 2024-05-07:
@bogd oh thanks for the feedback, I've only just seen it. I've had the same experience and have had enough of the old Ikea switches ;-). But of course RODRET would be nice.

paolapersico1, 2024-05-18:
My experience: one of my RODRET worked well while I had problems with another one, which didn't report the button presses.
I have tried several times repairing and cluster binding (I noticed that the working device had the "LevelCtrl", "OnOff", "genPollCtrl", and "PowerCfg" boxes checked, while the one which didn't work only the "genPollCtrl"), and so on. Nothing worked. 

Since the RODRET I bought came from a kit with an IKEA bulb, I decided to pair it directly to the bulb, and when I paired it by following the instructions - by pressing the internal button for ten seconds at a close distance from the bulb - to my surprise it also started working in Z2M: I can see the actions and the battery level.

Not sure why the same 10-seconds button press didn't work when RODRET was close to the coordinator instead of the bulb, but I thought this could help somebody.

fuomag9, 2024-08-31:
Turning "LevelCtrl", "OnOff", "genPollCtrl", and "PowerCfg" seems to have made it pair as well, "LevelCtrl" does error out tho

Xiobe, 2024-11-22:
I had an issue pairing them too. When going through the [IKEA manual](https://www.ikea.com/us/en/manuals/rodret-wireless-dimmer-power-switch-smart-white__AA-2389077-1-100.pdf) I read that the reset sequence was to quickly press the pairing button 4 times within 5 seconds was. When I tried that, I was able to pair them without any issue.

yarafie, 2024-12-03:
Just to share with all

My setup
Z2M 1.42.0 latest as of today
SLZB-06 Zigbee Coordinator 
CPU Model: CC2652P-ESP32-D0WD-V3
Firmware version: 20240710 v2.5.8
IKEA RODRET Dimmer E2202-W2402 (Came in package with a Tradfri Bulb)

I was able to pair succesfully in z2m using the 4 quick  presses and binded with z2m with below binds that showed up
LevelCtrl, OnOff, genPollCtrl, PowerCfg
Everything seemed to be exposed and worked (identify, battery, linkquality)
However no actions were logged when pressing the buttons.

I repeated the pairing above a few times. Sometimes removing the RODRET Completly from Z2M other times just re-pairing, sometimes removing the battery and waiting 10 minutes and retrying, always with the exact same results.

After reading some of the comments above, and with the RODRET already paired with Z2m I then did the 10 second push, I happened to be close to an IKEA Tradfri bulb already installed and paired with Z2M and the bulb   flashed. All Actions were now working and showing in Z2M and the bulb was being controlled by the RODRET. 

I'm using Home Assistant and spun off an Automation to control another light. Automation did work to control that light but it was also controlling the IKEA Tradfri Bulb.

So although this may be considered a success however its still controlling the Ikea Tradfri bulb along with whatever is put in the HA Automation.

wapolinar, 2024-12-30:
Just to add my experience here:
Z2M 1.42.0
SLZB-06 Coordinator
IKEA RODRET E2201 that came in a package with the E2204 TRETAKT power plug

My goal was to pair it only to the coordinator, not to the power plug since I do not want to use them together at all.

At first I had the issue that the E2201 did pair (4x quick presses on connect button near the battery), but left the network immediately. I fixed that by using a new battery instead of a rechargeable one.

Then, the device was paired, but no action was logged. Just by accident, I was able to start an (very slow, > 200 minutes long) OTA update which surprisingly succeeded directly after I successfully (re-)paired the device.

I was not able to bind the device to the coordinator by holding the button of the paired E2201 for 10 seconds and clicking the bind button for LevelCtrl, OnOff and genPollCtrl. But after my experience with the OTA update, I tried the following:
I removed the device from z2m and then paired it again. As soon as it was paired, I switch to the bind tab, selected the following checkboxes and clicked "bind" (bind was to the coordinator, of course):

[image]

Then, **the bind was successful and finally the actions showed up** in the mqtt log (and can be used to control something).

TB31388, 2025-02-02:
Any news on this. I have had the same problem and not been able to get any of the suggestions to work. I bought my E2201 as a stand alone product.

1vivy, 2025-03-31:
> I was not able to bind the device to the coordinator by holding the button of the paired E2201 for 10 seconds and clicking the bind button for LevelCtrl, OnOff and genPollCtrl. But after my [...]

I was able to press reset (4x presses on reset button), reinterview, and spam the bind. Now my actions register yay. Little easier than re-pairing and rushing to the bind page while z2m sets it up.

closed 2025-06-07 by github-actions[bot]