Koenkk/zigbee2mqtt #18349: [New device support]: QADZ2 Tuya Zigbee Dimmer 2 Channel

queroautomacao, 2023-07-18:
### Link

https://www.loja.queroautomacao.com.br/MLB-2874056381-mini-modulo-dimmer-zigbee-tuya-2-canais-profissional-_JM

### Database entry

{"id":3,"type":"Router","ieeeAddr":"0xa4c13899582c043c","nwkAddr":60083,"manufId":4417,"manufName":"_TZ3210_tkkb1ym8","powerSource":"Mains (single phase)","modelId":"TS110E","epList":[1,2,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[4,5,3,6,8,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"�sH,\u0013�zH,\u0013��H,\u0013","65506":54,"65508":0,"65534":0,"stackVersion":0,"dateCode":"","manufacturerName":"_TZ3210_tkkb1ym8","zclVersion":3,"appVersion":64,"modelId":"TS110E","powerSource":1}},"genOnOff":{"attributes":{"61440":0,"onOff":1,"onTime":0,"moesStartUpOnOff":0}},"genLevelCtrl":{"attributes":{"61440":828,"64514":0,"64515":323,"64516":831,"currentLevel":211}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x040d84fffe584724","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x040d84fffe584724","endpointID":1}],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":257,"inClusterList":[4,5,3,6,8,61184],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"61440":0,"onOff":1,"onTime":0,"moesStartUpOnOff":1}},"genLevelCtrl":{"attributes":{"61440":671,"64514":0,"64515":224,"64516":791,"currentLevel":171}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x040d84fffe584724","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x040d84fffe584724","endpointID":1}],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1479408709},"lastSeen":1689645719966,"defaultSendRequestWhen":"immediate"}

### Comments

I could not create an external converter, so I just changed the manufacturer name in tuya.ts file to test if it works.

The converter for this device works perfect, so my question is, how can I do an external converter exactly the same but change only the manufacturer?

Device: TS110E
From: _TZ3210_ngqk6jia 
To: _TZ3210_tkkb1ym8

queroautomacao, 2023-07-18:
zigbee-herdsman-converters/src/devices/tuya.ts

BlackStar-95, 2023-07-18:
and how did you managed to change the name in the file ?

Koenkk, 2023-07-18:
Added the `_TZ3210_tkkb1ym8`!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-07-18 by Koenkk

queroautomacao, 2023-07-18:
> Added the `_TZ3210_tkkb1ym8`!
> 
> Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

This is for the 2 channel Zigbee dimmer, please fix it and thank you very much for all great work! I will donate!