Koenkk/zigbee2mqtt #18352: Exception while calling fromZigbee converter: 'illuminance_calibration' is not a number, got string ()}

GShox, 2023-07-18:
### What happened?

All Philips Hue motion sensors are getting errors since ver. 1.32.0
Exception while calling fromZigbee converter: 'illuminance_calibration' is not a number, got string ()}

### What did you expect to happen?

No error

### Zigbee2MQTT version

1.32.0 / 1.32.1

### Adapter firmware version

20221226

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P

### Debug log

debug 2023-07-18 09:23:02: Received Zigbee message from 'Ute/HueEntre', type 'attributeReport', cluster 'msIlluminanceMeasurement', data '{"measuredValue":38535}' from endpoint 2 with groupID 0
error 2023-07-18 09:23:02: Exception while calling fromZigbee converter: 'illuminance_calibration' is not a number, got string ()}
debug 2023-07-18 09:23:02: Error: 'illuminance_calibration' is not a number, got string ()
    at toNumber (/app/node_modules/zigbee-herdsman-converters/src/lib/utils.ts:509:15)
    at calibrateAndPrecisionRoundOptions (/app/node_modules/zigbee-herdsman-converters/src/lib/utils.ts:89:29)
    at Object.convert (/app/node_modules/zigbee-herdsman-converters/src/converters/fromZigbee.js:471:30)
    at Receive.onDeviceMessage (/app/lib/extension/receive.ts:150:51)
    at EventEmitter.emit (node:events:525:35)
    at EventBus.emitDeviceMessage (/app/lib/eventBus.ts:102:22)
    at Controller.<anonymous> (/app/lib/zigbee.ts:106:27)
    at Controller.emit (node:events:513:28)
    at Controller.selfAndDeviceEmit (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:515:14)
    at Controller.onZclOrRawData (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:726:18)

debug 2023-07-18 09:23:02: Received Zigbee message from 'Sovrum/Hue', type 'attributeReport', cluster 'msIlluminanceMeasurement', data '{"measuredValue":0}' from endpoint 2 with groupID 0
error 2023-07-18 09:23:02: Exception while calling fromZigbee converter: 'illuminance_calibration' is not a number, got string ()}
debug 2023-07-18 09:23:02: Error: 'illuminance_calibration' is not a number, got string ()
    at toNumber (/app/node_modules/zigbee-herdsman-converters/src/lib/utils.ts:509:15)
    at calibrateAndPrecisionRoundOptions (/app/node_modules/zigbee-herdsman-converters/src/lib/utils.ts:89:29)
    at Object.convert (/app/node_modules/zigbee-herdsman-converters/src/converters/fromZigbee.js:471:30)
    at Receive.onDeviceMessage (/app/lib/extension/receive.ts:150:51)
    at EventEmitter.emit (node:events:525:35)
    at EventBus.emitDeviceMessage (/app/lib/eventBus.ts:102:22)
    at Controller.<anonymous> (/app/lib/zigbee.ts:106:27)
    at Controller.emit (node:events:513:28)
    at Controller.selfAndDeviceEmit (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:515:14)
    at Controller.onZclOrRawData (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:726:18)

labaland, 2023-07-18:
I get the same with my Aqara/Xiaomi Light sensor.

GShox, 2023-07-19:
Thanks for you assistance :)

I changed both these lines to be sure:
    illuminance_calibration: ''
    illuminance_lux_calibration: ''
to:
    illuminance_calibration: '0'
    illuminance_lux_calibration: '0'

Error is gone, and hope to see some lights on this evening!

labaland, 2023-08-19:
Still the same error!!! 😁

closed 2023-09-26 by github-actions[bot]

labaland, 2024-04-13:
Still getting theese errors spammed

smithbill17, 2024-11-08:
I'm getting this error in my HA Zigbee2MQTT, but it doesn't say which device is giving rise to the error.

Where do I find the "device config.yaml" that was mentioned above? I only have a "configuration.yaml" in Zigbee2MQTT but it doesn't contain illuminance_calibration and neither of the two 'external converters' I have either.

When I look through the 'Exposes' tab for all of my Zigbee2MQTT devices, none of them expose illuminance_calibration.

So I'm at a loss as to which device is causing this or where to find the mentioned "device config.yaml" is located?

smithbill17, 2024-11-08:
Okay, I found if I use the Zigbee2MQTT dashboard & navigate to each of my 'presence sensor' devices & then choose the 'Settings (specific)' tab, I can then enter a 0 into the edit-box & having done this for each device, it seems to resolve this error.

[image]

coasttech, 2025-02-08:
> Okay, I found if I use the Zigbee2MQTT dashboard & navigate to each of my 'presence sensor' devices & then choose the 'Settings (specific)' tab, I can then enter a 0 into the edit-box & having done this for each device, it seems to resolve this error.
> 
> [image]

Fixed my issue with the RB-SRAIN01

Thank you!