Koenkk/zigbee2mqtt #18360: Groups becomes unavailable on HA.

antibill51, 2023-07-19:
### What happened?

I have a similar problem  than[https://github.com/Koenkk/zigbee2mqtt/issues/9629]( https://github.com/Koenkk/zigbee2mqtt/issues/9629) but only concerns groups.
Sometimes groups become unavailable on the HA side.

  optimistic: true/false
  retain: true/false
in groups.yaml don't change anything.

Home Assistant 2023.7.2
Supervisor 2023.07.1
Operating System 10.3

Z2M 1.32.1-dev

### What did you expect to happen?

When restarting Z2M or HA, the discovery message should be sent for groups as well, not just for individual entities.

### How to reproduce it (minimal and precise)

create a group (of lights for example).
It is logically visible on the HA side.
Restarting the MQTT brocker. (OBLIGATORY?)
Restarting HA.
The group becomes unavailable.

The only way have found is to add an entity to each group that is already part of it so that the group entity reappears on the HA side.

### Zigbee2MQTT version

1.32.1-dev commit: 313865b

### Adapter firmware version

20220219

### Adapter

HamGeek CC2652P Tech

antibill51, 2023-08-21:
By adding 
`availability: false`
 for each group, the problem has not reoccurred for approximately 2 weeks.

Is it normal to have to go through this?
Thanks

rccoleman, 2023-08-27:
I've just been struggling with a similar issue where some groups have an `availability` state of `offline`, when I do not have availability enabled anywhere.  It only affected a subset of my groups, with no obvious commonality between the ones that were affected and the ones that were not.  I have `availability: false` at the bottom of my `configuration.yaml` (not put there explicitly, but I think it was automatic as the default), but I just added it to each group definition to see if it helps.

rccoleman, 2023-09-27:
Not stale, still looking into it.

closed 2023-11-04 by github-actions[bot]