Koenkk/zigbee2mqtt #18361: Sonoff ZBMINI - cannot pair - "Received message from unsupported device with Zigbee model undefined..."

S474N, 2023-07-19:
### What happened?

Cannot pair Sonoff ZBMINI

Number of Zigbee devices: 3

[image]

[image]

Received message from unsupported device with Zigbee model 'undefined' and manufactrurer name 'undefined':
[image]

[image]

Tested also second device, but same error:
[image]

Same device in Deconz/Phoscon via ConBee II:
[image]

### What did you expect to happen?

Pair Sonoff ZBMINI devices.

### How to reproduce it (minimal and precise)

Standard pairing procedure.

### Zigbee2MQTT version

1.32.1 (tested also with dev version)

### Adapter firmware version

20230507

### Adapter

SMLIGHT SLZB-05 CC2652P

### Debug log

[image]

francisp2, 2023-08-20:
Bring it close to the coordinator  and pair again

S474N, 2023-08-20:
@francisp2: tested 1m, 2m, 5m, 10m and nothing.

francisp2, 2023-08-20:
First remove it from Z2M, then pair again

[image]

francisp2, 2023-09-21:
Last one I installed was 'unsupported' too, but after clicking 'reconfigure' in the front-end it was found correctly.

closed 2023-10-30 by github-actions[bot]