Koenkk/zigbee2mqtt #18364: AduroSmart 81809 detected as Trust ZLED-RGB9 and power_on_behavior does not work

connelhooley, 2023-07-19:
### What happened?

I have an `AduroSmart 81809` but when I try and set the `power_on_behavior` attribute I get an error. I also see an error when setting `color_temp_startup`.

The bulb is listed as a `Trust ZLED-RGB9` (seems the opposite of #8313).

All other attributes work, I can turn the bulb off and on and I can change the bulb's colour etc.

I've tried using the `latest-dev` image but the behaviour is the same.

### What did you expect to happen?

I expected the attribute to be set as it is listed as supported on both bulbs.

I'm not sure if:
- The bulb doesn't actually support this functionality and the attribute should be removed from the UI.
- The bulb does support this functionality and the attribute should be renamed or changed in the UI.
- Other user's aren't seeing this and I have a local issue

### How to reproduce it (minimal and precise)

Pair the bulb and set the attribute

### Zigbee2MQTT version

1.32.1

### Adapter firmware version

6.10.3.0 build 297

### Adapter

SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20220813102406-if00

### Debug log

Pairing logs:
```
info  2023-07-19 11:46:49: Device '0x00158d000193e5e9' joined
info  2023-07-19 11:46:49: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00158d000193e5e9","ieee_address":"0x00158d000193e5e9"},"type":"device_joined"}'
info  2023-07-19 11:46:49: Starting interview of '0x00158d000193e5e9'
info  2023-07-19 11:46:50: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00158d000193e5e9","ieee_address":"0x00158d000193e5e9","status":"started"},"type":"device_interview"}'
info  2023-07-19 11:47:00: Successfully interviewed '0x00158d000193e5e9', device has successfully been paired
info  2023-07-19 11:47:00: Device '0x00158d000193e5e9' is supported, identified as: Trust Smart RGB LED bulb (ZLED-RGB9)
info  2023-07-19 11:47:00: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Smart RGB LED bulb","exposes":[{"features":[{"access":7,"description":"On/off state of this light","name":"state","property":"state","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"},{"access":7,"description":"Brightness of this light","name":"brightness","property":"brightness","type":"numeric","value_max":254,"value_min":0},{"access":7,"description":"Color temperature of this light","name":"color_temp","presets":[{"description":"Coolest temperature supported","name":"coolest","value":153},{"description":"Cool temperature (250 mireds / 4000 Kelvin)","name":"cool","value":250},{"description":"Neutral temperature (370 mireds / 2700 Kelvin)","name":"neutral","value":370},{"description":"Warm temperature (454 mireds / 2200 Kelvin)","name":"warm","value":454},{"description":"Warmest temperature supported","name":"warmest","value":500}],"property":"color_temp","type":"numeric","unit":"mired","value_max":500,"value_min":153},{"access":7,"description":"Color temperature after cold power on of this light","name":"color_temp_startup","presets":[{"description":"Coolest temperature supported","name":"coolest","value":153},{"description":"Cool temperature (250 mireds / 4000 Kelvin)","name":"cool","value":250},{"description":"Neutral temperature (370 mireds / 2700 Kelvin)","name":"neutral","value":370},{"description":"Warm temperature (454 mireds / 2200 Kelvin)","name":"warm","value":454},{"description":"Warmest temperature supported","name":"warmest","value":500},{"description":"Restore previous color_temp on cold power on","name":"previous","value":65535}],"property":"color_temp_startup","type":"numeric","unit":"mired","value_max":500,"value_min":153},{"access":7,"description":"Color of this light in the CIE 1931 color space (x/y)","features":[{"access":7,"name":"x","property":"x","type":"numeric"},{"access":7,"name":"y","property":"y","type":"numeric"}],"name":"color_xy","property":"color","type":"composite"}],"type":"light"},{"access":2,"description":"Triggers an effect on the light (e.g. make light blink for a few seconds)","name":"effect","property":"effect","type":"enum","values":["blink","breathe","okay","channel_change","finish_effect","stop_effect"]},{"access":7,"description":"Controls the behavior when the device is powered on after power loss","name":"power_on_behavior","property":"power_on_behavior","type":"enum","values":["off","on","toggle","previous"]},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"ZLED-RGB9","options":[{"access":2,"description":"Controls the transition time (in seconds) of on/off, brightness, color temperature (if applicable) and color (if applicable) changes. Defaults to `0` (no transition).","name":"transition","property":"transition","type":"numeric","value_min":0},{"access":2,"description":"When enabled colors will be synced, e.g. if the light supports both color x/y and color temperature a conversion from color x/y to color temperature will be done when setting the x/y color (default true).","name":"color_sync","property":"color_sync","type":"binary","value_off":false,"value_on":true},{"access":2,"description":"State actions will also be published as 'action' when true (default false).","name":"state_action","property":"state_action","type":"binary","value_off":false,"value_on":true}],"supports_ota":false,"vendor":"Trust"},"friendly_name":"0x00158d000193e5e9","ieee_address":"0x00158d000193e5e9","status":"successful","supported":true},"type":"device_interview"}'
info  2023-07-19 11:47:00: Configuring '0x00158d000193e5e9'
debug 2023-07-19 11:47:00: Received Zigbee message from '0x00158d000193e5e9', type 'readResponse', cluster 'lightingColorCtrl', data '{"colorCapabilities":31}' from endpoint 2 with groupID 0
debug 2023-07-19 11:47:00: Missing colorTempPhysicalMin and/or colorTempPhysicalMax for endpoint 0x00158d000193e5e9!
info  2023-07-19 11:47:00: MQTT publish: topic 'zigbee2mqtt/0x00158d000193e5e9', payload '{"color_mode":"xy","linkquality":160,"power_on_behavior":null,"state":null}'
debug 2023-07-19 11:47:00: Received Zigbee message from '0x00158d000193e5e9', type 'readResponse', cluster 'lightingColorCtrl', data '{"colorTempPhysicalMax":500,"colorTempPhysicalMin":153}' from endpoint 2 with groupID 0
info  2023-07-19 11:47:00: MQTT publish: topic 'zigbee2mqtt/0x00158d000193e5e9', payload '{"color_mode":"xy","linkquality":156,"power_on_behavior":null,"state":null}'
info  2023-07-19 11:47:00: Successfully configured '0x00158d000193e5e9'
```

Set attribute logs:
```
debug 2023-07-19 11:48:07: Received MQTT message on 'zigbee2mqtt/0x00158d000193e5e9/set' with data '{"power_on_behavior":"previous"}'
debug 2023-07-19 11:48:07: Publishing 'set' 'power_on_behavior' to '0x00158d000193e5e9'
error 2023-07-19 11:48:07: Publish 'set' 'power_on_behavior' to '0x00158d000193e5e9' failed: 'Error: Write 0x00158d000193e5e9/2 genOnOff({"startUpOnOff":255}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
debug 2023-07-19 11:48:07: Error: Write 0x00158d000193e5e9/2 genOnOff({"startUpOnOff":255}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:419:28)
    at Endpoint.write (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:488:22)
    at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/src/converters/toZigbee.js:116:13)
    at Publish.onMQTTMessage (/app/lib/extension/publish.ts:248:36)
```

Koenkk, 2023-07-19:
Can you provide the data/database.db entry of this device?

connelhooley, 2023-07-21:
> Can you provide the data/database.db entry of this device?

``` json
{
  "id": 35,
  "type": "Router",
  "ieeeAddr": "0x00158d000193e5e9",
  "nwkAddr": 48675,
  "manufId": 4151,
  "manufName": "Trust International B.V.\u0000",
  "powerSource": "Mains (single phase)",
  "modelId": "ZLL-ExtendedColo",
  "epList": [
    1,
    2
  ],
  "endpoints": {
    "1": {
      "profId": 49246,
      "epId": 1,
      "devId": 4096,
      "inClusterList": [
        4096
      ],
      "outClusterList": [
        4096
      ],
      "clusters": {},
      "binds": [],
      "configuredReportings": [],
      "meta": {}
    },
    "2": {
      "profId": 49246,
      "epId": 2,
      "devId": 528,
      "inClusterList": [
        0,
        4,
        3,
        6,
        8,
        5,
        768,
        65535,
        25
      ],
      "outClusterList": [
        0,
        4,
        3,
        6,
        8,
        5,
        768,
        25
      ],
      "clusters": {
        "lightingColorCtrl": {
          "attributes": {
            "colorCapabilities": 31,
            "colorTempPhysicalMin": 153,
            "colorTempPhysicalMax": 500
          }
        },
        "genLevelCtrl": {
          "attributes": {}
        }
      },
      "binds": [],
      "configuredReportings": [],
      "meta": {
        "onLevelSupported": false
      }
    }
  },
  "appVersion": 1,
  "stackVersion": 1,
  "hwVersion": 1,
  "dateCode": "20160916",
  "swBuildId": "1000-0003",
  "zclVersion": 1,
  "interviewCompleted": true,
  "meta": {
    "configured": 88764544
  },
  "lastSeen": 1689964491153,
  "defaultSendRequestWhen": "immediate"
}
```

Koenkk, 2023-07-23:
Seems this is a trust device internally so we cannot distinguish it (`"manufName": "Trust International B.V.\u0000",`). I've disabled the power on behaviour.

closed 2023-07-23 by Koenkk