Koenkk/zigbee2mqtt #18369: Aqara QBKG12LM / QBKG11LM wall switches randomly turns off in decoupled mode

midlan, 2023-07-19:
### What happened?

I have couple of QBKG12LM / QBKG11LM wall switches (version with neutral) in my home. I am using three of them in decoupled mode and the problem is, that the relay inside the switch goes off randomly in some of the switches pretty soon without any reason (no automation triggered). Then I only see this report in HA logbook:

[image]

I've found `"state":"OFF"` in logs:
```
Zigbee2MQTT:info  2023-07-19 22:05:38: MQTT publish: topic 'zigbee2mqtt/VYP dětský pokoj', payload '{"action":null,"consumption":13.72,"device_temperature":31,"energy":13.72,"linkquality":109,"operation_mode":"decoupled","power":0,"power_outage_count":38326,"state":"OFF","update":{"installed_version":94,"latest_version":94,"state":"idle"},"update_available":false}'
````
Full log file below.

I think the problem started as soon as I started using decoupled mode on three switches. Currently using decoupled with 2x QBKG11LM + 1x QBKG12LM. Previusly I used only 1x QBKG12LM in decoupled mode in whole network (+ the others in coupled mode).

### What did you expect to happen?

I expect the switches to stay in ON state all the time (unless changed to OFF manually).

### Zigbee2MQTT version

1.32.1 (HA OS addon 1.32.1-1)

### Adapter firmware version

0.9.9 (Mar 3 2023)

### Adapter

SLZB-06 ethernet mode + PoE

### Debug log

Logs file: 
[vyp_detsky_22_05_37.log](https://github.com/Koenkk/zigbee2mqtt/files/12100025/vyp_detsky_22_05_37.log)

midlan, 2023-08-19:
problem is still present

closed 2023-09-26 by github-actions[bot]

softweed, 2023-10-02:
Having the same problem

midlan, 2023-10-02:
I forgot to mention, the problem is present on all switches, not only the decoupled ones, as I wrote in the initial post. (except one switch I described in comment above)

midlan, 2023-10-17:
Ok, seems every switch at my home is affected. Some more, some less.

My zigbee2mqtt are flooded by such messages:
```
Debug 2023-10-17 14:02:28Received Zigbee message from 'VYP předsíň rozcestí (řídící)', type 'attributeReport', cluster 'genBasic', data '{"65521":0}' from endpoint 1 with groupID 0
Debug 2023-10-17 14:02:28QBKG11LM: unknown key 65521 with value 0
Debug 2023-10-17 14:02:28QBKG11LM: Processed data into payload {}
Debug 2023-10-17 14:02:30Received Zigbee message from 'VYP ložnice', type 'attributeReport', cluster 'genBasic', data '{"65521":0}' from endpoint 1 with groupID 0
Debug 2023-10-17 14:02:30QBKG11LM: unknown key 65521 with value 0
Debug 2023-10-17 14:02:30QBKG11LM: Processed data into payload {}
Debug 2023-10-17 14:02:31Received Zigbee message from 'VYP koupelna', type 'attributeReport', cluster 'genOnOff', data '{"61440":50300672,"onOff":0}' from endpoint 1 with groupID 0
Info 2023-10-17 14:02:31MQTT publish: topic 'zigbee2mqtt/VYP koupelna', payload '{"action":null,"consumption":67.43,"device_temperature":27,"energy":67.43,"linkquality":120,"operation_mode_left":null,"operation_mode_right":null,"power":0,"power_outage_count":26168,"state_left":"OFF","state_right":"OFF","update":{"installed_version":94,"latest_version":94,"state":"idle"},"update_available":false}'
Debug 2023-10-17 14:02:31Received Zigbee message from 'VYP koupelna', type 'attributeReport', cluster 'genOnOff', data '{"61440":50300672,"onOff":0}' from endpoint 2 with groupID 0
Info 2023-10-17 14:02:31MQTT publish: topic 'zigbee2mqtt/VYP koupelna', payload '{"action":null,"consumption":67.43,"device_temperature":27,"energy":67.43,"linkquality":116,"operation_mode_left":null,"operation_mode_right":null,"power":0,"power_outage_count":26168,"state_left":"OFF","state_right":"OFF","update":{"installed_version":94,"latest_version":94,"state":"idle"},"update_available":false}'
Debug 2023-10-17 14:02:32Received Zigbee message from 'VYP koupelna', type 'attributeReport', cluster 'genBasic', data '{"65281":{"100":0,"101":0,"152":0,"154":16,"3":27,"5":26170,"8":4446,"9":513}}' from endpoint 1 with groupID 0
Debug 2023-10-17 14:02:32QBKG12LM: Processed data into payload {"device_temperature":27,"power_outage_count":26169,"state_left":"OFF","state_right":"OFF","power":0}
Debug 2023-10-17 14:02:32QBKG12LM: Processed data into payload {"device_temperature":27,"power_outage_count":26169,"state_left":"OFF","state_right":"OFF","power":0}
Info 2023-10-17 14:02:32MQTT publish: topic 'zigbee2mqtt/VYP koupelna', payload '{"action":null,"consumption":67.43,"device_temperature":27,"energy":67.43,"linkquality":120,"operation_mode_left":null,"operation_mode_right":null,"power":0,"power_outage_count":26169,"state_left":"OFF","state_right":"OFF","update":{"installed_version":94,"latest_version":94,"state":"idle"},"update_available":false}'
Debug 2023-10-17 14:02:33Device 'VYP koupelna' announced itself
Info 2023-10-17 14:02:33MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"VYP koupelna","ieee_address":"0x00158d00045ae3dc"},"type":"device_announce"}'
Info 2023-10-17 14:02:33MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"VYP koupelna"},"type":"device_announced"}'
Debug 2023-10-17 14:02:34Received Zigbee message from 'VYP ložnice', type 'attributeReport', cluster 'genBasic', data '{"65521":0}' from endpoint 1 with groupID 0
Debug 2023-10-17 14:02:34QBKG11LM: unknown key 65521 with value 0
Debug 2023-10-17 14:02:34QBKG11LM: Processed data into payload {}
Debug 2023-10-17 14:02:35Retrieving state of 'VYP koupelna' after reconnect
Debug 2023-10-17 14:02:35Received Zigbee message from 'VYP koupelna', type 'readResponse', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Info 2023-10-17 14:02:35MQTT publish: topic 'zigbee2mqtt/VYP koupelna', payload '{"action":null,"consumption":67.43,"device_temperature":27,"energy":67.43,"linkquality":120,"operation_mode_left":null,"operation_mode_right":null,"power":0,"power_outage_count":26169,"state_left":"OFF","state_right":"OFF","update":{"installed_version":94,"latest_version":94,"state":"idle"},"update_available":false}'
Debug 2023-10-17 14:02:36Received Zigbee message from 'VYP koupelna', type 'read', cluster 'genTime', data '["time"]' from endpoint 1 with groupID 0
Debug 2023-10-17 14:02:36Received Zigbee message from 'VYP obývák vlevo (řídící)', type 'attributeReport', cluster 'genOnOff', data '{"61440":37519616,"onOff":0}' from endpoint 1 with groupID 0
Info 2023-10-17 14:02:36MQTT publish: topic 'zigbee2mqtt/VYP obývák vlevo (řídící)', payload '{"action":null,"consumption":114.95,"device_temperature":26,"energy":114.95,"linkquality":149,"operation_mode_left":"decoupled","operation_mode_right":"decoupled","power":0,"power_outage_count":9240,"state_left":"OFF","state_right":"OFF","update":{"installed_version":94,"latest_version":94,"state":"idle"},"update_available":false}'
Debug 2023-10-17 14:02:36Received Zigbee message from 'VYP obývák vlevo (řídící)', type 'attributeReport', cluster 'genOnOff', data '{"61440":37519616,"onOff":0}' from endpoint 2 with groupID 0
Info 2023-10-17 14:02:36MQTT publish: topic 'zigbee2mqtt/VYP obývák vlevo (řídící)', payload '{"action":null,"consumption":114.95,"device_temperature":26,"energy":114.95,"linkquality":142,"operation_mode_left":"decoupled","operation_mode_right":"decoupled","power":0,"power_outage_count":9240,"state_left":"OFF","state_right":"OFF","up [... cut ...]
Info 2023-10-17 14:02:37MQTT publish: topic 'zigbee2mqtt/VYP obývák vlevo (řídící)', payload '{"action":null,"consumption":114.95,"device_temperature":26,"energy":114.95,"linkquality":153,"operation_mode_left":"decoupled","operation_mode_right":"decoupled","power":0,"power_outage_count":9241,"state_left":"OFF","state_right":"OFF","update":{"installed_version":94,"latest_version":94,"state":"idle"},"update_available":false}'
```` 
Seems like the switches are disconnecting and reconnecting many times every minute. And sometimes they loose their state, so the relay goes off.

@softweed is your log flooded too?