Koenkk/zigbee2mqtt #18371: ZYXH Tuya 12 Way Relay Switch Module (_TZE204_dqolcpcp)

Franckybel, 2023-07-20:
### Link

https://www.aliexpress.com/item/1005005605518721.html

### Database entry

{"id":2,"type":"Router","ieeeAddr":"0xa4c138ffaad02e58","nwkAddr":56168,"manufId":4417,"manufName":"_TZE204_dqolcpcp","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"��J,i��J,i��J,i��J,i��J,i��J,i��J,i��J,i��J,i��J,i","65506":56,"65508":0,"65534":0,"modelId":"TS0601","manufacturerName":"_TZE204_dqolcpcp","powerSource":1,"zclVersion":3,"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-708457359},"lastSeen":1689781210113,"defaultSendRequestWhen":"immediate"}

### Comments

This Device already exists in supported devices list **but with only 8 way Relay Switch.**
For your information, it's existing also with 24 Way Relay Switch.

I have created the external converter based on the one already supported (8 way Relay Switch) by zigbee2Mqtt
https://www.zigbee2mqtt.io/devices/TS0601_switch_8.html

Device Picture
[image]

Thanks in advance to support this new hardware

Franck

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE204_dqolcpcp',
        },
    ],
    model: 'TS0601_12Swtichs',
    vendor: 'TuYa',
    description: '12 Switch Tuya ZYXH',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
     exposes: [
            tuya.exposes.switch().withEndpoint('l1'),
            tuya.exposes.switch().withEndpoint('l2'),
            tuya.exposes.switch().withEndpoint('l3'),
            tuya.exposes.switch().withEndpoint('l4'),
            tuya.exposes.switch().withEndpoint('l5'),
            tuya.exposes.switch().withEndpoint('l6'),
            tuya.exposes.switch().withEndpoint('l7'),
            tuya.exposes.switch().withEndpoint('l8'),
			tuya.exposes.switch().withEndpoint('l9'),
			tuya.exposes.switch().withEndpoint('l10'),
			tuya.exposes.switch().withEndpoint('l11'),
			tuya.exposes.switch().withEndpoint('l12'),

        ],
  endpoint: (device) => {
            return {'l1': 1, 'l2': 1, 'l3': 1, 'l4': 1, 'l5': 1, 'l6': 1, 'l7': 1, 'l8': 1,'l9': 1,'l10': 1,'l11': 1,'l12': 1};
        },
   meta: {
            multiEndpoint: true,
            tuyaDatapoints: [
                [1, 'state_l1', tuya.valueConverter.onOff],
                [2, 'state_l2', tuya.valueConverter.onOff],
                [3, 'state_l3', tuya.valueConverter.onOff],
                [4, 'state_l4', tuya.valueConverter.onOff],
                [5, 'state_l5', tuya.valueConverter.onOff],
                [6, 'state_l6', tuya.valueConverter.onOff],
                [0x65, 'state_l7', tuya.valueConverter.onOff],
                [0x66, 'state_l8', tuya.valueConverter.onOff],
				[0x67, 'state_l9', tuya.valueConverter.onOff],
				[0x68, 'state_l10', tuya.valueConverter.onOff],
				[0x69, 'state_l11', tuya.valueConverter.onOff],
				[0x6A, 'state_l12', tuya.valueConverter.onOff],
            ],
    },
  whiteLabel: [
          tuya.whitelabel('ZYXH', 'TS0601_switch_12', '12 Gang switch', ['_TZE204_dqolcpcp']),
       ],
};

module.exports = definition;
```

luixal, 2023-08-02:
Wow, just got this device today and noticed it wasn't supported. Hope this gets merged soon :)

Thanks for your work!

Franckybel, 2023-08-02:
Hello @luixal, you can create your proper device definition using unsupported device process without waiting official support device.
Sorry it's in French (you can use Google translator) but I have describe how to do it to Jeedom community.  See my last post [Jeedom community post](https://community.jeedom.com/t/comment-faire-reconnaitre-un-nouveau-materiel-zigbee-non-supporte-avec-zigbee2mqtt-carte-relais-12-voies-pour-arrosage-automatique/109241/15)

luixal, 2023-08-04:
Hi @Franckybel , I know, when it failed to pair I went into the advanced page to add support for the device. Then I search in the issues and find that you already did :)

Anyway, I'm still missing the power unit I'm gonna use with this devices so... no hurry at all. If supported hasn't been merged by then, I'll add it manually.

Thanks for your work! :)

roastcom, 2023-08-14:
Amazing to find the device already listed, but does it look like the converter only supports what they call 'interlocking mode' on the device page? I.e. standard switch on-off behaviour.

How would I change that to 'press to on' / auto off behaviour?

Franckybel, 2023-08-14:
Hello
It looks that these options cannot be set into hardware itself (like other can), you will need the tuya app and remote wireless. 
I'm managing these features via my home automation.

Franckybel, 2023-09-15:
After 2 months, this device is still not part of the official device supporter.
How can I be sure that this topic has been taken into account for future device integration?

luixal, 2023-09-15:
Hi,

I had a look at the page for [adding support for new devices](https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_4-add-device-picture-to-zigbee2mqtt-io-documentation) . In the 4th point they explain how to make a PR here for adding documentation and another PR in this [other repo](https://github.com/Koenkk/zigbee-herdsman-converters) for it to be supported.

Will be great if you could do that so it gets officially supported :)

Koenkk, 2023-09-16:
Added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-09-16 by Koenkk