Koenkk/zigbee2mqtt #18374: Cannot start zigbee2mqtt SyntaxError: Unexpected token '('

giulup, 2023-07-20:
### What happened?

Hey everyone.

I have a major problem. 
I had zigbee2mqtt working fine. Today i received some new devices and for them to work i had to update (for the first time) zigbee2mqtt from version 1.24.

For this i uninstalled zigbee2mqtt and reinstalled according to the  docunmentation. Using my old configuration.yaml and at the point of starting i receive errors:

> zigbee2mqtt@1.32.1 start /opt/zigbee2mqtt
> node index.js

/opt/zigbee2mqtt/node_modules/rimraf/node_modules/minimatch/dist/cjs/ast.js:86
    #fillNegs() {
             ^

SyntaxError: Unexpected token '('
    at wrapSafe (internal/modules/cjs/loader.js:915:16)
    at Module._compile (internal/modules/cjs/loader.js:963:27)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1027:10)
    at Module.load (internal/modules/cjs/loader.js:863:32)
    at Function.Module._load (internal/modules/cjs/loader.js:708:14)
    at Module.require (internal/modules/cjs/loader.js:887:19)
    at require (internal/modules/cjs/helpers.js:74:18)
    at Object.<anonymous> (/opt/zigbee2mqtt/node_modules/rimraf/node_modules/minimatch/dist/cjs/index.js:9:18)
    at Module._compile (internal/modules/cjs/loader.js:999:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1027:10)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.32.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.32.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/USER/.npm/_logs/2023-07-20T14_21_36_787Z-debug.log

Did someone run into such problems or knows how to solve?

### Zigbee2MQTT version

1.3.1

### Adapter firmware version

000

### Adapter

Sonoff_Zigbee_3.0_USB_Dongle_Plus

Petorrrrr, 2023-07-25:
I have almost the exact same error. Referencing slightly different rows/columns though:

/opt/zigbee2mqtt/node_modules/rimraf/node_modules/minimatch/dist/cjs/ast.js:86
    #fillNegs() {
                  ^
SyntaxError: Unexpected token '('
    at wrapSafe (internal/modules/cjs/loader.js:1053:16)
    at Module._compile (internal/modules/cjs/loader.js:1101:27)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1157:10)
    at Module.load (internal/modules/cjs/loader.js:985:32)
    at Function.Module._load (internal/modules/cjs/loader.js:878:14)
    at Module.require (internal/modules/cjs/loader.js:1025:19)
    at require (internal/modules/cjs/helpers.js:72:18)
    at Object.<anonymous> (/opt/zigbee2mqtt/node_modules/rimraf/node_modules/minimatch/dist/cjs/index.js:9:18)
    at Module._compile (internal/modules/cjs/loader.js:1137:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1157:10)

Koenkk, 2023-07-25:
This is indeed caused by an outdated and unsupported Node version, updating your node version should fix it.

closed 2023-07-25 by Koenkk