Koenkk/zigbee2mqtt #18379: Issue with installing it on proxmox and using usb passthrough

Londonlad95, 2023-07-20:
### What happened?

When I try to launch zigbee2mqtt I get the following errors:

Error: Error while opening serialport 'Error: Error: Inappropriate ioctl for device setting custom baud rate of 115200'
    at SerialPort.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:146:28)
    at SerialPort._error (/opt/zigbee2mqtt/node_modules/@serialport/stream/dist/index.js:75:22)
    at /opt/zigbee2mqtt/node_modules/@serialport/stream/dist/index.js:111:18

My config file is as follows:
```
homeassistant:
 legacy_triggers: true
  status_topic: hass/status
  legacy_entity_attributes: true
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://homeassistant.local/:1883
  user: ***
  password: ***
  keepalive: 60
  reject_unauthorized: true
  version: 4
  include_device_information: true
serial:
  port:  /dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_*****
frontend:
  port: 9442
  url: null
  host: 0.0.0.0
advanced:
  network_key:
    ***[Redacted]***
  homeassistant_legacy_entity_attributes: false
  legacy_api: true
  legacy_availability_payload: true
  log_syslog:
    app_name: Zigbee2MQTT
    eol: /n
    host: localhost
    localhost: localhost
    path: /dev/log
    pid: process.pid
    port: 123
    protocol: tcp4
    type: '5424'
  last_seen: ISO_8601_local
  log_level: debug
device_options:
  legacy: false
devices:
 minimized
groups:
 minimized
availability:
  active:
    timeout: 10
  passive:
    timeout: 1500

```

### What did you expect to happen?

For zigbee2mqtt to launch

### How to reproduce it (minimal and precise)

All I do is enter this command:
cd /opt/zigbee2mqtt && npm start

### Zigbee2MQTT version

1.32.1

### Adapter firmware version

unsure as I can't launch zigbee2mqtt currently

### Adapter

Sonoff

platte-73, 2023-07-21:
Did you add hardware USB device in the proxmox sever

Londonlad95, 2023-07-23:
> Did you add hardware USB device in the proxmox sever

Yes I did. 

The issue was I was following a guide which listed the config path incorrectly.

For anyone with the same issue as me, use the correct path. 

This is what solved it for me: [https://smarthomescene.com/guides/how-to-separate-zigbee2mqtt-from-home-assistant-in-proxmox/#comment-2519](url)

Londonlad95, 2023-07-23:
All now sorted

closed 2023-07-23 by Londonlad95