Koenkk/zigbee2mqtt #18381: IKEA Rodret can't be paired

Jaggenite, 2023-07-21:
### What happened?

Trying to pair IKEA Rodret to z2m, but get this error message: 

Device 'Oxb43522fffe0a057e'
with Zigbee model 'RODRET
Dimmer' and manufacturer
name 'IKEA of Sweden' is NOT
supported

### What did you expect to happen?

It should be paired

### How to reproduce it (minimal and precise)

Reset the RODRET device and wait for pairing to complete 

### Zigbee2MQTT version

1.32.1

### Adapter firmware version

7.1.1.0 build 273

### Adapter

Home Assistant Skyconnect

Koenkk, 2023-07-23:
This device will be supported in the next z2m release on 1 August

closed 2023-07-23 by Koenkk

Jaggenite, 2023-07-23:
> This device will be supported in the next z2m release on 1 August

Thanks :)

mcnallylaurence, 2023-08-06:
Still not seeing this supported?

bogd, 2023-08-07:
It does show up with a name this time (correctly identified as "Rodret dimmer"), but it is still unsupported. 

Any chance we could reopen this?