Koenkk/zigbee2mqtt #18382: Ikea FYRTUR cover `request.frame.Payload.filter is not a function`

arough007, 2023-07-21:
### What happened?

When I try to change the height of the cover I get the following error and the position of the cover doesn't change.

```
Error 2023-07-21 09:54:41Publish 'set' 'position' to 'Rollo Wohnzimmer Max' failed: 'TypeError: Command 0x2c1165fffea17a4d/1 closuresWindowCovering.goToLiftPercentage({"percentageliftvalue":65}, {"sendWhen":"active","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (request.frame.Payload.filter is not a function)'
Debug 2023-07-21 09:54:41TypeError: Command 0x2c1165fffea17a4d/1 closuresWindowCovering.goToLiftPercentage({"percentageliftvalue":65}, {"sendWhen":"active","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (request.frame.Payload.filter is not a function) at Endpoint.filterRequests (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:346:67) at Endpoint.sendRequest (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:383:18) at Endpoint.command (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:833:39) at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/src/converters/toZigbee.js:436:26) at Publish.onMQTTMessage (/app/lib/extension/publish.ts:248:52) at EventEmitter.emit (node:events:525:35) at EventBus.emitMQTTMessage (/app/lib/eventBus.ts:109:22) at MQTT.onMessage (/app/lib/mqtt.ts:140:27) at MqttClient.emit (node:events:513:28) at MqttClient._handlePublish (/app/node_modules/mqtt/lib/client.js:1547:12)
```

### What did you expect to happen?

I expected the cover to change position.

### How to reproduce it (minimal and precise)

Just change the position of the cover.
Either via z2m directly or via HA.

### Zigbee2MQTT version

1.32.1

### Adapter firmware version

20210708

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/12117942/log.txt)

freelancer42, 2023-08-01:
I have the same (or at least similar) issue with my IKEA Fyrtur (firmware 24.4.11), using Z2M 1.32.1 with a SONOFF ZBDongle-E.
The issue seems to happen when the blind hasn't been active (moved) for a while. Trying to set it to a specific position yields the error message in the OP.
Trying to set the state to OPEN or CLOSE instead of setting a position always seems to work, and after doing that I can set the position without issue. But then it stops working again the next day.

dwradcliffe, 2023-08-08:
Seeing the same error with my IKEA PRAKTLYSING also on 24.4.11.

Linuus, 2023-08-15:
I get the same with my Fyrtur now after updating to 1.32.2.

Any progress on this issue?

Koenkk, 2023-08-15:
Pushed a fix, let me know if it works.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-09-23 by github-actions[bot]