Koenkk/zigbee2mqtt #18383: [New device support]:

shing6326, 2023-07-21:
### Link

https://la.manuals.plus/philips/hue-go-bt-portable-lamp-manual

### Database entry

{"id":8,"type":"Router","ieeeAddr":"0x001788010cb452cf","nwkAddr":30237,"manufId":4107,"manufName":"Signify Netherlands B.V.","powerSource":"Mains (single phase)","modelId":"929003128601","epList":[11,242],"endpoints":{"11":{"profId":260,"epId":11,"devId":269,"inClusterList":[0,3,4,5,6,8,4096,768,64513],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"929003128601","manufacturerName":"Signify Netherlands B.V.","powerSource":1,"zclVersion":2,"appVersion":2,"stackVersion":1,"hwVersion":0,"dateCode":"20210305","swBuildId":"1.82.10"}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":1,"hwVersion":0,"dateCode":"20210305","swBuildId":"1.82.10","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1689931921744,"defaultSendRequestWhen":"immediate"}

### Comments

https://www.zigbee2mqtt.io/devices/929003128401.html
the philips hue lamp contains different sku from different countries and color, my 929003128601 doesn't support it from current zigbee2mqtt

SKU | Color | Country
929003128401 | Album | EU/UK
929003128501 | Black | EU/UK
929003128601 | Album | AUSTER
929003128701 | Black | AUSTER
929003521401 | Album | EU/UK
929003521501 | Black | EU/UK
929003521601 | Album | AUSTER
929003521701 | Black | AUSTER
929003128801 | Album | AU/NZ
929003128901 | Album | CN
929003129001 | Album | JP/K/TW

shing6326, 2023-07-22:
https://github.com/Koenkk/zigbee-herdsman-converters/pull/5994

Koenkk, 2023-07-23:
PR merged so assuming this can be closed now.

closed 2023-07-23 by Koenkk