Koenkk/zigbee2mqtt #18384: [New device support]: 3 Phase Electricity Meter Unsupported device model 'TS0601' and manufacturer name '_TZE200_ywbltqzw'

evsmorodin, 2023-07-21:
### Link

3 Phase Electricity Meter Unsupported device model 'TS0601' and manufacturer name '_TZE200_ywbltqzw'

### Database entry

{"id":21,"type":"Router","ieeeAddr":"0xa4c138d5be107401","nwkAddr":27327,"manufId":4417,"manufName":"_TZE200_ywbltqzw","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"��\u0014+i��\u0014+i��\u0014+i��\u0014+i��\u0014+i��\u0014+i��\u0014+i��\u0014+i��\u0014+i��\u0014+i","65506":48,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE200_ywbltqzw","powerSource":1,"zclVersion":3,"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1669465017123,"defaultSendRequestWhen":"immediate"}

### Comments

The device does not show any parameters and they cannot be controlled
[image]
[image]
[image]
[image]
[image]

I tried to create a converter but it came to nothing...
[image]
[image]

For example [XOCA DAC2161C (MatSee Plus) ](https://www.zigbee2mqtt.io/devices/DAC2161C.html)
[image]
[image]

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE200_ywbltqzw',
        },
    ],
    model: 'PJ-DAC7361C',
    vendor: 'TuYa',
    description: 'Zigbee DIN 3-phase energy meter',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    whiteLabel: [{vendor: 'XOCA', model: 'DAC7361C'}],
    //onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [
        e.voltage(), e.current(), e.power(), e.energy(),
        tuya.exposes.switch(),
        tuya.exposes.energyWithPhase('a'), tuya.exposes.energyWithPhase('b'), tuya.exposes.energyWithPhase('c'),
        tuya.exposes.voltageWithPhase('a'), tuya.exposes.voltageWithPhase('b'), tuya.exposes.voltageWithPhase('c'),
        tuya.exposes.powerWithPhase('a'), tuya.exposes.powerWithPhase('b'), tuya.exposes.powerWithPhase('c'),
        tuya.exposes.currentWithPhase('a'), tuya.exposes.currentWithPhase('b'), tuya.exposes.currentWithPhase('c'),
        tuya.exposes.powerFactorWithPhase('a'), tuya.exposes.powerFactorWithPhase('b'), tuya.exposes.powerFactorWithPhase('c'),
        exposes.enum('fault', ea.STATE, ['clear', 'over_current_threshold', 'over_power_threshold',
            'over_voltage threshold', 'wrong_frequency_threshold']).withDescription('Fault status of the device (clear = nothing)'),
        exposes.enum('threshold_1', ea.STATE, ['not_set', 'over_current_threshold', 'over_voltage_threshold'])
            .withDescription('State of threshold_1'),
        exposes.binary('threshold_1_protection', ea.STATE, 'ON', 'OFF')
            .withDescription('OFF - alarm only, ON - relay will be off when threshold reached'),
        exposes.numeric('threshold_1_value', ea.STATE)
            .withDescription('Can be in Volt or Ampere depending on threshold setting. Setup the value on the device'),
        exposes.enum('threshold_2', ea.STATE, ['not_set', 'over_current_threshold', 'over_voltage_threshold'])
            .withDescription('State of threshold_2'),
        exposes.binary('threshold_2_protection', ea.STATE, 'ON', 'OFF')
            .withDescription('OFF - alarm only, ON - relay will be off when threshold reached'),
        exposes.numeric('threshold_2_value', ea.STATE)
            .withDescription('Setup value on the device'),
        exposes.binary('clear_fault', ea.STATE_SET, 'ON', 'OFF')
            .withDescription('Turn ON to clear last the fault'),
            exposes.text('meter_id', ea.STATE).withDescription('Meter ID (ID of device)'),
    ],
    meta: {
        // All datapoints go in here
        tuyaDatapoints: [
            [1, 'energy', tuya.valueConverter.divideBy100],
            [3, null, null], // Monthly, but sends data only after request
            [4, null, null], // Dayly, but sends data only after request
            [6, null, tuya.valueConverter.phaseVariant2WithPhase('a')],
            [7, null, tuya.valueConverter.phaseVariant2WithPhase('b')],
            [8, null, tuya.valueConverter.phaseVariant2WithPhase('c')],
            [9, 'power', tuya.valueConverter.raw],
            [10, 'fault', tuya.valueConverterBasic.lookup({'clear': 0, 'over_current_threshold': 1,
                'over_power_threshold': 2, 'over_voltage_threshold': 4, 'wrong_frequency_threshold': 8})],
            [11, null, null], // Frozen - strange function, in native app - nothing is clear
            [16, 'state', tuya.valueConverter.onOff],
            [17, null, tuya.valueConverter.threshold], // It's settable, but can't write converter
            [18, 'meter_id', tuya.valueConverter.raw],
            [20, 'clear_fault', tuya.valueConverter.onOff], // Clear fault
            [21, null, null], // Forward Energy T1 - don't know what this
            [22, null, null], // Forward Energy T2 - don't know what this
            [23, null, null], // Forward Energy T3 - don't know what this
            [24, null, null], // Forward Energy T4 - don't know what this
        ],
    },
};

module.exports = definition;
```

markus-barta, 2023-07-29:
Almost the same hardware here:
Zigbee Model: TS0601
Zigbee Manufacturer: _TZE204_x8fp01wi

evsmorodin, 2023-08-29:
Issue is still relevant

evsmorodin, 2023-09-07:
I made a mistake and the converter didn't load
after correction, there are results
[image]
[image]
[image]
[image]

evsmorodin, 2023-10-16:
@Koenkk please open this issue

jujumo, 2023-11-12:
Same hardware here and same problem.

evsmorodin, 2024-05-11:
> @evsmorodin to properly support this device, you have to figure out all the datapoints: https://www.zigbee2mqtt.io/advanced/support-new-devices/02_support_new_tuya_devices.html

Hello, @Koenkk! For a long time there was no time for smart devices. Today I set aside time for this. So, the datapoints for the device are:
```
{
    "result": [
        {
            "dpId": 1,
            "dpName": "Forward Energy Total"
        },
        {
            "dpId": 3,
            "dpName": "Month energy"
        },
        {
            "dpId": 4,
            "dpName": "Daily energy"
        },
        {
            "dpId": 6,
            "dpName": "Phase A"
        },
        {
            "dpId": 7,
            "dpName": "Phase B"
        },
        {
            "dpId": 8,
            "dpName": "Phase C"
        },
        {
            "dpId": 9,
            "dpName": "Total Power "
        },
        {
            "dpId": 10,
            "dpName": "Fault"
        },
        {
            "dpId": 11,
            "dpName": "Frozen set"
        },
        {
            "dpId": 16,
            "dpName": "Switch"
        },
        {
            "dpId": 17,
            "dpName": "Alarm set"
        },
        {
            "dpId": 18,
            "dpName": "Meter ID"
        },
        {
            "dpId": 20,
            "dpName": "Clear event"
        },
        {
            "dpId": 21,
            "dpName": "Forward Energy T1"
        },
        {
            "dpId": 22,
            "dpName": "Forward Energy T2"
        },
        {
            "dpId": 23,
            "dpName": "Forward Energy T3"
        },
        {
            "dpId": 24,
            "dpName": "Forward Energy T4"
        }
    ],
    "t": 1715425098886,
    "success": true,
    "status": "ok"
}
```

evsmorodin, 2024-11-28:
Hello
@Koenkk 
Can you help with adding support for this device?

Koenkk, 2024-11-29:
@evsmorodin where did you get stuck with creating the external converter? You can use [similar devices](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/tuya.ts) for inspiration.

evsmorodin, 2024-12-15:
Last week I did some tests.

First I decided to search for issues. Maybe there is a solution or a way to find it in neighboring branches

I found this topic: https://github.com/Koenkk/zigbee2mqtt/discussions/20155

Added a converter for my device _TZE200_ywbltqzw by analogy with _TZE200_x8fp01wi, _TZE204_x8fp01wi and got a satisfactory result. The device is displayed in the interface, data is read. This is enough to get readings from the meter

[image]
[image]
[image]
[image]

evsmorodin, 2024-12-15:
I also found this topic: https://github.com/Koenkk/zigbee2mqtt/issues/18750

I used the converter from it and it worked too. It has advanced features compared to those described above

<details><summary>#️⃣ Converter code</summary>

```javascript
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE200_ywbltqzw',
        },
    ],
    model: 'PJ-DAC7361C',
    vendor: 'MatSee Plus',
    description: 'Zigbee DIN 3-phase energy meter',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    whiteLabel: [{vendor: 'XOCA', model: 'DAC7361C'}],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [
        e.switch().setAccess('state', ea.STATE_SET),
        e.power(),
        e.energy(),
        e.produced_energy(),
        tuya.exposes.voltageWithPhase('a'),
        tuya.exposes.voltageWithPhase('b'),
        tuya.exposes.voltageWithPhase('c'),
        tuya.exposes.powerWithPhase('a'),
        tuya.exposes.powerWithPhase('b'),
        tuya.exposes.powerWithPhase('c'),
        tuya.exposes.currentWithPhase('a'),
        tuya.exposes.currentWithPhase('b'),
        tuya.exposes.currentWithPhase('c'),
        e.text('meter_id', ea.STATE).withDescription('Meter ID (ID of device)'),
        e.numeric('daily_energy', ea.STATE).withUnit('kWh').withDescription('Daily sum of consumed energy'),
        e.numeric('monthly_energy', ea.STATE).withUnit('kWh').withDescription('Monthly sum of consumed energy'),
        e.enum('fault', ea.STATE, [
            'clear',
            'ov_cr',
            'unbalance_alarm',
            'ov_vol',
            'undervoltage_alarm',
            'miss_phase_alarm',
            'outage_alarm',
            'magnetism_alarm',
            'terminal_alarm',
            'cover_alarm',
            'credit_alarm',
            'no_balance_alarm',
            'battery_alarm',
            'meter_hardware_alarm'
        ]).withDescription('Fault status of the device (clear = nothing)'),
        e.binary('clear_fault', ea.STATE_SET, 'ON', 'OFF').withDescription('Turn ON to clear last the fault'),
        e.numeric('monthly_energy_rst_day', ea.STATE_SET).withValueMin(1).withValueMax(31).withDescription('Day of the month for monthly consumed energy meter reset'),
        e.numeric('daily_energy_rst_hour', ea.STATE_SET).withValueMin(0).withValueMax(24).withDescription('Hour of the day for daily consumed energy meter reset'),
[... 49 lines cut ...]
module.exports = definition;
```

</details> 

<details><summary>🖼️ Interface screenshots</summary>
<p>
[image]
[image]
[image]
</p>
</details> 

Visually, all data is readable, but there are still several problems

<details><summary>🖼️ Problem screenshots</summary>
<p>
[image]
[image]
</p>
</details> 

Both of these values ​​cannot be set, only read. @PetrozPL on your device, this data can be set via z2m?

There was also a problem with the state switch, but I need a little more time to test it.

PetrozPL, 2024-12-16:
> Frozen set could be the day of the month/hour of the day when daily/monthly values are reset:
> 
> ![image](https://private-user-images.githubusercontent.com/76878782/263499409-b88eb0a7-649d-4d42-b28f-2bb49951cc91.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQzNzkyMTksIm5iZiI6MTczNDM3ODkxOSwicGF0aCI6Ii83Njg3ODc4Mi8yNjM0OTk0MDktYjg4ZWIwYTctNjQ5ZC00ZDQyLWIyOGYtMmJiNDk5NTFjYzkxLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMTYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjE2VDE5NTUxOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTk5ZGYyNzNkNWY0MTFkMzg5ZWFkZDU1MzJjOTkyNzM5NWRmY2I4MzJiZjU0MDllYjIwZGQ4MGRhMWE1NmE1YzImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.bTmVTTNICeEoiCxMdu0c6pAnOE11hHeUFgzxa4r2_yE)
> 
> EDIT : Yes, it seems that I was correct. Daily power consumption was cleared when time reported by device reached selected hour. Tomorrow I'll check if monthly power consumption is cleared when the day of the month reaches selected day, .
> 
> EDIT: Also monthly power consumption was cleared when both - day and hour selected was reached.

I remember that it was working correctly, but for the devices that I was testing initially. I think that I might have not test it for the device that I bought later for myself. I don't have it installed right now, so I can't check it at this moment.

closed 2025-02-23 by github-actions[bot]