Koenkk/zigbee2mqtt #18401: Problem of Thermostat Moes BHT-002-GCLZB

DChe47, 2023-07-22:
### What happened?

I connected the thermostat Moes BHT-002-GCLZB (defined as _TZE200_aoclfnxz) to Home Assistant Supervised used Zegbee2MQTT. At first glance, everything works. But I looked to Log of Home Assistant Core and there are a lot of errors:

`Logger: homeassistant.components.mqtt.models
Source: core.py:1269
Integration: MQTT ([documentation](https://www.home-assistant.io/integrations/mqtt), [issues](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+mqtt%22))
First occurred: 18:16:03 (9318 occurrences)
Last logged: 22:44:15

Exception raised when updating state of sensor.0x003c84fffec01c90_program, topic: 'zigbee2mqtt/Термостат Прихожая' with payload: b'{"child_lock":"UNLOCK","current_heating_setpoint":15,"deadzone_temperature":1,"heat":"OFF","linkquality":32,"local_temperature":22,"local_temperature_calibration":-2,"max_temperature":35,"max_temperature_limit":30,"min_temperature_limit":2,"preset":"hold","preset_mode":"hold","program":{"saturday_p1_hour":6,"saturday_p1_minute":0,"saturday_p1_temperature":24,"saturday_p2_hour":12,"saturday_p2_minute":0,"saturday_p2_temperature":23,"saturday_p3_hour":14,"saturday_p3_minute":30,"saturday_p3_temperature":22,"saturday_p4_hour":17,"saturday_p4_minute":30,"saturday_p4_temperature":21,"sunday_p1_hour":6,"sunday_p1_minute":0,"sunday_p1_temperature":19,"sunday_p2_hour":12,"sunday_p2_minute":30,"sunday_p2_temperature":20,"sunday_p3_hour":14,"sunday_p3_minute":30,"sunday_p3_temperature":21,"sunday_p4_hour":18,"sunday_p4_minute":30,"sunday_p4_temperature":20,"weekdays_p1_hour":6,"weekdays_p1_minute":0,"weekdays_p1_temperature":20,"weekdays_p2_hour":11,"weekdays_p2_minute":30,"weekdays_p2_temperature":21,"weekdays_p3_hour":13,"weekdays_p3_minute":30,"weekdays_p3_temperature":22,"weekdays_p4_hour":17,"weekdays_p4_minute":30,"weekdays_p4_temperature":23},"running_state":"idle","sensor":"AL","system_mode":"heat"}'
Exception raised when updating state of sensor.0x003c84fffec01c90_program, topic: 'zigbee2mqtt/Термостат Прихожая' with payload: b'{"child_lock":"UNLOCK","current_heating_setpoint":15,"deadzone_temperature":1,"heat":"OFF","linkquality":51,"local_temperature":22,"local_temperature_calibration":-2,"max_temperature":35,"max_temperature_limit":30,"min_temperature_limit":2,"preset":"hold","preset_mode":"hold","program":{"saturday_p1_hour":6,"saturday_p1_minute":0,"saturday_p1_temperature":24,"saturday_p2_hour":12,"saturday_p2_minute":0,"saturday_p2_temperature":23,"saturday_p3_hour":14,"saturday_p3_minute":30,"saturday_p3_temperature":22,"saturday_p4_hour":17,"saturday_p4_minute":30,"saturday_p4_temperature":21,"sunday_p1_hour":6,"sunday_p1_minute":0,"sunday_p1_temperature":19,"sunday_p2_hour":12,"sunday_p2_minute":30,"sunday_p2_temperature":20,"sunday_p3_hour":14,"sunday_p3_minute":30,"sunday_p3_temperature":21,"sunday_p4_hour":18,"sunday_p4_minute":30,"sunday_p4_temperature":20,"weekdays_p1_hour":6,"weekdays_p1_minute":0,"weekdays_p1_temperature":20,"weekdays_p2_hour":11,"weekdays_p2_minute":30,"weekdays_p2_temperature":21,"weekdays_p3_hour":13,"weekdays_p3_minute":30,"weekdays_p3_temperature":22,"weekdays_p4_hour":17,"weekdays_p4_minute":30,"weekdays_p4_temperature":23},"running_state":"idle","sensor":"AL","system_mode":"heat"}'
Exception raised when updating state of sensor.0x003c84fffec01c90_program, topic: 'zigbee2mqtt/Термостат Прихожая' with payload: b'{"child_lock":"UNLOCK","current_heating_setpoint":15,"deadzone_temperature":1,"heat":"OFF","linkquality":21,"local_temperature":22,"local_temperature_calibration":-2,"max_temperature":35,"max_temperature_limit":30,"min_temperature_limit":2,"preset":"hold","preset_mode":"hold","program":{"saturday_p1_hour":6,"saturday_p1_minute":0,"saturday_p1_temperature":24,"saturday_p2_hour":12,"saturday_p2_minute":0,"saturday_p2_temperature":23,"saturday_p3_hour":14,"saturday_p3_minute":30,"saturday_p3_temperature":22,"saturday_p4_hour":17,"saturday_p4_minute":30,"saturday_p4_temperature":21,"sunday_p1_hour":6,"sunday_p1_minute":0,"sunday_p1_temperature":19,"sunday_p2_hour":12,"sunday_p2_minute":30,"sunday_p2_temperature":20,"sunday_p3_hour":14,"sunday_p3_minute":30,"sunday_p3_temperature":21,"sunday_p4_hour":18,"sunday_p4_minute":30,"sunday_p4_temperature":20,"weekdays_p1_hour":6,"weekdays_p1_minute":0,"weekdays_p1_temperature":20,"weekdays_p2_hour":11,"weekdays_p2_minute":30,"weekdays_p2_temperature":21,"weekdays_p3_hour":13,"weekdays_p3_minute":30,"weekdays_p3_temperature":22,"weekdays_p4_hour":17,"weekdays_p4_minute":30,"weekdays_p4_temperature":23},"running_state":"idle","sensor":"AL","system_mode":"heat"}'
Exception raised when updating state of sensor.0x003c84fffec01c90_program, topic: 'zigbee2mqtt/Термостат Прихожая' with payload: b'{"child_lock":"UNLOCK","current_heating_setpoint":15,"deadzone_temperature":1,"heat":"OFF","linkquality":18,"local_temperature":22,"local_temperature_calibration":-2,"max_temperature":35,"max_temperature_limit":30,"min_temperature_limit":2,"preset":"hold","preset_mode":"hold","program":{"saturday_p1_hour":6,"saturday_p1_minute":0,"saturday_p1_temperature":24,"saturday_p2_hour":12,"saturday_p2_minute":0,"saturday_p2_temperature":23,"saturday_p3_hour":14,"saturday_p3_minute":30,"saturday_p3_temperature":22,"saturday_p4_hour":17,"saturday_p4_minute":30,"saturday_p4_temperature":21,"sunday_p1_hour":6,"sunday_p1_minute":0,"sunday_p1_temperature":19,"sunday_p2_hour":12,"sunday_p2_minute":30,"sunday_p2_temperature":20,"sunday_p3_hour":14,"sunday_p3_minute":30,"sunday_p3_temperature":21,"sunday_p4_hour":18,"sunday_p4_minute":30,"sunday_p4_temperature":20,"weekdays_p1_hour":6,"weekdays_p1_minute":0,"weekdays_p1_temperature":20,"weekdays_p2_hour":11,"weekdays_p2_minute":30,"weekdays_p2_temperature":21,"weekdays_p3_hour":13,"weekdays_p3_minute":30,"weekdays_p3_temperature":22,"weekdays_p4_hour":17,"weekdays_p4_minute":30,"weekdays_p4_temperature":23},"running_state":"idle","sensor":"AL","system_mode":"heat"}'
Exception raised when updating state of sensor.0x003c84fffec01c90_program, topic: 'zigbee2mqtt/Термостат Прихожая' with payload: b'{"child_lock":"UNLOCK","current_heating_setpoint":15,"deadzone_temperature":1,"heat":"OFF","linkquality":25,"local_temperature":22,"local_temperature_calibration":-2,"max_temperature":35,"max_temperature_limit":30,"min_temperature_limit":2,"preset":"hold","preset_mode":"hold","program":{"saturday_p1_hour":6,"saturday_p1_minute":0,"saturday_p1_temperature":24,"saturday_p2_hour":12,"saturday_p2_minute":0,"saturday_p2_temperature":23,"saturday_p3_hour":14,"saturday_p3_minute":30,"saturday_p3_temperature":22,"saturday_p4_hour":17,"saturday_p4_minute":30,"saturday_p4_temperature":21,"sunday_p1_hour":6,"sunday_p1_minute":0,"sunday_p1_temperature":19,"sunday_p2_hour":12,"sunday_p2_minute":30,"sunday_p2_temperature":20,"sunday_p3_hour":14,"sunday_p3_minute":30,"sunday_p3_temperature":21,"sunday_p4_hour":18,"sunday_p4_minute":30,"sunday_p4_temperature":20,"weekdays_p1_hour":6,"weekdays_p1_minute":0,"weekdays_p1_temperature":20,"weekdays_p2_hour":11,"weekdays_p2_minute":30,"weekdays_p2_temperature":21,"weekdays_p3_hour":13,"weekdays_p3_minute":30,"weekdays_p3_temperature":22,"weekdays_p4_hour":17,"weekdays_p4_minute":30,"weekdays_p4_temperature":23},"running_state":"idle","sensor":"AL","system_mode":"heat"}'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/mqtt/models.py", line 270, in process_write_state_requests
    entity.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 742, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 867, in _async_write_ha_state
    hass.states.async_set(entity_id, state, attr, self.force_update, self._context)
  File "/usr/src/homeassistant/homeassistant/core.py", line 1651, in async_set
    state = State(
            ^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 1269, in __init__
    raise InvalidStateError(
homeassistant.exceptions.InvalidStateError: Invalid state encountered for entity ID: sensor.0x003c84fffec01c90_program. State max length is 255 characters.`

I also send the state of this thermostat:

`{
    "child_lock": "UNLOCK",
    "current_heating_setpoint": 15,
    "deadzone_temperature": 1,
    "heat": "OFF",
    "linkquality": 40,
    "local_temperature": 22,
    "local_temperature_calibration": -2,
    "max_temperature": 35,
[... cut ...]
        "sunday_p4_temperature": 20,
        "weekdays_p1_hour": 6,
[... 5 lines cut ...]
        "weekdays_p3_hour": 13,
        "weekdays_p3_minute": 30,
        "weekdays_p3_temperature": 22,
        "weekdays_p4_hour": 17,
        "weekdays_p4_minute": 30,
        "weekdays_p4_temperature": 23
    },
    "running_state": "idle",
    "sensor": "AL",
    "system_mode": "heat"
}`

How do fix these errors?

### Zigbee2MQTT version

1.32.1

### Adapter firmware version

20220302

### Adapter

zStack3x0

DChe47, 2023-07-24:
I found that this problem was not only with me, but I did not find a solution: [https://github.com/home-assistant/core/issues/92895](url). Help me, please

DChe47, 2023-07-24:
Additional I send diagnostic data:
[mqtt-3f90dc1b7b3d111e89e835812995e654-Термостат Прихожая-6ce1b9ba28ffe4d0f9a594f0401739fe.json.txt](https://github.com/Koenkk/zigbee2mqtt/files/12141830/mqtt-3f90dc1b7b3d111e89e835812995e654-.-6ce1b9ba28ffe4d0f9a594f0401739fe.json.txt)

DChe47, 2023-07-27:
And this warning in Home Assistant Core Log also from this thermostat:
`Logger: homeassistant.helpers.template
Source: helpers/template.py:735
First occurred: 26 июля 2023 г. в 10:16:01 (35 occurrences)
Last logged: 13:22:21

Template variable warning: 'dict object' has no attribute 'current_heating_setpoint' when rendering '{{ value_json.current_heating_setpoint }}'`

csszokolai, 2023-07-30:
Logger: homeassistant.helpers.template
Source: helpers/template.py:735 
First occurred: 11:39:04 (17 occurrences) 
Last logged: 17:07:16

Template variable warning: 'dict object' has no attribute 'current_heating_setpoint' when rendering '{{ value_json.current_heating_setpoint }}'

csszokolai, 2023-08-30:
The issue doesn't solved yet.

DChe47, 2023-09-01:
This problem is not completely solved, but I found a way around it.
In the settings for this device, I have installed:
`    filtered_attributes:
      - program
`
This may be a temporary solution.

closed 2023-10-09 by github-actions[bot]