Koenkk/zigbee2mqtt #18406: set gradient doesn't always work with blue colours - Philips Hue gradient lightstrip 929002994901

jafrouk, 2023-07-23:
### What happened?

I've been testing setting the gradient with the Philips Hue gradient lightstrip 929002994901 via Home Assistant.

It seems that the strip (or perhaps the translation code) does not support certain values of blue.

Setting to red and green work fine, but if you try to set to blue the command does not work.

### What did you expect to happen?

I expected the light to change when i set the new value (and if the Hue Gradient Lightstrip doesn't support the colour I selected that it would "clip" it to a supported colour)

### How to reproduce it (minimal and precise)

First do this, changing device name, (works):
```
service: mqtt.publish
data:
  topic: zigbee2mqtt/Sitting Light Strip 1/set/
  payload: >-
    {"gradient":["#ff0000","#00ff00","#ff0000"]}
```
Then do this, changing device name, (does not work):

service: mqtt.publish
data:
  topic: zigbee2mqtt/Sitting Light Strip 1/set/
  payload: >-
    {"gradient":["#ff0000","#00ff00","#0000ff"]}
```

### Zigbee2MQTT version

1.32.1 

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Home assistant yellow (not sure which adapter you mean or where to find it)

### Debug log

```
Debug 2023-07-23 09:24:15Received MQTT message on 'zigbee2mqtt/Sitting Light Strip 1/set/' with data '{"gradient":["#ff0000","#00ff00","#ff0000"]}'
Debug 2023-07-23 09:24:15Publishing 'set' 'gradient' to 'Sitting Light Strip 1'
Debug 2023-07-23 09:24:16Received Zigbee message from 'Sitting Light Strip 1', type 'attributeReport', cluster 'manuSpecificPhilips2', data '{"48":32,"50":0,"51":0,"52":2,"53":100,"54":136,"state":{"data":[75,1,1,117,1,177,234,78,13,48,0,0,0,14,159,93,180,131,212,14,159,93,24,0],"type":"Buffer"}}' from endpoint 11 with groupID 0
Info 2023-07-23 09:24:16MQTT publish: topic 'zigbee2mqtt/Sitting Light Strip 1', payload '{"brightness":117,"color":{"h":185,"hue":185,"s":8,"saturation":8,"x":0.31299999356269836,"y":0.32899999618530273},"color_mode":"xy","color_temp":154,"elapsed":15051,"gradient":["#ff0500","#00ff0e","#ff0500"],"linkquality":255,"power_on_behavior":"previous","state":"ON","update":{"installed_version":16782340,"latest_version":16782340,"state":"idle"},"update_available":null}'
Debug 2023-07-23 09:24:21Received MQTT message on 'zigbee2mqtt/Sitting Light Strip 1/set/' with data '{"gradient":["#ff0000","#00ff00","#0000ff"]}'
Debug 2023-07-23 09:24:21Publishing 'set' 'gradient' to 'Sitting Light Strip 1'
```

jafrouk, 2023-09-26:
@Koenkk @tader do you think you'll be able to help here?

tader, 2023-09-26:
I can't really help you with this.

Summary of information for you to get started on your quest:

- a single zone seems to bleed into adjacent zones, averaging each others' colors (https://github.com/Koenkk/zigbee2mqtt/issues/10674#issuecomment-1160255437)
- the colors are converted from RGB to XY coordinates (https://github.com/Koenkk/zigbee2mqtt/issues/10674#issuecomment-1284527609)
- construct payload of colors and their distribution over zones (https://github.com/Koenkk/zigbee2mqtt/issues/10674#issuecomment-1284592826)

Most likely it could be because of terms like "gamut" and "color science" that influence the conversion van RGB to XY, there is a publication by Signify bragging about it (see PDF on https://www.signify.com/global/lighting-academy/browser/publication/color-science)

closed 2023-11-04 by github-actions[bot]