Koenkk/zigbee2mqtt #18408: RTX ZVG1 wrong days in normal timer

Talw3g, 2023-07-23:
### What happened?

When sending the command `{"normal_schedule_timer_1": "05:30 / 35.0 / TuThSaSu / 1"}` to the ZVG1, it is acknowledged as `'05:30 / 35 / SuMoWeFr / 1'` : everything is OK except the active days.
The device does go ON on the ACKED days.
Here is the log with the sent command and received status: 
After trying to enable each day one by one, I found out that the days are just in reversed order (except Sunday):
| Day wanted | Day actually active |
|------------|---------------------|
| Monday     | Saturday            |
| Tuesday    | Friday              |
| Wednesday  | Thursday            |
| Thursday   | Wednesday           |
| Friday     | Tuesday             |
| Saturday   | Monday              |
| Sunday     | Sunday              |

### What did you expect to happen?

Active days should be the ones sent in the initial command.

### Zigbee2MQTT version

1.32.1-dev commit: dab7e6a

### Adapter firmware version

20230507

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P

### Debug log

debug 2023-07-23 16:21:13Received MQTT message on 'zigbee2mqtt/prog_arrosage/set' with data '{"normal_schedule_timer_1": "05:30 / 35.0 / TuThSaSu / 1"}'
debug 2023-07-23 16:21:13Publishing 'set' 'normal_schedule_timer_1' to 'prog_arrosage'
info 2023-07-23 16:21:13MQTT publish: topic 'zigbee2mqtt/prog_arrosage/state', payload 'OFF'
info 2023-07-23 16:21:13MQTT publish: topic 'zigbee2mqtt/prog_arrosage/timer_state', payload 'enabled'
info 2023-07-23 16:21:13MQTT publish: topic 'zigbee2mqtt/prog_arrosage/battery', payload '50'
info 2023-07-23 16:21:13MQTT publish: topic 'zigbee2mqtt/prog_arrosage/linkquality', payload '43'
info 2023-07-23 16:21:13MQTT publish: topic 'zigbee2mqtt/prog_arrosage/timer_time_left', payload '1'
info 2023-07-23 16:21:13MQTT publish: topic 'zigbee2mqtt/prog_arrosage/last_valve_open_duration', payload '0'
info 2023-07-23 16:21:13MQTT publish: topic 'zigbee2mqtt/prog_arrosage/water_consumed', payload '0.00'
info 2023-07-23 16:21:13MQTT publish: topic 'zigbee2mqtt/prog_arrosage/timer', payload '60'
info 2023-07-23 16:21:13MQTT publish: topic 'zigbee2mqtt/prog_arrosage/cycle_timer_1', payload ''
info 2023-07-23 16:21:13MQTT publish: topic 'zigbee2mqtt/prog_arrosage/cycle_timer_2', payload ''
info 2023-07-23 16:21:13MQTT publish: topic 'zigbee2mqtt/prog_arrosage/cycle_timer_3', payload ''
info 2023-07-23 16:21:13MQTT publish: topic 'zigbee2mqtt/prog_arrosage/cycle_timer_4', payload ''
info 2023-07-23 16:21:13MQTT publish: topic 'zigbee2mqtt/prog_arrosage/normal_schedule_timer_1', payload '05:30 / 35.0 / TuThSaSu / 1'
info 2023-07-23 16:21:13MQTT publish: topic 'zigbee2mqtt/prog_arrosage/normal_schedule_timer_2', payload ''
info 2023-07-23 16:21:13MQTT publish: topic 'zigbee2mqtt/prog_arrosage/normal_schedule_timer_3', payload ''
info 2023-07-23 16:21:13MQTT publish: topic 'zigbee2mqtt/prog_arrosage/normal_schedule_timer_4', payload ''
debug 2023-07-23 16:21:13Received Zigbee message from 'prog_arrosage', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,1,1,74,0,35,43,64,1,7,230,8,1,1],"type":"Buffer"},"datatype":0,"dp":17}],"seq":33280}' from endpoint 1 with groupID 0
info 2023-07-23 16:21:13MQTT publish: topic 'zigbee2mqtt/prog_arrosage/state', payload 'OFF'
info 2023-07-23 16:21:13MQTT publish: topic 'zigbee2mqtt/prog_arrosage/timer_state', payload 'enabled'
info 2023-07-23 16:21:13MQTT publish: topic 'zigbee2mqtt/prog_arrosage/battery', payload '50'
info 2023-07-23 16:21:13MQTT publish: topic 'zigbee2mqtt/prog_arrosage/linkquality', payload '43'
info 2023-07-23 16:21:13MQTT publish: topic 'zigbee2mqtt/prog_arrosage/timer_time_left', payload '1'
info 2023-07-23 16:21:13MQTT publish: topic 'zigbee2mqtt/prog_arrosage/last_valve_open_duration', payload '0'
info 2023-07-23 16:21:13MQTT publish: topic 'zigbee2mqtt/prog_arrosage/water_consumed', payload '0.00'
info 2023-07-23 16:21:13MQTT publish: topic 'zigbee2mqtt/prog_arrosage/timer', payload '60'
info 2023-07-23 16:21:13MQTT publish: topic 'zigbee2mqtt/prog_arrosage/cycle_timer_1', payload ''
info 2023-07-23 16:21:13MQTT publish: topic 'zigbee2mqtt/prog_arrosage/cycle_timer_2', payload ''
info 2023-07-23 16:21:13MQTT publish: topic 'zigbee2mqtt/prog_arrosage/cycle_timer_3', payload ''
info 2023-07-23 16:21:13MQTT publish: topic 'zigbee2mqtt/prog_arrosage/cycle_timer_4', payload ''
info 2023-07-23 16:21:13MQTT publish: topic 'zigbee2mqtt/prog_arrosage/normal_schedule_timer_1', payload '05:30 / 35 / SuMoWeFr / 1'
info 2023-07-23 16:21:13MQTT publish: topic 'zigbee2mqtt/prog_arrosage/normal_schedule_timer_2', payload ''
info 2023-07-23 16:21:13MQTT publish: topic 'zigbee2mqtt/prog_arrosage/normal_schedule_timer_3', payload ''
info 2023-07-23 16:21:13MQTT publish: topic 'zigbee2mqtt/prog_arrosage/normal_schedule_timer_4', payload ''

Talw3g, 2023-08-16:
I've also noticed that the valve opens for only 10min, whatever the time selected in the schedule timer.
Since it's on the same topic, I'm not sure if it deserves a new issue...

closed 2023-09-23 by github-actions[bot]

ianmtaylor1, 2024-03-17:
Bumping a stale issue here, but I'm experiencing this exact same thing with this exact same device.

Zigbee2MQTT version: 1.36.0, running in official docker container.

ianmtaylor1, 2024-03-17:
The day of week scrambling also occurs in the cycle timers, FYI.

ianmtaylor1, 2024-03-22:
I just wanted to follow up here for the OP and anyone who finds this on google.

Based on my tests, this day of week issue appears to be fixed in the dev branch as of 1.36.0-dev, commit 73423eb. Another issue with the cycle timers where they would be set, disappear from the ui/current device state, but still be active, also seems to have been resolved.

I don't see any obvious commits that would have fixed this so the fix may have been in zigbee-herdsman or zigbee-herdsman-converters.