Koenkk/zigbee2mqtt #18409: No converter manuSpecificBosch TS0002_TZ3000_zmy4lslw

dnothing, 2023-07-23:
### What happened?

Hi,

In my logs i found out, that one of TS0002 is displaying info regarding missing converter. Is there any sollution for this?

Debug 2023-07-23 17:35:34Received Zigbee message from '0xa4c13867684a7b06', type 'raw', cluster 'manuSpecificBosch', data '{"data":[8,31,10,2,208,72,2,0,10],"type":"Buffer"}' from endpoint 1 with groupID 0
Debug 2023-07-23 17:35:34No converter available for 'TS0002_switch_module' with cluster 'manuSpecificBosch' and type 'raw' and data '{"data":[8,31,10,2,208,72,2,0,10],"type":"Buffer"}'
Debug 2023-07-23 17:35:34Received Zigbee message from '0xa4c13867684a7b06', type 'raw', cluster 'manuSpecificBosch', data '{"data":[8,31,10,2,208,72,2,0,10],"type":"Buffer"}' from endpoint 1 with groupID 0
Debug 2023-07-23 17:35:34No converter available for 'TS0002_switch_module' with cluster 'manuSpecificBosch' and type 'raw' and data '{"data":[8,31,10,2,208,72,2,0,10],"type":"Buffer"}'
Debug 2023-07-23 17:35:35Received Zigbee message from '0xa4c13867684a7b06', type 'raw', cluster 'manuSpecificBosch', data '{"data":[8,31,10,2,208,72,2,0,10],"type":"Buffer"}' from endpoint 1 with groupID 0
Debug 2023-07-23 17:35:35No converter available for 'TS0002_switch_module' with cluster 'manuSpecificBosch' and type 'raw' and data '{"data":[8,31,10,2,208,72,2,0,10],"type":"Buffer"}'
Debug 2023-07-23 17:35:35Received Zigbee message from '0xa4c13867684a7b06', type 'raw', cluster 'manuSpecificBosch', data '{"data":[8,31,10,2,208,72,2,0,10],"type":"Buffer"}' from endpoint 1 with groupID 0
Debug 2023-07-23 17:35:35No converter available for 'TS0002_switch_module' with cluster 'manuSpecificBosch' and type 'raw' and data '{"data":[8,31,10,2,208,72,2,0,10],"type":"Buffer"}'
Debug 2023-07-23 17:35:35Received Zigbee message from '0xa4c13867684a7b06', type 'raw', cluster 'manuSpecificBosch', data '{"data":[8,31,10,2,208,72,2,0,10],"type":"Buffer"}' from endpoint 1 with groupID 0
Debug 2023-07-23 17:35:35No converter available for 'TS0002_switch_module' with cluster 'manuSpecificBosch' and type 'raw' and data '{"data":[8,31,10,2,208,72,2,0,10],"type":"Buffer"}'
Debug 2023-07-23 17:35:35Received Zigbee message from '0xa4c13867684a7b06', type 'raw', cluster 'manuSpecificBosch', data '{"data":[8,31,10,2,208,72,2,0,10],"type":"Buffer"}' from endpoint 1 with groupID 0
Debug 2023-07-23 17:35:35No converter available for 'TS0002_switch_module' with cluster 'manuSpecificBosch' and type 'raw' and data '{"data":[8,31,10,2,208,72,2,0,10],"type":"Buffer"}'
Debug 2023-07-23 17:35:35Received Zigbee message from '0xa4c13867684a7b06', type 'raw', cluster 'manuSpecificBosch', data '{"data":[8,31,10,2,208,72,2,0,10],"type":"Buffer"}' from endpoint 1 with groupID 0

### Zigbee2MQTT version

1.32.1-dev commit: 57d031d5

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Sonoff ZBDongle Plus-e

closed 2023-08-30 by github-actions[bot]

tfarras, 2024-01-23:
I have the same issue with `Sonoff ZBDongle Plus-e` and `TS0002` switch modules from `_TZ3000_zmy4lslw`.
After some investigation I noticed that the issue is not constant. Usually when I change `channel` and completely remove all the `backups` and `database.db` from `data` directory - the devices that have that issue are changing too. Also their data in `database.db` for such devices is a bit strange (different from same devices that are working fine)

I also noticed that devices which have such issues - do not publish their state immediately. If you'll change their state physically - z2m will know about that change only after 1-2 seconds. Also you can experience issues with their update from within z2m, sometimes actions are just failed due to timeout.

I tried flashing the dongle and tried downgrading the z2m version. Tried docker version and standalone node process, everything behaves the same, some of the devices cannot work properly.

The issue is not related to the LQI or the distance between dongle and the device, I experienced such issues even when device was in 30cm from the dongle and LQI was 240.

For now I gave up trying to fix the issue and ordered ZBDongle-P, that is `recomended` and not `experimental`.
But if somebody knows how to fix the issue - would love to discuss it.

Koenkk, 2024-03-12:
Is the device not working or is it just the spamming? Not sure if we can fix the spamming part.

chojnicki, 2024-03-12:
@Koenkk not working while this is happening. My idea about spamming was that since Z2M cannot recognize this message, maybe device is waiting for some response. 

BUT while just removing/repaitong them did not helped, today I disconnected them from power. After reconnecting, device send like 20 same messages again and then went silent. And it is working normal again. With second one the same procedure same good result.

So in summary - hard reset works (at least for me) but not sure why. This is alarming for me, because it can happen again randomly again without my knowledge, killing my network and SD card. Even after diagnozing, it's pain to fix it this way because module is hidden inside wall behind switch :P Not sure if Z2M can help here, but I hope so. How to decode what this message means? It never shows up when device is working.

Koenkk, 2024-03-13:
>  How to decode what this message means?

That's very hard, it just a random set of bytes, only the manufacuter knows the meaning of it.

bgornicki, 2024-06-10:
I experienced the same issue (random delays in command execution, spamming errors with debug log level) and I decided to dive into it. I can contribute to solve it but I need some help to determine the best way to do it.

I found some pretty good Tuya Developer documentation pages with details regarding custom clusters and DataPoints for the products mentioned above. Documentation for TS0002 can be found here: [Multi-Gang Switch](https://developer.tuya.com/en/docs/connect-subdevices-to-gateways/tuya-zigbee-multiple-switch-access-standard?id=K9ik6zvnqr09m)

Supported Clusters:
[image]

The first issue is that the current config incorrectly recognizes `Tuya private cluster 0 (0xE000)` as `manuSpecificBosch` (with the same id). The second one `Tuya private cluster 1 (0xE001)` is handled correctly as [`manuSpecificTuya_3`](https://github.com/Koenkk/zigbee-herdsman/blob/master/src/zspec/zcl/definition/cluster.ts#L4581) and used to determine power on behavior for the switch (it's working fine).

I would like to add support for `Tuya private cluster 0 (0xE000)` because it exposes a couple of interesting functionalities:

[image]

Everything works correctly when using the official Tuya Gateway so it would be great to support it in z2m. There's also an explanation of the DataPoints.

I was thinking about adding the cluster definition to `clusters.js` in `zigbee-herdsman` package but I encountered a different issue. It is that `Tuya private cluster 0 (0xE000)` cluster does not seem to be consistent between different type of products. There is also documentation for [Socket with Energy Monitoring](https://developer.tuya.com/en/docs/connect-subdevices-to-gateways/tuya-zigbee-measuring-smart-plug-access-standard?id=K9ik6zvofpzqk) which also describes `0xE000` cluster but it has a different name (`TUYA_COMMON_PRIVATE_CLUSTER`), different attribute IDs (different order...) and different data types.

So my question is: what would be the best way to add support for it? Is it possible to create custom cluster definitions for specific products? I found [this discussion)(https://github.com/Koenkk/zigbee2mqtt/discussions/16786) regarding custom clusters and it seems that the answer is no? Or maybe it would be best to just add two different clusters and specify which one is used in which device? What do you think @Koenkk?

Koenkk, 2024-06-11:
@bgornicki since recent z2m supports custom clusters, [example](https://github.com/Koenkk/zigbee-herdsman-converters/blob/e204850a4ed41eb5eb85d65f5ae63b81ab61ed60/src/devices/sonoff.ts#L43)

sakcaj, 2024-06-25:
Got the same issue as @bgornicki for the same device type, same errors in the logs with debug enabled, the same symptoms i.e. the switch is not reliable. LMK if there's anything I can do to test if there's a fix.