Koenkk/zigbee2mqtt #18411: Livarno Lux LED Strip 22W 2m - not working in latest version

szporwolik, 2023-07-23:
### What happened?

I've been using Livarno Lux LED Strip 22W 2m (_TZ3000_riwp3k79) with a luck since the support was implemented. Today I've updated zigbee2mqtt to latest docker revision and it seems that device is not recognized anymore.

### What did you expect to happen?

Device was supported and behaved stable in the last 1.5-2 years.

### How to reproduce it (minimal and precise)

Pair _TZ3000_riwp3k79 Lightstrip.

### Zigbee2MQTT version

1.32.1 commit: 4ccfdfc

### Adapter firmware version

0x26580700

### Adapter

 ConBee2/RaspBee2

### Debug log

[image]

Koenkk, 2023-07-24:
Can you try to:
- force remove device 
- restart z2m
- re-pair device.

szporwolik, 2023-07-24:
Yes, that helped. After re-paring, device was properly recognized and is working correctly.
[image]

Great thanks!

Is there something I can provide (logs?) that I can help to fix this long term ?

Koenkk, 2023-07-24:
I don't know how it happend (it ofc shouldn't) and haven't seen other reports about this. So let's treat it as an incident for now.

szporwolik, 2023-07-24:
Copy, thank you again! :)

closed 2023-07-24 by Koenkk