Koenkk/zigbee2mqtt #18412: NodOn SIN-4-RS-20 state note updated

jiquem, 2023-07-23:
### What happened?

The status (OPEN, CLOSE, STOP) of my roller shutter module NodOn SIN-4-RS-20 is not updated very regularly in Home Assistant. From time to time it's updated, and from time to time it isn't.
I have to go into zigbee2mqtt to perform a manual refresh. 

### What did you expect to happen?

State to be updated like for others devices

### Zigbee2MQTT version

1.32.1 

### Adapter firmware version

20221226

### Adapter

zStack3x0

Wireheadbe, 2023-07-27:
Have the same issue. Manually changing reporting settings fixes this for a while, or powercycling them.
Possible firmware issue @AlexisPolegato ?
(There's a newer firmware posted to the OTA repo a couple weeks back, but trying to install it, gives an error ```Update of 'SaJefRolluik' failed (No image available for imageType '265')```)

jiquem, 2023-07-28:
Thanks @Wireheadbe 
Which reporting setting do you change ?

jiquem, 2023-08-02:
I've got this message after updating z2m to 1.32.2

`Info 2023-08-02 14:01:12MQTT publish: topic 'zigbee2mqtt/cover_cuisine', payload '{"last_seen":1690977522426,"linkquality":72,"position":100,"state":"OPEN","tilt":0,"update":{"installed_version":523,"latest_version":541,"state":"available"},"update_available":null}'
Info 2023-08-02 14:01:12MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"cover_cuisine"},"error":"Update of 'cover_cuisine' failed (Timeout: device did not request any image blocks)","status":"error","transaction":"n3pzs-1"}'
Error 2023-08-02 14:01:12Update of 'cover_cuisine' failed (Timeout: device did not request any image blocks`

jiquem, 2023-09-03:
Still the same behavior with the 1.33

trollix, 2023-10-04:
Idem for me, update failed again in 1.33-1
[image]

Koenkk, 2023-10-09:
Some improvements have been pushed regarding the OTA update mechanism, please check if this issue is fixed now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

jiquem, 2023-10-09:
Hello, I've just tested on HA with '''1.33.1-dev commit: [9cff19b](https://github.com/Koenkk/zigbee2mqtt/commit/9cff19b)''' and got the same error

Wireheadbe, 2023-11-12:
> Have the same issue. Manually changing reporting settings fixes this for a while, or powercycling them. Possible firmware issue @AlexisPolegato ? (There's a newer firmware posted to the OTA repo a [...]

No updates from NodOn. States not updated a lot of times (covers open, but reported closed, unless forcing a position refresh, which i know do as work-around every 5 minutes...). Sometimes, in the middle of the night, they go up a bit a down again. No idea why. Instant wake-up..

Firmware updates don't work, even though someone from the company actively posts firmware to the repo. 
Can't recommend these modules. Had hoped this would be quality brand...

AlexisPolegato, 2023-11-14:
Hello @Wireheadbe ,

This should be fixed in the latest version of the device firmware.

Unfortunately, there's an issue in the mechanism of firmware update in Z2M, it can work if you try several times but it's quite random.
Alternatively, you could try to update your products through the NodOn Connect Application (available on your Android/iOS).

Wireheadbe, 2023-11-14:
Hi @AlexisPolegato  - thanks for coming back.

However, on my Moto G100 running Android 12, the app crashes when trying to upgrade. On my wife's A52s, the app isn't available since NodOn Connect isn't compatible with Android 13.  Thus, no dice to be able to upgrade.

Why would the firmware upgrade fail - because I've updated numerous devices from e.g. Philips, Ikea,... Would you mind sharing tips/tricks? I've noticed in the app, that the modules disappear after a while. So i did disconnect a modules from the mains before trying to upgrade: app still crashes on my phone, and the upgrade doesn't work via Z2M as well. Do I have faulty modules?

aurel32, 2023-11-14:
> Alternatively, you could try to update your products through the NodOn Connect Application (available on your Android/iOS).

Is there some documentation available about that application? I have tried to get it working with my SIN-4-RS-20 devices, and the application does not find any devices. I have tried to powercycle them, but it doesn't work either.

AlexisPolegato, 2023-11-14:
> Hi @AlexisPolegato - thanks for coming back.
> 
> However, on my Moto G100 running Android 12, the app crashes when trying to upgrade. On my wife's A52s, the app isn't available since NodOn Connect isn't compatible with Android 13. Thus, no dice to be able to upgrade.
> 
> Why would the firmware upgrade fail - because I've updated numerous devices from e.g. Philips, Ikea,... Would you mind sharing tips/tricks? I've noticed in the app, that the modules disappear after a while. So i did reconnect a module to the mains before trying to upgrade: app still crashes on my phone, and the upgrade doesn't work via Z2M as well. Do I have faulty modules?

There are 2 Firmwares in the Device. The Device asks an update on a certain Firmware and the Gateway should answer with this Firmware.
But Zigbee2Mqtt does not currently handle that, so when the device asks a certain Firmware, the Gateway does not answer with the right Firmware and the Update does not happen...

AlexisPolegato, 2023-11-14:
> Is there some documentation available about that application? I have tried to get it working with my SIN-4-RS-20 devices, and the application does not find any devices. I have tried to powercycle [...]

Here but it's in French: https://nodon.fr/nodon/application-nodon-connect/

When you power cycle the device, it does not appear in the App ?

Wireheadbe, 2023-11-17:
> There are 2 Firmwares in the Device. The Device asks an update on a certain Firmware and the Gateway should answer with this Firmware. But Zigbee2Mqtt does not currently handle that, so when the [...]

Hi @AlexisPolegato - thanks for the info. Any way we can work together with @Koenkk if you provide some more technical details to this mechanism you have implemented?

(would appreciate a fix on Android 12 and a build for Android 13 as well 😉 )

AlexisPolegato, 2023-11-21:
@Wireheadbe  What do you mean exaclty by "the state wasn't updated automatically" ?
You changed the state locally (from the Switch ?) or from Z2M ? And you didn't get the right status back ?

If you check the logs, you don't see any reports ?
Something like that:

`Info 2023-11-21 16:03:12MQTT publish: topic 'zigbee2mqtt/SIN-4-RS-20', payload '{"linkquality":76,"position":0,"state":"OPEN","tilt":100,"update":{"installed_version":543,"latest_version":543,"state":"idle"},"update_available":null}'`

dgseten, 2023-11-27:
Hi @AlexisPolegato , I have installed few of [ADEO SIN-4-RS-20_LEX](https://www.zigbee2mqtt.io/devices/SIN-4-RS-20_LEX.html) and I love them, one of them is working fine but in the rest of then I am seeing the issue explained in this page. I guess that the issue is the same.

I think the product is the same than this [NodOn SIN-4-RS-20](https://www.zigbee2mqtt.io/devices/SIN-4-RS-20.html) and Nodon is the real manufacturer of this product, but OTA updates are not supporthed neither.

I just check with the Nodon Connect app, but there is not also options to update the version of this product. Could you help us?

[image]

aurel32, 2023-12-03:
> Here but it's in French: https://nodon.fr/nodon/application-nodon-connect/
>
> When you power cycle the device, it does not appear in the App ?

No it didn't work. But I have found the issue. I was trying from an Android 13 device. When I tried from an old phone running IIRC Android 11, the device got discovered, and I was able to upgrade the firmware. The upgrade went fine, the only issue I have noticed is that the up and down switch inputs got swapped in the new firmware, so I had to rewire it.

Wireheadbe, 2024-01-30:
Worked fine for a while on the newest firmware, but sadly, at some point, they stop reporting the status of the shutter (e.g. trigger shutter close, they do the actuation, but they do not report back the position (in this case, closed)) - need to manually refresh.

Balooforever, 2024-02-07:
> Worked fine for a while on the newest firmware, but sadly, at some point, they stop reporting the status of the shutter (e.g. trigger shutter close, they do the actuation, but they do not report back the position (in this case, closed)) - need to manually refresh.

Same for me. I don’t know if the update state is on z2m or on module.. With my Old module in a very old firmware, all work fine ..

cedricbarbecue, 2024-02-08:
@Wireheadbe same behaviour but I use the z2m report apply option on z2m interface on my nodon module and everything was then corrected.

Wireheadbe, 2024-02-10:
> @Wireheadbe same behaviour but I use the z2m report apply option on z2m interface on my nodon module and everything was then corrected.

That works for a while, but then at some point it stops again..I've even factory-reset them, on the latest firmware, and re-paired them. Nothing helps. I have heaps of zigbee devices, and these are the only ones I have headaches with. Even at night, sometimes, it'll just open a shutter for a bit, and close it again. Like it's doing some form of calibration... But then wakes you! I've reverted to a node-red flow, which force-refreshes the status of Nodon devices every 5 minutes.

dguihal, 2024-02-19:
OK got a "fix" (probably more a workaround) for this

1. Get to your device in z2m interface
2. Get to "Reporting" Tab
3. Remove everything where attribute is related to "TiltPercentage" (I sometime got 2 here ...). Goal is to keep only one "currentPositionLiftPercentage"

And that's it, my device magically started to report correctly again

I still don't get why this "Tilt" feature has been added to this device as there is no physical tilt connectors in it. You may want to control some tilt with the regular lift output but you certainly can't have both and this looks like the source of all of this trouble

Wireheadbe, 2024-02-19:
This only works for a while. At some point, the device will again stop reporting, unfortunately.

AlexisPolegato, 2024-02-19:
@Wireheadbe How long would you say before reporting stops ?

cweber26, 2024-02-22:
Hello, I think the issue is not exclusively linked to the NodOn module.
Because I was had the same issue with one of my Proxalus cover motor which have a zigbee 3 embedded
It solve the issue by clicking on the button Apply button of the Reporting tab of the cover on Z2M.
And as I said in a previous comment I also had the same issue and workaround with the live power consommation info on a connected plug.

cweber26, 2024-02-22:
@Wireheadbe can you share us the command to do the refresh by an automation ? Thanks

jiquem, 2024-02-24:
Hello
I have 3 covers. I put in place an ugly automation to update the reporting parameters every hour.
This workaround prevent me to face reporting issues.

```
alias: "[ADMIN] Reporting cover"
description: ""
trigger:
  - platform: time_pattern
    hours: "*"
    minutes: "0"
    seconds: "0"
condition: []
action:
  - service: mqtt.publish
    data:
      payload: >-
        {"attribute":"currentPositionTiltPercentage","cluster":"closuresWindowCovering","id":"cover_salon/1","maximum_report_interval":60,"minimum_report_interval":1,"reportable_change":1}
      topic: zigbee2mqtt/bridge/request/device/configure_reporting
  - service: mqtt.publish
    data:
      payload: >-
        {"attribute":"currentPositionLiftPercentage","cluster":"closuresWindowCovering","id":"cover_salon/1","maximum_report_interval":60,"minimum_report_interval":1,"reportable_change":1}
      topic: zigbee2mqtt/bridge/request/device/configure_reporting
  - service: mqtt.publish
    data:
      payload: >-
        {"attribute":"currentPositionTiltPercentage","cluster":"closuresWindowCovering","id":"cover_sejour/1","maximum_report_interval":60,"minimum_report_interval":1,"reportable_change":1}
      topic: zigbee2mqtt/bridge/request/device/configure_reporting
  - service: mqtt.publish
    data:
      payload: >-
        {"attribute":"currentPositionLiftPercentage","cluster":"closuresWindowCovering","id":"cover_sejour/1","maximum_report_interval":60,"minimum_report_interval":1,"reportable_change":1}
      topic: zigbee2mqtt/bridge/request/device/configure_reporting
  - service: mqtt.publish
    data:
      payload: >-
        {"attribute":"currentPositionTiltPercentage","cluster":"closuresWindowCovering","id":"cover_cuisine/1","maximum_report_interval":60,"minimum_report_interval":1,"reportable_change":1}
      topic: zigbee2mqtt/bridge/request/device/configure_reporting
  - service: mqtt.publish
    data:
      payload: >-
        {"attribute":"currentPositionLiftPercentage","cluster":"closuresWindowCovering","id":"cover_cuisine/1","maximum_report_interval":60,"minimum_report_interval":1,"reportable_change":1}
      topic: zigbee2mqtt/bridge/request/device/configure_reporting
mode: single
```

Wireheadbe, 2024-03-01:
> @Wireheadbe can you share us the command to do the refresh by an automation ? Thanks

Hi, I'm using Node-Red for this.. I basically make sure to save the MQTT output of zigbee2mqtt/bridge/devices in a global var (eek! global vars! - I know 😆 )
```
let devices = msg.payload;
global.set('devices',devices);
```

And then via a timer, every 5 minutes, I retrieve what's stored in the global var:
```
msg.payload = global.get('devices')
return msg;
```
I then pass that on to a "split" node, which splits using "\n"

And then I send that output to another function node:
```
if ( msg.payload.definition != null) {
    let model = msg.payload.definition.model;
    let name = msg.payload.friendly_name;

    if (model === "SIN-4-RS-20") {
        msg.topic = "zigbee2mqtt/" + name + '/get' ;
        msg.payload = {
            "position": ""
        }
        return msg;
    }
}
```
Which then sends that back to MQTT - by searching for every device occurence of "SIN-4-RS-20" and sending a payload to force-get the position.

lieblinger, 2024-03-07:
@Wireheadbe Could you share your Node-RED flow? I'm facing the same issue and it's really annoying. 

Thanks in advance!

cweber26, 2024-04-16:
> Hello I have 3 covers. I put in place an ugly automation to update the reporting parameters every hour. This workaround prevent me to face reporting issues.
>
> ```
> alias: "[ADMIN] Reporting cover" [...]

Thanks that exactly what I needed.
I've just create the automation and run it, I see on Z2M that the reporting configuration is ok.
I think that a pretty good workaround.

EderlMarkus, 2024-06-01:
> Hello I have 3 covers. I put in place an ugly automation to update the reporting parameters every hour. This workaround prevent me to face reporting issues.
>
> ```
> alias: "[ADMIN] Reporting cover" [...]

I also have the same problem, I've got 10 SIN-4-RS-20 and I've tried the workaroud, although I'm trying mine in Node-Red:
[flows.json](https://github.com/user-attachments/files/15522518/flows.json)

But still, theres always 1 or 2 that simply stop reporting their position from one day to the next. Still a sustainable solution would make my day :D

Is there anything I could take a look at in the Source Code or are there any logs i can provide that could help?

AlexisPolegato, 2024-11-12:
It has been fixed on latest Firmware of the Device (V3.5.0).
=> https://github.com/Koenkk/zigbee-OTA/pull/599

calou79, 2024-11-19:
> It has been fixed on latest Firmware of the Device (V3.5.0). => [Koenkk/zigbee-OTA#599](https://github.com/Koenkk/zigbee-OTA/pull/599)

Can you explain to me the procedure to follow to upgrade the nodon images to 3.5.0 ? or updating zigbee2qtt is enough? 
because I still can't update them in zigbee2mqtt 
in French if you are French otherwise English 
thank you in advance

dguihal, 2024-11-19:
Just trigger a ota update search on z2m interface, it will find it

EderlMarkus, 2024-11-21:
> Can you explain to me the procedure to follow to upgrade the nodon images to 3.5.0 ? or updating zigbee2qtt is enough? because I still can't update them in zigbee2mqtt in French if you are French [...]

I have Zigbee2Mqtt Version 1.40.1 
[image]
In this Version you should be seeing the OTA Tab in the Frontend:
[image]
There you should be able to upgrade:
[image]

calou79, 2024-11-21:
ok thank you very much, the error must come from my sonoff e because indeed I can launch the updates but they fail each time with these errors :(I will look at this again thanks)
z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Volet Salon Grand TV/mode', payload '{"cover":null}'
error 2024-11-21 17:38:25zh:ember:uart:ash: Received ERROR from adapter, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT.
error 2024-11-21 17:38:25zh:ember:uart:ash: ASH disconnected | Adapter status: ASH_NCP_FATAL_ERROR
error 2024-11-21 17:38:25zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
error 2024-11-21 17:38:25zh:ember: Adapter fatal error: HOST_FATAL_ERROR
info 2024-11-21 17:38:25zh:ember:uart:ash: ASH COUNTERS since last clear:
info 2024-11-21 17:38:25zh:ember:uart:ash: Total frames: RX=3730, TX=4868
info 2024-11-21 17:38:25zh:ember:uart:ash: Cancelled : RX=0, TX=0
info 2024-11-21 17:38:25zh:ember:uart:ash: DATA frames : RX=3724, TX=1136
info 2024-11-21 17:38:25zh:ember:uart:ash: DATA bytes : RX=84980, TX=67605
info 2024-11-21 17:38:25zh:ember:uart:ash: Retry frames: RX=5, TX=3
info 2024-11-21 17:38:25zh:ember:uart:ash: ACK frames : RX=0, TX=3729
info 2024-11-21 17:38:25zh:ember:uart:ash: NAK frames : RX=0, TX=0
info 2024-11-21 17:38:25zh:ember:uart:ash: nRdy frames : RX=0, TX=0
info 2024-11-21 17:38:25zh:ember:uart:ash: CRC errors : RX=0
info 2024-11-21 17:38:25zh:ember:uart:ash: Comm errors : RX=0
info 2024-11-21 17:38:25zh:ember:uart:ash: Length < minimum: RX=0
info 2024-11-21 17:38:25zh:ember:uart:ash: Length > maximum: RX=0
info 2024-11-21 17:38:25zh:ember:uart:ash: Bad controls : RX=0
info 2024-11-21 17:38:25zh:ember:uart:ash: Bad lengths : RX=0
info 2024-11-21 17:38:25zh:ember:uart:ash: Bad ACK numbers : RX=0
info 2024-11-21 17:38:25zh:ember:uart:ash: Out of buffers : RX=0
info 2024-11-21 17:38:25zh:ember:uart:ash: Retry dupes : RX=5
info 2024-11-21 17:38:25zh:ember:uart:ash: Out of sequence : RX=0
info 2024-11-21 17:38:25zh:ember:uart:ash: ACK timeouts : RX=3
error 2024-11-21 17:38:33zh:ember:ezsp: ERROR Transaction failure; status=ASH_ERROR_TIMEOUTS. Last Frame: [FRAME: ID=52:"SEND_UNICAST" Seq=221 Len=7].
error 2024-11-21 17:38:33zh:ember:ezsp: ERROR Transaction failure; status=ASH_NCP_FATAL_ERROR. Last Frame: [FRAME: ID=52:"SEND_UNICAST" Seq=221 Len=7].
info 2024-11-21 17:38:57zh:ember:uart:ash: Port closed.
info 2024-11-21 17:38:57zh:ember:uart:ash: ======== ASH stopped ========
info 2024-11-21 17:38:57zh:ember:ezsp: ======== EZSP stopped ========
info 2024-11-21 17:38:57zh:ember: ======== Ember Adapter Stopped ========
error 2024-11-21 17:38:57z2m: Adapter disconnected, stopping

EderlMarkus, 2024-12-03:
So far the Shutters are all sending correct Data and the States of the individual Shutters are updated correctly. It seems the Upgrade of the Firmware has fixed the issue. I'm cautiously optimistic.

Wireheadbe, 2024-12-03:
Using 3.5.0-1.1.2 - the states keep being updated. No more need for hacks via NodeRed etc..

closed 2025-02-09 by github-actions[bot]