Koenkk/zigbee2mqtt #18413: [Feature request]: ZM85EL-2Z (_TZE200_cf1sl3tj) please add battery sensor DP 13

VikeDragon, 2023-07-24:
### Is your feature request related to a problem? Please describe

Greetings!
ZM85EL-2Z (_TZE200_cf1sl3tj). Please add battery sensor DP 13
Previously, there was a converter option with a battery charge, now after the changes hit the release, they disappeared after the last update.
https://github.com/Koenkk/zigbee2mqtt/issues/11251#issuecomment-1285207519

### Describe the solution you'd like

Previously, there was a converter option with a battery charge, now after the changes hit the release, they disappeared after the last update. Moreover, the battery charge is displayed on the first tab of the device, but not as a separate sensor.
https://github.com/Koenkk/zigbee2mqtt/issues/11251#issuecomment-1285207519

### Describe alternatives you've considered

ZM85EL-2Z (_TZE200_cf1sl3tj). Please add battery sensor DP 13

### Additional context

https://github.com/Koenkk/zigbee2mqtt/issues/11251#issuecomment-1285207519
Moreover, the battery charge is displayed on the first tab of the device, but not as a separate sensor.
[image]
[image]
[image]

nimmerpeople, 2023-09-17:
Please...

VikeDragon, 2023-09-21:
Greetings, Koenkk!

Thank you for the converter! I see the charging icon appeared in the list, but this prop is not transmitted to MQTT. I tried to reconnect the device, but there is no battery props in MQTT.

Koenkk, 2023-09-21:
can you make a screenshot of the device -> state tab?

cloudbr34k84, 2023-09-22:
@Koenkk 
[image]

VikeDragon, 2023-09-23:
> can you make a screenshot of the device -> state tab?

[image]

Koenkk, 2023-09-23:
I made a mistake in the ext converter, with https://gist.github.com/Koenkk/0ed3bbdc0f6246e2b6d73b47aa649a15 battery should show up under exposes, it may take some time before the battery value is reported (24 hours max I expect)

cloudbr34k84, 2023-09-23:
> I made a mistake in the ext converter, with https://gist.github.com/Koenkk/0ed3bbdc0f6246e2b6d73b47aa649a15 battery should show up under exposes, it may take some time before the battery value is reported (24 hours max I expect)

When will this be pushed through to edge? I prefer to wait till I don't need it as a converter

Koenkk, 2023-09-23:
> When will this be pushed through to edge? I prefer to wait till I don't need it as a converter

When someone confirms the ext converter works.

VikeDragon, 2023-09-24:
> I made a mistake in the ext converter, with https://gist.github.com/Koenkk/0ed3bbdc0f6246e2b6d73b47aa649a15 battery should show up under exposes, it may take some time before the battery value is reported (24 hours max I expect)

Greetings, Koenkk!
I try the modified converter, I wait 24 hours until the battery appears.

VikeDragon, 2023-09-25:
Greetings, Koenkk!
The battery charge has not appeared with the new version of the converter. Here are screenshots. Repaired one of the drives, I'll try what will happen tomorrow.
[image]
[image]
[image]

VikeDragon, 2023-09-27:
Greetings, Koenkk!
The problem remains. The new converter does not show charge and does not create charge entities in MQTT.

Koenkk, 2023-09-27:
what was the latest z2m version where this still worked?

VikeDragon, 2023-09-28:
> what was the latest z2m version where this still worked?

Greetings, Koenkk!
I removed both engines again, cleaned the MQTT from the props. I have overloaded the Home Assistant several times.  I checked the converter that it is connected correctly and works. I connected the engines and everything was created correctly! Everything is wonderful! Batteries shows, new entities have been created! Thank you so much for the converter!!!
[image]
[image]
[image]
[image]

VikeDragon, 2023-09-28:
Thank you for your support and help!

closed 2023-09-28 by VikeDragon

Koenkk, 2023-10-03:
Integrated the fix, you can remove the ext converter after updating to todays release.

VikeDragon, 2023-10-05:
> Integrated the fix, you can remove the ext converter after updating to todays release.

Thank you! Everything works great after the update, without the converter!