Koenkk/zigbee2mqtt #18414: [Feature request]: New property NodOn SIN-4-RS-20, SIN-4-RS-20_LEX and SIN-4-RS-20_PRO

rty3108, 2023-07-24:
### Is your feature request related to a problem? Please describe

Please, can you add the Calibration property for the NodOn devices SIN-4-RS-20, SIN-4-RS-20_LEX and SIN-4-RS-20_PRO?
These devices support the option to calibrate through zigbee commands.

The attribute to be used for module calibration is 'Mode' is code 0x0017 from 'Window Covering Settings Attribute Set' (7.4.2.1.3).
The bit for its activation is the 'bit1' table (7.4.2.1.3.8)

### Describe the solution you'd like

New property for devices SIN-4-RS-20, SIN-4-RS-20_LEX and SIN-4-RS-20_PRO

### Describe alternatives you've considered

I don't know any other way to get this information

### Additional context

.

rty3108, 2023-08-24:
Please can implement this features?
Thancks

Sraoune, 2023-08-25:
Yeah, would be nice to have

Sraoune, 2023-09-25:
Still needed :)

closed 2023-11-04 by github-actions[bot]

eherranzr, 2024-01-20:
Hi guys,
I just bought the device and position only shows 0% or 100%. Is the same for you?
I would say calibrating would solve this issue.
If we can support somehow, please let us know

josejn, 2024-03-15:
> Hi guys, I just bought the device and position only shows 0% or 100%. Is the same for you? I would say calibrating would solve this issue. If I can support somehow, please let me know

Hi, I have bought some of them and they work like a charm (lex / leroy merlin version)
You have to calibrate them by pushing 5 times the phisical switch at the module. The blind will go full up, then full down and them up again.

The only issue I have is at the first pairing. I have to reboot mosquito brocker in order to get they publishing the state and follow the commands properly.

Having this feature would be great since they are ver inconvinient to disassemble because they barely fit under the wall buttons.

Regards.

amanfromfrance, 2024-06-04:
I also would like this feature to be implemented as the auto-calibration does not work with my Bubendorff motors.

lippertmarkus, 2024-06-11:
did one of you were able to start the manual calibration?

amanfromfrance, 2024-06-11:
If you mean sending commands using this panel:
[image]

Then no it never worked.

lippertmarkus, 2024-06-11:
I just were able to manually set the calibration bit:
[image]

but setting this just starts the automatic calibration (just like pressing the button on the module 5 times) and afterwards disables the automatic calibration again (bit1 = 0)

amanfromfrance, 2024-06-17:
So you are sending something like 0x0010?

lippertmarkus, 2024-06-17:
yes writing the attribute, but as a decimal (2)

amanfromfrance, 2024-06-20:
I was able to get the complete list of attributes from the module using **zha_toolkit.scan_device** command:

```yaml
zha_toolkit_version: v1.1.10
zigpy_version: 0.64.1
zigpy_rf_version: 0.39.1
ieee_org:
  - 165
  - 93
  - 25
  - 254
  - 255
  - 177
  - 179
  - 244
ieee: f4:b3:b1:ff:fe:19:5d:a5
command: scan_device
command_data: null
start_time: "2024-06-20T16:18:26.653801+00:00"
errors: []
params:
  dir: 0
  tries: 1
  expect_reply: true
  args: []
  read_before_write: true
  read_after_write: true
scan:
  ieee: f4:b3:b1:ff:fe:19:5d:a5
  nwk: "0x5447"
  model: SIN-4-RS-20
  manufacturer: NodOn
  manufacturer_id: "0x4747"
  endpoints:
    - id: 1
      device_type: "0x0202"
      profile: "0x0104"
      in_clusters:
        "0x0000":
          cluster_id: "0x0000"
          title: Basic
          name: basic
          attributes:
            "0x0000":
              attribute_id: "0x0000"
              attribute_name: zcl_version
              value_type:
                - "0x20"
                - uint8_t
                - Analog
              access: READ|REPORT
              access_acl: 5
              attribute_value: 8
            "0x0001":
              attribute_id: "0x0001"
              attribute_name: app_version
              value_type:
                - "0x20"
                - uint8_t
                - Analog
              access: READ|REPORT
              access_acl: 5
              attribute_value: 3
            "0x0002":
              attribute_id: "0x0002"
              attribute_name: stack_version
              value_type:
                - "0x20"
                - uint8_t
                - Analog
[... 674 lines cut ...]

- **Calibration Vertical Run Time - Up**
   - **_Attribute_**: 0x0000
   - **_Type_**: uint16
   - **_Manufacturer Code_**: 0x1288
   - **_Description_**: Value of the current Up Vertical Run Time in ten of ms . By default, the value is set at OxFFFF, which means the Calibration has not been done. Once the Calibration has been done, the value changed to the Run Time calculated during the Calibration process.
- **Calibration Vertical Run Time - Down**
   - **_Attribute_**: 0x0001
   - **_Type_**: uint16
   - **_Manufacturer Code_**: 0x128B
   - **_Description_**: Value of the current Down Vertical Run Time in ten of ms By default, the value is set at OxFFFF, which means the Calibration has not been done. Once the Calibration has been done, the value changed to the Run Time calculated during the Calibration process.
- **Calibration Rotation Run Time - Up**
   - **_Attribute_**: 0x0002
   - **_Type_**: uint16
   - **_Manufacturer Code_**: 0x1288
   - **_Description_**: Value of the current Up Rotation Run Time in ten of ms By default, the value is set at 0x00C8 (= 200 => 2000 ms). The Rotation Run Time is not calculated by the Calibration process and must be set by the user using this attribute.
- **Calibration Rotation Run Time - Down**
   - **_Attribute_**: 0x0003
   - **_Type_**: uint16
   - **_Manufacturer Code_**: 0x1288
   - **_Description_**: Value of the current Down Rotation Run Time in ten of ms By default, the value is set at 0x00C8 (200 => 2000 ms). The Rotation Run Time is not calculated by the Calibration process and must be set by the user using this attribute.

But as you can see these Manufacturers attribues do not appear in the YAML... so I'm confused.

From what the customer support told me, it works perfectly using Z2M but to be honest I am reluctant to migrate...

lippertmarkus, 2024-06-20:
Those manufacturer specific attributes have been implemented in Z2M here: https://github.com/AlexisPolegato/zigbee-herdsman-converters/blob/b0bf537d9509e8be158e1297d8749429cc5ff863/src/devices/nodon.ts#L21-L76

but changing them has no effect for me. I also asked customer support about using those attributes this but they weren't able to help.

amanfromfrance, 2024-06-20:
Sure, I'll give it a try.

But maybe with a cheap alternate Zigbee key as I spent many hours getting my Tuya garage opener module working and tweaking it... I suppose I would have to start from 0 once more with Z2M otherwise.

By any chance, would you have a recommendation for such a cheap alternate Zigbee key? (I am using SkyConnect at the moment).

lippertmarkus, 2024-06-20:
I'm super happy with this one: https://www.ebay.de/itm/384530667411

amanfromfrance, 2024-06-24:
So! I've received and installed the Sonoff Zigbee dongle, installed and configured Z2M and the NodOn module is now integrated through Z2M on my HA installation.

I can now see the manufacturer specific attributes and by default they are not set:
[image]

But changing them using the sliders or the dedicated field has absolutely no effect... while the NodOn customer support ensured me that it works for them.

Maybe that is due to my Bubendorff 4 wire shutters?? It's really a shame that, I feel like there is no solution.

amanfromfrance, 2024-06-24:
Sure, I'll try to call them again this week. Because for them what I was experiencing was related to the fact that I was using ZHA instead of Z2M... which has been proven wrong.

One thing I really don't understand is that as soon as the module is detected and integrated (wether it is using ZHA or Z2M), it automatically calibrates itself:
* Makes the shutter go up for 20cm,
* Makes it go down for 20cm,
* Makes it go up for 20cm,
* Makes it go down for 20cm,
* Finally makes it go up for 20cm.

Which according to NodOn customer support should not happen.

The installed FW on my module is **3.0.0-1.1.2**.

And when it comes to the **NodOn Connect** app, it tells me that my module is not up to date but the firmware update stalls at 0%... and I am unable to remove the module from the list.

lippertmarkus, 2024-06-24:
Thank you very much! 

For me it's also automatically calibrated when paired with Z2M. I have firmware 3.0.0-1.3.0.

AlexisPolegato, 2024-06-24:
Hi @amanfromfrance ,

> Sure, I'll try to call them again this week. Because for them what I was experiencing was related to the fact that I was using ZHA instead of Z2M... which has been proven wrong.
>
> One thing I really [...]

Well, this should actually happen, but it is not the Calibration, it's the Identification Process (when the Device is paired in a Zigbee Network, it identifies itself).
Once you've added the Device in the Network, you can calibrate it by doing a 5x press on the button.
Altenatively, you can launch the Calibration from your Z2M interface.

> The installed FW on my module is **3.0.0-1.1.2**.
> 
> And when it comes to the **NodOn Connect** app, it tells me that my module is not up to date but the firmware update stalls at 0%... and I am unable to remove the module from the list.

I will test the Manuel Calibration with this version in few days to check if there's any issue, I'll get back to you on this point.

lippertmarkus, 2024-06-24:
With a hint from @AlexisPolegato  I just updated to firmware 3.3.0-1.3.0 via Z2M and since then the configuration attributes seem to work

amanfromfrance, 2024-06-24:
That's great! How were you able to update to this firmware? Using the NodOn app?

lippertmarkus, 2024-06-24:
In Z2M under "OTA" you can "check for new updates" for the NodOn modules and update them afterwards

amanfromfrance, 2024-06-24:
Actually I tried it numerous times this weekend and it never worked... 1h ago, it detected the new firmware version and I was able to launch the update 👍

lippertmarkus, 2024-06-24:
I just updated my second module to 3.3.0-1.3.0 as well and tried setting the calibration attributes but for that module it didn't work :(

lippertmarkus, 2024-06-25:
Did it work for you after the firmware update?

amanfromfrance, 2024-06-25:
I'll only be able to test tomorrow when I'll be back home, but for now what I can say is that the firmware update I was able to start yesterday (I could see a progress indication) seemingly has gone wrong because the version displayed is still the same:

[image]

And now the OTA update tells me the module is up to date 🤔

[image]

lippertmarkus, 2024-06-26:
@amanfromfrance you mentioned in an earlier comment that you had firmware 3.0.0-1.1.2 before so at least the first version part seems to have increased after the firmware update

amanfromfrance, 2024-06-27:
I've just tested and the calibration attributes still have no effect unfortunately :(

lippertmarkus, 2024-06-27:
I did further tests and found out that even after setting the manual calibration attributes the NodOn modules still seem to watch the current to detect start end end positions. 

Setting the manual calibration attributes did only work for roller shutters for which the automatic calibration also works (roller shutters which are detected correctly because of the current rising/falling edges).

What I tried:
1. Add a module to a "supported" roller shutter
1. Run automated calibration ==> roller shutter correctly drives to closed or open positions
1. Set manual calibration attributes ==> automated calibration is correctly overriden by the set runtime values and roller shutter drives the correct amount of time that I manually set
1. Disconnect the Nodon module from the supported and connect it to the "unsupported" roller shutter (for me those are two roller shutters paired and driven together) without reset/reconfigure
1. I would expect that the manual runtimes I set before in step 3 are still applying for the "unsupported" roller shutter I switched to but this is not the case. Setting a position now doesn't move the roller shutter at all, using "OPEN" and "CLOSE" from Z2M only moves the roller shutter for 1s and then stops (instead of the manual runtime I set in 3)

I would expect that as soon as I set the manual calibration attributes the module no longer watches the current and only drives the exact amount of seconds I set it to. But it seems that even with the manually set runtime the module still watches the current (which "give the wrong signal" when e.g. two roller shutters are paired)

@AlexisPolegato  Are you able to change that behaviour or make it configurable? I really like the modules but this behaviour really limits the use cases 🙁

amanfromfrance, 2024-06-28:
I've also noticed something annoying when the module is installed and I want to open or close the shutter using the physical switch: I have to press the up/down button 2 times for the shutter to go up/down:
* 1st press: nothing happens
* Press on the middle/stop button to be able to go up/down a 2nd time
* 2nd press: the shutter goes up/down

I suppose that's not a normal behavior...

I have these switches installed:
https://www.leroymerlin.fr/produits/electricite-et-domotique/interrupteur-et-prise/interrupteur-et-prise-schneider-electric/schneider-electric-odace/interrupteur-schneider-electric-odace/interrupteur-volet-roulant-a-assembler-odace-schneider-electric-blanc-68749695.html

Sraoune, 2024-06-28:
> I've also noticed something annoying when the module is installed and I want to open or close the shutter using the physical switch (10cm by 10cm...): I have to press the up/down button 2 times for the shutter to go up/down:
> 
>     * 1st press: nothing happens
> 
>     * Press on the middle/stop button to be able to go up/down a 2nd time
> 
>     * 2nd press: the shutter goes up/down
> 
> 
> I suppose that's not a normal behavior...
> 
> I have these switches installed: https://www.leroymerlin.fr/produits/electricite-et-domotique/interrupteur-et-prise/interrupteur-et-prise-schneider-electric/schneider-electric-odace/interrupteur-schneider-electric-odace/interrupteur-volet-roulant-a-assembler-odace-schneider-electric-blanc-68749695.html

I experience the exact same behaviour, in the same context (Sin-4-RS-20 + Schneider Odace). I beleive this is related to the switch type (bistable switch with a reset button). If i do a power reset of the NodOn, i get back to what is the standard comportment : pressing up or down launch the shutter going up/down, and the middle button stop the motion.

amanfromfrance, 2024-08-17:
It's been a long while now, I suppose we'll have to do with what we have :(

AlexisPolegato, 2024-08-23:
> It's been a long while now, I suppose we'll have to do with what we have :(

You are talking about the behavior with the Physical Switch ?
There's a learning mechanism at the first use of the the Switch by the device. If it doesn't work properly, it might come from here.

So you can try to power off and on your device and then make a Simple Press on the Switch.

amanfromfrance, 2024-08-24:
Hello @AlexisPolegato,

I am talking more generally of the fact that the NodOn module isn't working at all in my case, .e.g. with my Bubendorff shutters:
* Shutter is only going up or down for 20cm at a time regardless of all the tests I've made. The module is always identifying itself as you explained but after that the calibration never works, whether it is by pressing the button 5 times or setting the value in Z2M.
* No position is available but tfrom my discussions with NodOn support hat was expected with my Bubendorff shutters.

And yes on top of that there is the Physical Switch problem.

Last week I tried to use 2 **MOES Zigbee + RF curtain switch modules** (MS-108ZR) and it worked at first try... bonus is that with these modules I get to set the shutters position using a percentage.

I find it sad that a 40/45€ module is unable to accomplish what a 12€ module is able to do.