Koenkk/zigbee2mqtt #18419: [New device support]: Tuya TS0601 TZE204_81yrt3lo

dams74, 2023-07-24:
### Link

https://fr.aliexpress.com/item/1005005782430275.html

### Database entry

{     "ieee": "a4:c1:38:b5:cd:41:32:70",     "nwk": 7491,     "status": 2,     "lqi": "255",     "rssi": "-55",     "last_seen": "1690192304.159175",     "node_descriptor": "01:40:8e:41:11:42:42:00:00:2a:42:00:00",     "endpoints": [         {             "id": 1,             "status": 1,             "device_type": 81,             "profile_id": 260,             "manufacturer": "_TZE204_81yrt3lo",             "model": "TS0601",             "output_clusters": [                 {                     "id": 25,                     "name": "Ota",                     "attributes": []                 },                 {                     "id": 10,                     "name": "Time",                     "attributes": []                 }             ],             "input_clusters": [                 {                     "id": 4,                     "name": "Groups",                     "attributes": []                 },                 {                     "id": 5,                     "name": "Scenes",                     "attributes": []                 },                 {                     "id": 61184,                     "name": "Cluster",                     "attributes": []                 },                 {                     "id": 0,                     "name": "Basic",                     "attributes": [                         {                             "id": 0,                             "name": "zcl_version",                             "value": 3                         },                         {                             "id": 1,                             "name": "app_version",                             "value": 74                         },                         {                             "id": 2,                             "name": "stack_version",                             "value": 0                         },                         {                             "id": 3,                             "name": "hw_version",                             "value": 1                         },                         {                             "id": 4,                             "name": "manufacturer",                             "value": "_TZE204_81yrt3lo"                         },                         {                             "id": 5,                             "name": "model",                             "value": "TS0601"                         },                         {                             "id": 6,                             "name": "date_code",                             "value": ""                         },                         {                             "id": 7,                             "name": "power_source",                             "value": 1                         }                     ]                 }             ]         },         {             "id": 242,             "status": 1,             "device_type": 97,             "profile_id": 41440,             "manufacturer": null,             "model": null,             "output_clusters": [                 {                     "id": 33,                     "name": "GreenPowerProxy",                     "attributes": []                 }             ],             "input_clusters": []         }     ],     "signature": {         "manufacturer": "_TZE204_81yrt3lo",         "model": "TS0601",         "node_desc": {             "logical_type": 1,             "complex_descriptor_available": 0,             "user_descriptor_available": 0,             "reserved": 0,             "aps_flags": 0,             "frequency_band": 8,             "mac_capability_flags": 142,             "manufacturer_code": 4417,             "maximum_buffer_size": 66,             "maximum_incoming_transfer_size": 66,             "server_mask": 10752,             "maximum_outgoing_transfer_size": 66,             "descriptor_capability_field": 0         },         "endpoints": {             "1": {                 "profile_id": 260,                 "device_type": 81,                 "input_clusters": [                     4,                     5,                     61184,                     0                 ],                 "output_clusters": [                     25,                     10                 ]             },             "242": {                 "profile_id": 41440,                 "device_type": 97,                 "input_clusters": [],                 "output_clusters": [                     33                 ]             }         }     },     "class": "zigpy.device" }

### Comments

Can you help to create a converter?
I do not have the product, it'f for an other user
Thanks

dronisuav, 2023-07-30:
I get  this information from manufacturer:

1 | 115 | DPID_POWER_TOTAL_ID | only report | 4 | 1. report the total power (A+B)2. big-endian, 0.1W, X103. signed (32bits)
-- | -- | -- | -- | -- | --
2 | 101 | DPID_POWER_ID_A | only report | 4 | 1. report the total power (A)2. big-endian, 0.1W, X103. unsigned int (32bits)
3 | 105 | DPID_POWER_ID_B | only report | 4 | 1. report the total power (B)2.big-endian, 0.1W, X103. unsigned int (32bits)
4 | 102 | DPID_POWER_DIRECTION_ID_A | only report | 1 | 0: Forward, 1:Reverse
5 | 104 | DPID_POWER_DIRECTION_ID_B | only report | 1 | 0: Forward, 1:Reverse
6 | 106 | DPID_FORWARD_ENERGY_TOTAL_A | only report | 4 | 1. report the forward energy (A)2. big-endian,  X100,0.01KWH3. unsigned int (32bits)
7 | 107 | DPID_REVERSE_ENERGY_TOTAL_A | only report | 4 | 1. report the reverse energy  (A)2. big-endian,  X100, 0.01KWH3. unsigned int (32bits)
8 | 108 | DPID_FORWARD_ENERGY_TOTAL_B | only report | 4 | 1. report the forward energy (B)2. big-endian,  X100,0.01KWH3. unsigned int (32bits)
9 | 109 | DPID_REVERSE_ENERGY_TOTAL_B | only report | 4 | 1. report the reverse energy  (B)2. big-endian,  X100, 0.01KWH3. unsigned int (32bits)
10 | 110 | DPID_POWER_FACTOR_A | only report | 4 | 1. report the power factor  (A)2. big-endian,  X1003. unsigned int (32bits)
11 | 121 | DPID_POWER_FACTOR_B | only report | 4 | 1. report the power factor  (B)2. big-endian,  X1003. unsigned int (32bits)
12 | 111 | DPID_POWER_FREQ | only report | 4 | 1. report the AC freq 2. big-endian,  X1003. unsigned int (32bits)
13 | 112 | DPID_VOLTAGE_A | only report | 4 | 1. report the Voltage 2. big-endian,  X1003. unsigned int (32bits)
14 | 113 | DPID_CURRENT_A | only report | 4 | 1. report the Current(A) 2. big-endian,  X1003. unsigned int (32bits)
15 | 114 | DPID_CURRENT_B | only report | 4 | 1. report the Current(B) 2. big-endian,  X1003. unsigned int (32bits)
16 | 129 | DPID_UPDATE_RATE | report/setting | 4 | 1. report the update rate2. big-endian,  (3-60s)3. unsigned int (32bits)
17 | 116 | DPID_VOLTAGE_A_COEF | report/setting | 4 | 1. calibration Voltage2. big-endian,  X10003. unsigned int (32bits)
18 | 117 | DPID_CURRENT_A_COEF | report/setting | 4 | 1. Calibration current_A2. big-endian,  X10003. unsigned int (32bits)
19 | 118 | DPID_POWER_A_COEF | report/setting | 4 | 1. Calibration power_A2. big-endian,  X10003. unsigned int (32bits)
20 | 119 | DPID_ENERGY_A_COEF | report/setting | 4 | 1. Calibration forward energy_A2. big-endian,  X10003. unsigned int (32bits)
21 | 127 | DPID_ENERGY_A_COEF_REV | report/setting | 4 | 1. Calibration reverse energy_A2. big-endian,  X10003. unsigned int (32bits)
22 | 122 | DPID_FREQ_COEF | report/setting | 4 | 1. Calibration AC freq2. big-endian,  X10003. unsigned int (32bits)
23 | 123 | DPID_CURRENT_B_COEF | report/setting | 4 | 1. Calibration Current_B2. big-endian,  X10003. unsigned int (32bits)
24 | 124 | DPID_POWER_B_COEF | report/setting | 4 | 1. Calibration Power_B2. big-endian,  X10003. unsigned int (32bits)
25 | 125 | DPID_ENERGY_B_COEF | report/setting | 4 | 1. Calibration forward energy_B2. big-endian,  X10003. unsigned int (32bits)
26 | 128 | DPID_ENERGY_B_COEF_REV | report/setting | 4 | 1. Calibration Revese energy_B2. big-endian,  X10003. unsigned int (32bits)
  |   |   |   |   |

samartandek, 2023-07-30:
Hi guys, i bought the same unit couple days ago. Some guy wrote that external converter from  [TS0601_bidirectional_energy meter](https://www.zigbee2mqtt.io/devices/TS0601_bidirectional_energy_meter.html) work for him after replace _TZE200_lsanae15 to our  TZE204_81yrt3lo 

So i try it too, device was recognized but almost every exposes values return null.

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
//    fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE204_81yrt3lo'}],
    fingerprint: tuya.fingerprint('TS0601', ['_TZE204_81yrt3lo', '_TZE204_81yrt3lo']),
    model: 'TS0601 bidirectional clamp energy meter',
    vendor: 'TuYa',
    description: 'Tuya Smart ZigBee Energy Meter Bidirectional Dual Phases 80A ',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    configure: tuya.configureMagicPacket,
        exposes: [tuya.exposes.switch(), e.energy(), e.power(), e.voltage(), e.current(),
            e.enum('fault', ea.STATE, ['clear', 'over_current_threshold', 'over_power_threshold',
                'over_voltage threshold', 'wrong_frequency_threshold']).withDescription('Fault status of the device (clear = nothing)'),
            e.enum('threshold_1', ea.STATE, ['not_set', 'over_current_threshold', 'over_voltage_threshold'])
                .withDescription('State of threshold_1'),
            e.binary('threshold_1_protection', ea.STATE, 'ON', 'OFF')
                .withDescription('OFF - alarm only, ON - relay will be off when threshold reached'),
            e.numeric('threshold_1_value', ea.STATE)
                .withDescription('Can be in Volt or Ampere depending on threshold setting. Setup the value on the device'),
            e.enum('threshold_2', ea.STATE, ['not_set', 'over_current_threshold', 'over_voltage_threshold'])
                .withDescription('State of threshold_2'),
            e.binary('threshold_2_protection', ea.STATE, 'ON', 'OFF')
                .withDescription('OFF - alarm only, ON - relay will be off when threshold reached'),
            e.numeric('threshold_2_value', ea.STATE)
                .withDescription('Setup value on the device'),
            e.binary('clear_fault', ea.STATE_SET, 'ON', 'OFF')
                .withDescription('Turn ON to clear last the fault'),
            e.text('meter_id', ea.STATE).withDescription('Meter ID (ID of device)'),
        ],
        meta: {
            tuyaDatapoints: [
                [1, 'energy', tuya.valueConverter.divideBy100],
                [3, null, null], // Monthly, but sends data only after request
                [4, null, null], // Dayly, but sends data only after request
                [6, null, tuya.valueConverter.phaseVariant2], // voltage and current
                [10, 'fault', tuya.valueConverterBasic.lookup({'clear': 0, 'over_current_threshold': 1,
                    'over_power_threshold': 2, 'over_voltage_threshold': 4, 'wrong_frequency_threshold': 8})],
                [11, null, null], // Frozen - strange function, in native app - nothing is clear
                [16, 'state', tuya.valueConverter.onOff],
                [17, null, tuya.valueConverter.threshold], // It's settable, but can't write converter
                [18, 'meter_id', tuya.valueConverter.raw],
                [20, 'clear_fault', tuya.valueConverter.onOff], // Clear fault
                [21, null, null], // Forward Energy T1 - don't know what this
                [22, null, null], // Forward Energy T2 - don't know what this
                [23, null, null], // Forward Energy T3 - don't know what this
                [24, null, null], // Forward Energy T4 - don't know what this
            ],
        },
};

module.exports = definition;`

after that i found external converter for
[TS0601_3_phase_clamp_meter.js](https://gist.github.com/Koenkk/0046f4c0d453bcc1f8d25b0077694c1d)
which was made by guru Koenkk

[... 59 lines cut ...]
        tuyaDatapoints: [
            [111, 'ac_frequency', tuya.valueConverter.raw],
            [106, 'energy', tuya.valueConverter.divideBy100],
            [101, 'energy_a', valueConverter.divideBy1000],
            [111, 'energy_b', valueConverter.divideBy1000],
            [131, 'current', valueConverter.divideBy1000],
            [115, 'power', tuya.valueConverter.raw],
            [110, 'power_factor_a', tuya.valueConverter.raw],
            [121, 'power_factor_b', tuya.valueConverter.raw],
            [6, null, valueConverter.phaseVariant3('a')],
            [7, null, valueConverter.phaseVariant3('b')],
            [134, 'device_status', tuya.valueConverter.raw],
        ],
    },
};

module.exports = definition;`

but with my second external converter version zigbee2mqtt  doesn't even want to get up.

Look like both versions are so different.
I think that problem is with energy meter devices which we bought.

Maybe somebody have knowledge about that how to wrote proper external converter.
I don't have tuya zigbee router. I use sonoff usb dongle plus with Home Assistant.

samartandek, 2023-08-02:
Hi guys and Sophie Zhang :-),
i have the same unit and i made external converter for it.
It's first version but all data from report only DPid all covered.
From report/setting pids are only Voltage calibration  supported now by default in HA=>Zigbee2mqtt addon=>setting (specific).

I don't know how to config report/setting pids in external converter.
I don't have tuya zigbee gateway, i switched debug on in zigbee2mqtt to test and verify is it working properly.
If you know how to test it more and add support to report/setting pids give me sign.

I got only one bad info in zigbee2mqtt log: "No converter available for 'PJ-1203A' with cluster 'manuSpecificTuya' and type 'commandMcuSyncTime' and data '{"payloadSize":XXX}'" but it's known subject by koenkk.

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;
const {Buffer} = require('buffer');

const valueConverter = {
	divideBy10: tuya.valueConverterBasic.divideBy(10),
	divideBy100: tuya.valueConverterBasic.divideBy(100),
};

const definition = {
    fingerprint: [
        {
            modelID: 'TS0601',
            manufacturerName: '_TZE204_81yrt3lo',
        },
    ],
    model: 'TS0601_Bidirectional_Energy_Meter_with_80A_Current_Clamp',
    vendor: 'TuYa',
    description: 'Bidirectional Energy Meter with 80A Current Clamp',   
    fromZigbee: [tuya.fz.datapoints], //, fz.ignore_tuya_set_time not working for No converter available for friendly_name with cluster manuSpecificTuya and type commandMcuSyncTime and data {payloadSize:XXX}
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
	whiteLabel: [ tuya.whitelabel('Tuya', 'PJ-1203A', 'Bidirectional Energy Meter with 80A Current Clamp', ['_TZE204_81yrt3lo']), ],
//    whiteLabel: [{vendor: 'MatSeePlus', model: 'PJ-1203A'}],	
    exposes: [ //only report data
	    e.ac_frequency(),
        exposes.numeric('total_power_A', ea.STATE).withUnit('W').withDescription('Total power A'),
        exposes.numeric('total_power_B', ea.STATE).withUnit('W').withDescription('Total power B'),
        exposes.numeric('total_power_AB', ea.STATE).withUnit('W').withDescription('Total power A'),
        exposes.numeric('voltage', ea.STATE).withUnit('V').withDescription('Voltage'),		
        exposes.numeric('current_A', ea.STATE).withUnit('A').withDescription('Current A'),
        exposes.numeric('current_B', ea.STATE).withUnit('A').withDescription('Current B'),
        exposes.numeric('power_factor_A', ea.STATE).withUnit('%').withDescription('Instantaneous measured power factor A'),
        exposes.numeric('power_factor_B', ea.STATE).withUnit('%').withDescription('Instantaneous measured power factor B'),
        exposes.numeric('power_direction A', ea.STATE).withDescription('Power direction A 0/1 for forward/reverse'),
        exposes.numeric('power_direction B', ea.STATE).withDescription('Power direction B 0/1 for forward/reverse'),
        exposes.numeric('energy_forward_A', ea.STATE).withUnit('kWh').withDescription('Total energy A forward'),
        exposes.numeric('energy_forward_B', ea.STATE).withUnit('kWh').withDescription('Total energy B forward'),       
        exposes.numeric('energy_reverse_A', ea.STATE).withUnit('kWh').withDescription('Total energy A reverse'),
        exposes.numeric('energy_reverse_B', ea.STATE).withUnit('kWh').withDescription('Total energy B reverse'), 
        exposes.numeric('update_frequency', ea.STATE).withUnit('sec').withDescription('Update frequency'), 		
    ],
    
    meta: {
        tuyaDatapoints: [//only report data
		    [111, 'ac_frequency', tuya.valueConverter.divideBy100],
            [101, 'total_power_A', tuya.valueConverter.divideBy10],
            [105, 'total_power_B', tuya.valueConverter.divideBy10],
            [115, 'total_power_AB', tuya.valueConverter.divideBy10],
			[112, 'voltage', tuya.valueConverter.divideBy10],
			[113, 'current_A', tuya.valueConverter.divideBy100],
			[114, 'current_B', tuya.valueConverter.divideBy100],
			[110, 'power_factor_A', tuya.valueConverter.divideBy100],
			[121, 'power_factor_B', tuya.valueConverter.divideBy100],
            [102, 'power_direction A', tuya.valueConverter.raw],
            [104, 'power_direction B', tuya.valueConverter.raw],
            [106, 'energy_forward_A', tuya.valueConverter.divideBy100],
            [108, 'energy_forward_B', tuya.valueConverter.divideBy100],
            [107, 'energy_reverse_A', tuya.valueConverter.divideBy100],
            [109, 'energy_reverse_B', tuya.valueConverter.divideBy100],
			[129, 'update_frequency' , tuya.valueConverter.raw],
        ],
    },
};

module.exports = definition;

screenshoots:
[image]

[image]

[image]

FlowFrank, 2023-08-02:
> Hi guys and Sophie Zhang :-), i have the same unit and i made external converter for it. It's first version but all data from report only DPid all covered. From report/setting pids are only Voltage [...]

Hey there. I am new to Zigbee2MQTT and I would like to know, where I have to enter to code you send above. I have the same device and switched from ZHA to Zigbee2MQTT because of it.

FlowFrank, 2023-08-02:
> Hey @samartandek. I am new to Zigbee2MQTT and I would like to know, where I have to enter the code you send above. I have the same device and switched from ZHA to Zigbee2MQTT because of it.

Found it!

But there is a error message... :-( Have no idea what's wrong.

```
can not read a block mapping entry; a multiline key may not be an implicit key (55:14)

 52 | const ea = exposes.access;
 53 | const {Buffer} = require('buffer');
 54 | const valueConverter = {
 55 |     dvideBy10: tuya.valueConverterBasic.divi ...
-------------------^
 56 |     divideBy100: tuya.valueConverterBasic.di ...
```

samartandek, 2023-08-02:
Copy and past all code which i post to new file called
Tuya_TZE204_81yrt3lo.js
After that add 2 lines like that below to configuration.yaml where you has
zigbe2mqtt installation directory.
external_converters:
  - Tuya_TZE204_81yrt3lo.js

You have error in line 55 because you ate one sign i
must be divide not dvide

best regards
smartandek

SoulOnFire, 2023-08-03:
> Hi guys and Sophie Zhang :-), i have the same unit and i made external converter for it. It's first version but all data from report only DPid all covered. From report/setting pids are only Voltage [...]

I have successfully implemented this as you... but I'm always getting 0 values and total powers... did you get real values?

FlowFrank, 2023-08-03:
It's working for me too. However, the measured values cannot be correct.

samartandek, 2023-08-03:
I checked returned values with different zigbee energy meter socket with
25w bulb. Different clamp dirrection on the same power line can return you
Total power 0.
Check values with only one clamp Connected to single power L wire.

SoulOnFire, 2023-08-03:
> I checked returned values with different zigbee energy meter socket with 25w bulb. Different clamp dirrection on the same power line can return you Total power 0. Check values with only one clamp [...]

I Just find out the problem! I was using a 4w led lamp and that was too low to measure. Just tried it with a 40w and everything works as expected.

dams74, 2023-08-04:
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee'); const tz = require('zigbee-herdsman-converters/converters/toZigbee'); const exposes = [...]

Hello
Thanks to do this
It's the first time i do this. I'm new user of Zigbee2mqtt.
If i understand:

1) Make a new file called Tuya_TZE204_81yrt3lo.js with this inside:

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;
const {Buffer} = require('buffer');

const valueConverter = {
divideBy10: tuya.valueConverterBasic.divideBy(10),
divideBy100: tuya.valueConverterBasic.divideBy(100),
};

const definition = {
fingerprint: [
{
modelID: 'TS0601',
manufacturerName: '_TZE204_81yrt3lo',
},
],
model: 'TS0601_Bidirectional_Energy_Meter_with_80A_Current_Clamp',
vendor: 'TuYa',
description: 'Bidirectional Energy Meter with 80A Current Clamp',
fromZigbee: [tuya.fz.datapoints], //, fz.ignore_tuya_set_time not working for No converter available for friendly_name with cluster manuSpecificTuya and type commandMcuSyncTime and data {payloadSize:XXX}
toZigbee: [tuya.tz.datapoints],
onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
configure: tuya.configureMagicPacket,
whiteLabel: [ tuya.whitelabel('Tuya', 'PJ-1203A', 'Bidirectional Energy Meter with 80A Current Clamp', ['_TZE204_81yrt3lo']), ],
// whiteLabel: [{vendor: 'MatSeePlus', model: 'PJ-1203A'}],
exposes: [ //only report data
e.ac_frequency(),
exposes.numeric('total_power_A', ea.STATE).withUnit('W').withDescription('Total power A'),
exposes.numeric('total_power_B', ea.STATE).withUnit('W').withDescription('Total power B'),
exposes.numeric('total_power_AB', ea.STATE).withUnit('W').withDescription('Total power A'),
exposes.numeric('voltage', ea.STATE).withUnit('V').withDescription('Voltage'),
exposes.numeric('current_A', ea.STATE).withUnit('A').withDescription('Current A'),
exposes.numeric('current_B', ea.STATE).withUnit('A').withDescription('Current B'),
exposes.numeric('power_factor_A', ea.STATE).withUnit('%').withDescription('Instantaneous measured power factor A'),
exposes.numeric('power_factor_B', ea.STATE).withUnit('%').withDescription('Instantaneous measured power factor B'),
exposes.numeric('power_direction A', ea.STATE).withDescription('Power direction A 0/1 for forward/reverse'),
exposes.numeric('power_direction B', ea.STATE).withDescription('Power direction B 0/1 for forward/reverse'),
exposes.numeric('energy_forward_A', ea.STATE).withUnit('kWh').withDescription('Total energy A forward'),
exposes.numeric('energy_forward_B', ea.STATE).withUnit('kWh').withDescription('Total energy B forward'),
exposes.numeric('energy_reverse_A', ea.STATE).withUnit('kWh').withDescription('Total energy A reverse'),
exposes.numeric('energy_reverse_B', ea.STATE).withUnit('kWh').withDescription('Total energy B reverse'),
exposes.numeric('update_frequency', ea.STATE).withUnit('sec').withDescription('Update frequency'),
],

meta: {
    tuyaDatapoints: [//only report data
	    [111, 'ac_frequency', tuya.valueConverter.divideBy100],
        [101, 'total_power_A', tuya.valueConverter.divideBy10],
        [105, 'total_power_B', tuya.valueConverter.divideBy10],
        [115, 'total_power_AB', tuya.valueConverter.divideBy10],
		[112, 'voltage', tuya.valueConverter.divideBy10],
		[113, 'current_A', tuya.valueConverter.divideBy100],
		[114, 'current_B', tuya.valueConverter.divideBy100],
		[110, 'power_factor_A', tuya.valueConverter.divideBy100],
		[121, 'power_factor_B', tuya.valueConverter.divideBy100],
        [102, 'power_direction A', tuya.valueConverter.raw],
        [104, 'power_direction B', tuya.valueConverter.raw],
        [106, 'energy_forward_A', tuya.valueConverter.divideBy100],
        [108, 'energy_forward_B', tuya.valueConverter.divideBy100],
        [107, 'energy_reverse_A', tuya.valueConverter.divideBy100],
        [109, 'energy_reverse_B', tuya.valueConverter.divideBy100],
		[129, 'update_frequency' , tuya.valueConverter.raw],
    ],
},
};

module.exports = definition;`

2) Add 2 lines in to configuration.yaml where is zigbe2mqtt installation directory.
```
external_converters:
  - Tuya_TZE204_81yrt3lo.js
```

Is it correct?

Dams74

Toma82, 2023-08-05:
Hi
thanks @samartandek

in my case i had to change "[113, 'current_A', tuya.valueConverter.divideBy1000]" because current_A was displayed wrong

anonymous66va, 2023-08-06:
> Hi Thanks @samartandek
> 
> In my case i had to change "[113, 'current_A', tuya.valueConverter.divideBy1000]" because current_A was displayed wrong

I also added this line but I don't know if it was necessary

[image]

samartandek, 2023-08-06:
Hi guys
Divideby(1000) is not nessesary in my opinion because divide by 1000 not
used by report data in pids table.
I have pids table from another/correct source and reports only pids use
only divider by 10 and 100.

But you can put your values as you want.

anonymous66va, 2023-08-06:
> Hi guys Divideby(1000) is not nessesary in my opinion because divide by 1000 not used by report data in pids table. I have pids table from another/correct source and reports only pids use only [...]

This was the status before divide by 1000
Let me thank you for your great job!!
[image]

Toma82, 2023-08-06:
This is what it looks like when I changed it to 1000

tuyaDatapoints: [//only report data
	    [111, 'ac_frequency', tuya.valueConverter.divideBy100],
        [101, 'total_power_A', tuya.valueConverter.divideBy10],
        [105, 'total_power_B', tuya.valueConverter.divideBy10],
        [115, 'total_power_AB', tuya.valueConverter.divideBy10],
		[112, 'voltage', tuya.valueConverter.divideBy10],
		[113, 'current_A', tuya.valueConverter.divideBy1000],
		[114, 'current_B', tuya.valueConverter.divideBy1000],
		[110, 'power_factor_A', tuya.valueConverter.divideBy100],
		[121, 'power_factor_B', tuya.valueConverter.divideBy100],
[image]

and as it is in the original, i.e. 100
As you can see the amps are distorted
[image]

samartandek, 2023-08-06:
Sorry guys, specially for anonymous66va and Toma82.
You have totally right, current A & B must by divided by 1000 because Ohm  rules.

I got table with pids from manufacturer with wrong divider for current.
I updated my post with external converter.
Thank for your alert.

[image]

anonymous66va, 2023-08-06:
> Sorry guys, specially for anonymous66va and Toma82. You have totally right, current A & B must by divided by 1000 because Ohm rules.
> 
> I got table with pids from manufacturer with wrong divider for current. I updated my post with external converter. Thank for your alert.
> 
> [image]

Ottimo lavoro !

anonymous66va, 2023-08-06:
Dear @samartandek ,
could you modify the external converter file to adjust the powers calibrations / coefficients (118,124)?
It would be great i we can adjust them form the z2m settings (specific )interface.
many thanks

thibaultgrad, 2023-08-09:
> Hi guys and Sophie Zhang :-), i have the same unit and i made external converter for it. It's first version but all data from report only DPid all covered. From report/setting pids are only Voltage calibration supported now by default in HA=>Zigbee2mqtt addon=>setting (specific).
> 
> I don't know how to config report/setting pids in external converter. I don't have tuya zigbee gateway, i switched debug on in zigbee2mqtt to test and verify is it working properly. If you know how to test it more and add support to report/setting pids give me sign.
> 
> I got only one bad info in zigbee2mqtt log: "No converter available for 'PJ-1203A' with cluster 'manuSpecificTuya' and type 'commandMcuSyncTime' and data '{"payloadSize":XXX}'" but it's known subject by koenkk.
> 
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee'); const tz = require('zigbee-herdsman-converters/converters/toZigbee'); const exposes = require('zigbee-herdsman-converters/lib/exposes'); const reporting = require('zigbee-herdsman-converters/lib/reporting'); const extend = require('zigbee-herdsman-converters/lib/extend'); const tuya = require('zigbee-herdsman-converters/lib/tuya'); const utils = require('zigbee-herdsman-converters/lib/utils'); const e = exposes.presets; const ea = exposes.access; const {Buffer} = require('buffer');
> 
> const valueConverter = { divideBy10: tuya.valueConverterBasic.divideBy(10), divideBy100: tuya.valueConverterBasic.divideBy(100), divideBy1000: tuya.valueConverterBasic.divideBy(1000), };
> 
> const definition = { fingerprint: [ { modelID: 'TS0601', manufacturerName: '_TZE204_81yrt3lo', }, ], model: 'TS0601_Bidirectional_Energy_Meter_with_80A_Current_Clamp', vendor: 'TuYa', description: 'Bidirectional Energy Meter with 80A Current Clamp', fromZigbee: [tuya.fz.datapoints], toZigbee: [tuya.tz.datapoints], onEvent: tuya.onEventSetTime, configure: tuya.configureMagicPacket, whiteLabel: [ tuya.whitelabel('Tuya', 'PJ-1203A', 'Bidirectional Energy Meter with 80A Current Clamp', ['_TZE204_81yrt3lo']), ], // whiteLabel: [{vendor: 'MatSeePlus', model: 'PJ-1203A'}], exposes: [ //only report data e.ac_frequency(), exposes.numeric('total_power_A', ea.STATE).withUnit('W').withDescription('Total power A'), exposes.numeric('total_power_B', ea.STATE).withUnit('W').withDescription('Total power B'), exposes.numeric('total_power_AB', ea.STATE).withUnit('W').withDescription('Total power A'), exposes.numeric('voltage', ea.STATE).withUnit('V').withDescription('Voltage'), exposes.numeric('current_A', ea.STATE).withUnit('A').withDescription('Current A'), exposes.numeric('current_B', ea.STATE).withUnit('A').withDescription('Current B'), exposes.numeric('power_factor_A', ea.STATE).withUnit('%').withDescription('Instantaneous measured power factor A'), exposes.numeric('power_factor_B', ea.STATE).withUnit('%').withDescription('Instantaneous measured power factor B'), exposes.numeric('power_direction A', ea.STATE).withDescription('Power direction A 0/1 for forward/reverse'), exposes.numeric('power_direction B', ea.STATE).withDescription('Power direction B 0/1 for forward/reverse'), exposes.numeric('energy_forward_A', ea.STATE).withUnit('kWh').withDescription('Total energy A forward'), exposes.numeric('energy_forward_B', ea.STATE).withUnit('kWh').withDescription('Total energy B forward'), exposes.numeric('energy_reverse_A', ea.STATE).withUnit('kWh').withDescription('Total energy A reverse'), exposes.numeric('energy_reverse_B', ea.STATE).withUnit('kWh').withDescription('Total energy B reverse'), exposes.numeric('update_frequency', ea.STATE).withUnit('sec').withDescription('Update frequency'), ],
> 
> ```
> meta: {
>     tuyaDatapoints: [//only report data
>       [111, 'ac_frequency', tuya.valueConverter.divideBy100],
>       [101, 'total_power_A', tuya.valueConverter.divideBy10],
>       [105, 'total_power_B', tuya.valueConverter.divideBy10],
>       [115, 'total_power_AB', tuya.valueConverter.divideBy10],
>       [112, 'voltage', tuya.valueConverter.divideBy10],
>       [113, 'current_A', tuya.valueConverter.divideBy1000],
>       [114, 'current_B', tuya.valueConverter.divideBy1000],
>       [110, 'power_factor_A', tuya.valueConverter.divideBy100],
>       [121, 'power_factor_B', tuya.valueConverter.divideBy100],
>       [102, 'power_direction A', tuya.valueConverter.raw],
>       [104, 'power_direction B', tuya.valueConverter.raw],
>       [106, 'energy_forward_A', tuya.valueConverter.divideBy100],
>       [108, 'energy_forward_B', tuya.valueConverter.divideBy100],
>       [107, 'energy_reverse_A', tuya.valueConverter.divideBy100],
>       [109, 'energy_reverse_B', tuya.valueConverter.divideBy100],
>       [129, 'update_frequency' , tuya.valueConverter.raw],
>     ],
> },
> ```
> 
> };
> 
> module.exports = definition;
> 
> screenshoots: [image]
> 
> [image]
> 
> [image]

Works like a charm on my device thank you !!

samartandek, 2023-08-09:
> Dear @samartandek , could you modify the external converter file to adjust the powers calibrations / coefficients (118,124)? It would be great if we can adjust them from the z2m settings (specific )interface. many thanks

Hi,
as I wrote before:
"I don't know how to config report/setting pids in external converter.
I don't have tuya zigbee gateway, i switched debug on in zigbee2mqtt to test and verify is it working properly.
If you know how to test it more and add support to report/setting pids give me sign."

jmpaiva, 2023-08-10:
Hi everyone, i have one of this units, z2m is seeing it but i can't get any reading other than the voltage. I have the feeling i'm missing something stupid but still i can't pin point it. I have both clamps on a water boiler and i can't get any reading at all. I tried reversing the wires and still no luck
 Anyone has any idea on what to try next? Do i need to send some instruction so the unit jnows it has clamps? Any help is welcome

jmpaiva, 2023-08-10:
Some messages i see in the logs

2023-08-10 08:21:45Received Zigbee message from '0xa4c138720ce406ab', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,0],"type":"Buffer"},"datatype":2,"dp":113}],"seq":64000}' from endpoint 1 with groupID 0
2023-08-10 08:21:45Received Zigbee message from '0xa4c138720ce406ab', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,0],"type":"Buffer"},"datatype":2,"dp":101}],"seq":64256}' from endpoint 1 with groupID 0
2023-08-10 08:21:45Received Zigbee message from '0xa4c138720ce406ab', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,100],"type":"Buffer"},"datatype":2,"dp":110}],"seq":64512}' from endpoint 1 with groupID 0
2023-08-10 08:21:45Received Zigbee message from '0xa4c138720ce406ab', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,19,35],"type":"Buffer"},"datatype":2,"dp":111}],"seq":64768}' from endpoint 1 with groupID 0
2023-08-10 08:21:45

samartandek, 2023-08-10:
Clamp must be connected only to fireware L without others (no N, no ground).

Check with device which take more than 5W.

jmpaiva, 2023-08-12:
Silly me, on the rush to test it, the clanp was embracing the whole cable and not just the L wire. It's working fine now. I even tested the device in comparisson to another power meter and it reads very similar values. Good little device

jwillmer, 2023-08-16:
@samartandek would you mind creating a pull request to get the device into the official version?

dams74, 2023-08-21:
@samartandek i can create en pull request if you don't can do it

samartandek, 2023-08-21:
Hi dams74,
if you know how to do it use my updated external converter from post:
https://github.com/Koenkk/zigbee2mqtt/issues/18419#issuecomment-1662389536 
and create a pull request

samartandek, 2023-08-21:
[image]
150x150 picture to add to doc

anonymous66va, 2023-08-26:
The device works perfectly fine, but upon inspecting the log, I've noticed these two highlighted messages indicated by the red arrows.
Therefore, I've decided to modify the file Tuya_TZE204_81yrt3lo.js by adding the "_" symbol to the TAGS enclosed in red.
I've also attempted to exclude and re-include the device, but the MQTT log consistently returns the same error, and the power_direction X entities are not being exposed.
Where am I going wrong? Thank you for the support.
[image]

samartandek, 2023-08-26:
Look at piece of external converter
text in withDescription('something a or b or whatever') section doesn't matter but names like 'total_power_A' from exposes and names 'total_power_A' in meta must be the same in 100% percent size matter

exposes: [ //only report data
...............................
exposes.numeric('total_power_A', ea.STATE).withUnit('W').withDescription('Total power A'),
exposes.numeric('total_power_B', ea.STATE).withUnit('W').withDescription('Total power B'),
.....................................................
],

meta: {
    tuyaDatapoints: [//only report data
...................................
      [101, 'total_power_A', tuya.valueConverter.divideBy10],
      [105, 'total_power_B', tuya.valueConverter.divideBy10],
............................................................	  
    ],
},

remove device with force always when you change something in external converter and at the end restart zigbee2mqtt and add device

qkddn49, 2023-08-29:
update_frequency Is it possible to modify it's?

qkddn49, 2023-08-29:
There are no external devices

dams74, 2023-08-29:
> Hi dams74, if you know how to do it use my updated external converter from post: [#18419 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/18419#issuecomment-1662389536) and create a pull request

Hi
Sorry i'm late
It's done

yuriyb80, 2023-08-29:
> > update_frequency Is it possible to modify it's?
> 
> I think yes by tuya app or if you expand external converter definition.

where I can find some examples of how to add change to the external converter for changing update_frequency?

samartandek, 2023-08-29:
> > Hi dams74, if you know how to do it use my updated external converter from post: [#18419 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/18419#issuecomment-1662389536) and create a pull request
> 
> Hi Sorry i'm late It's done

Good job :-)

but now pull status is "All checks have failed" because:

Check failure on line 5950 in src/devices/tuya.ts

GitHub
Actions
/ ci
',' expected.

and you don't use my update external coverter but it's not big deal you must
in line "exposes.numeric('total_power_AB', ea.STATE).withUnit('W').withDescription('Total power A')," replace "Total power A" to "Total power A+B"

Koenkk, 2023-08-30:
Device is supported now, thanks all! Assuming this can be closed

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-08-30 by Koenkk

anonymous66va, 2023-08-31:
> Hello, thanks for this job. You have to add underscore to power_direction_____ A to be able to see it in Home Assistant :
> 
> ```
>         exposes.numeric('power_factor_A', ea.STATE).withUnit('%').withDescription('Instantaneous measured power factor A'),
>         exposes.numeric('power_factor_B', ea.STATE).withUnit('%').withDescription('Instantaneous measured power factor B'),
> ```
> 
> and
> 
> ```
>             [102, 'power_direction_A', tuya.valueConverter.raw],
>             [104, 'power_direction_B', tuya.valueConverter.raw],
> ```

this is what I already wrote a few posts above

dams74, 2023-08-31:
In vendor file it's ok?
`
    {
        fingerprint: tuya.fingerprint('TS0601', ['_TZE204_81yrt3lo']),
        model: 'PJ-1203A',
        vendor: 'TuYa',
        description: 'Bidirectional energy meter with 80A current clamp',
        fromZigbee: [tuya.fz.datapoints],
        toZigbee: [tuya.tz.datapoints],
        onEvent: tuya.onEventSetTime,
        configure: tuya.configureMagicPacket,
        exposes: [
            e.ac_frequency(), e.voltage(),
            tuya.exposes.powerWithPhase('a'), tuya.exposes.powerWithPhase('b'), tuya.exposes.powerWithPhase('ab'),
            tuya.exposes.currentWithPhase('a'), tuya.exposes.currentWithPhase('b'),
            tuya.exposes.powerFactorWithPhase('a'), tuya.exposes.powerFactorWithPhase('b'),
            tuya.exposes.energyFlowWithPhase('a'), tuya.exposes.energyFlowWithPhase('b'),
            tuya.exposes.energyWithPhase('a'), tuya.exposes.energyWithPhase('b'),
            tuya.exposes.energyProducedWithPhase('a'), tuya.exposes.energyProducedWithPhase('b'),
            e.numeric('update_frequency', ea.STATE).withUnit('s').withDescription('Update frequency'),
        ],
        meta: {
            tuyaDatapoints: [
                [111, 'ac_frequency', tuya.valueConverter.divideBy100],
                [101, 'power_a', tuya.valueConverter.divideBy10],
                [105, 'power_b', tuya.valueConverter.divideBy10],
                [115, 'power_ab', tuya.valueConverter.divideBy10],
                [112, 'voltage', tuya.valueConverter.divideBy10],
                [113, 'current_a', tuya.valueConverter.divideBy1000],
                [114, 'current_b', tuya.valueConverter.divideBy1000],
                [110, 'power_factor_a', tuya.valueConverter.divideBy100],
                [121, 'power_factor_b', tuya.valueConverter.divideBy100],
                [102, 'energy_flow_a', tuya.valueConverterBasic.lookup({'consuming': 0, 'producing': 1})],
                [104, 'energy_flow_b', tuya.valueConverterBasic.lookup({'consuming': 0, 'producing': 1})],
                [106, 'energy_a', tuya.valueConverter.divideBy100],
                [108, 'energy_b', tuya.valueConverter.divideBy100],
                [107, 'energy_produced_a', tuya.valueConverter.divideBy100],
                [109, 'energy_produced_b', tuya.valueConverter.divideBy100],
                [129, 'update_frequency', tuya.valueConverter.raw],
            ],
        },
    },
];`

pyzimmer, 2023-08-31:
@dams74 perfect it's ok !
How to use the last vendor file with Zigbee2mqtt extension for Home Assistant ?

samartandek, 2023-08-31:
//        tuya.exposes.energyFlowWithPhase('b'),
//        tuya.exposes.energyFlowWithPhase('a'),
//        tuya.exposes.energyProducedWithPhase('a'),
//        tuya.exposes.energyProducedWithPhase('b'),

This 4 doesn't work yet because is currently in
https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/lib/tuya.ts

It's extend tuyaExposes for energy flow and produce with phase

samartandek, 2023-09-02:
Device is officially supported now https://www.zigbee2mqtt.io/devices/PJ-1203A.html

andregoncalvespires, 2023-09-02:
Great job.

vladii07, 2023-10-15:
I'm having some problems with the supported device. I'm running z2m on docker (1.33.1-dev commit: [31541bd]). All the exposed attributes are coming through, but it seems like they're out of order. Frequency is 2.4Hz and power direction is 60? It looks like the right info, but out of order. Is there a way I can supersede a known device with an external source, or is there a way to change attribute order? Sorry, new to z2m. The device did work and report correctly for the last 2 weeks on a Tuya gateway, so I know it's reading fine. I tried adding it a couple of times.

diegoflyer, 2023-12-11:
Dear guys! I had this  PJ-1203A new that is completely freezing my system, it is sending data to zigbee2mqtt every 10 seconds but it is generating too many message to my domoticz system that the rasp 3b+ is not able to handle:
[image]
For every received zigbee message an mqtt is generated and it is too much, do you have some advice on how to manange this issue?
Is it the device itself with problems?
Is it my rasp 3b+ with no enough power?

Last: from the instructions it looks possible to change the update frequency, is it possible to change it via mqtt in some way?

diegoflyer, 2023-12-12:
Regarding to many messages issue it is my fault, I activated the debounce option with 1 second and now I have 1 message every 10 seconds.
What about the possibility to change the update frequency, is it possible to change it via zigbee2mqtt in some way?

diegoflyer, 2023-12-13:
@smartandek  Hi!
I am using zigbee2mqtt with domoticz and this device give strange data.
We have 2 data power (a and b) and 2 energy data (a and b).
In domoticz data should become 4 device:
1. power a
2. power b
3. Power a + Energy a
4. Power b + Energy b

but it is not like that!

On the device Power b + Energy B the power data is the data equal to power a.
From my rough analysis the error is not in domoticz....

KonsGit, 2024-03-07:
This is my converter. It can configure the update rate and energy calibration. The rest can be added at will - I didn’t need it.
Maybe it will be useful to someone.

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;
const {Buffer} = require('buffer');

const valueConverter = {
divideBy10: tuya.valueConverterBasic.divideBy(10),
divideBy100: tuya.valueConverterBasic.divideBy(100),
divideBy1000: tuya.valueConverterBasic.divideBy(1000),
};

const tzDatapoints = {
    ...tuya.tz.datapoints,
    key: ['update_rate', 'calibration_forward_energy_A', 'calibration_forward_energy_B', 'calibration_ac_freq']
};

const definition = {
fingerprint: [
{
modelID: 'TS0601',
manufacturerName: '_TZE204_81yrt3lo',
},
],
    model: 'PJ-1203A',
    vendor: 'TuYa',
    description: 'Bidirectional Energy Meter with 80A Current Clamp',
    fromZigbee: [tuya.fz.datapoints, fz.ignore_tuya_set_time],  
    toZigbee: [tzDatapoints],
    onEvent: tuya.onEventSetTime,
    configure: tuya.configureMagicPacket,
//    whiteLabel: [ tuya.whitelabel('MatSeePlus', 'PJ-1203A', 'Bidirectional Energy Meter with 80A Current Clamp', ['_TZE204_81yrt3lo']), ],
    // whiteLabel: [{vendor: 'MatSeePlus', model: 'PJ-1203A'}],
    exposes: [
//        e.ac_frequency(),
        exposes.numeric('ac_freq', ea.STATE).withUnit('Hz').withDescription('AC frequency'),
        exposes.numeric('total_power_A', ea.STATE).withUnit('W').withDescription('Total power A'),
        exposes.numeric('total_power_B', ea.STATE).withUnit('W').withDescription('Total power B'),
        exposes.numeric('total_power_AB', ea.STATE).withUnit('W').withDescription('Total power A+B'),
        e.voltage(),
        exposes.numeric('current_A', ea.STATE).withUnit('A').withDescription('Current A'),
        exposes.numeric('current_B', ea.STATE).withUnit('A').withDescription('Current B'),
        exposes.numeric('power_factor_A', ea.STATE).withUnit('%').withDescription('Power factor A'),
        exposes.numeric('power_factor_B', ea.STATE).withUnit('%').withDescription('Power factor B'),
        exposes.numeric('power_direction A', ea.STATE).withDescription('Power direction A'),
        exposes.numeric('power_direction B', ea.STATE).withDescription('Power direction B'),
        exposes.numeric('energy_forward_A', ea.STATE).withUnit('kWh').withDescription('Total energy A forward'),
        exposes.numeric('energy_forward_B', ea.STATE).withUnit('kWh').withDescription('Total energy B forward'),
        exposes.numeric('energy_reverse_A', ea.STATE).withUnit('kWh').withDescription('Total energy A reverse'),
        exposes.numeric('energy_reverse_B', ea.STATE).withUnit('kWh').withDescription('Total energy B reverse'),
        e.numeric('update_rate', ea.STATE_SET).withUnit('sec').withDescription('Report the update rate').withValueMin(3).withValueMax(60),
        e.numeric('calibration_forward_energy_A', ea.STATE_SET).withDescription('Calibration forward energy A'),
        e.numeric('calibration_forward_energy_B', ea.STATE_SET).withDescription('Calibration forward energy B'),
        e.numeric('calibration_ac_freq', ea.STATE_SET).withDescription('Calibration AC freq'),
    ],
    meta: {
        tuyaDatapoints: [
            [111, 'ac_freq', tuya.valueConverter.divideBy100],
            [101, 'total_power_A', tuya.valueConverter.divideBy10],
            [105, 'total_power_B', tuya.valueConverter.divideBy10],
            [115, 'total_power_AB', tuya.valueConverter.divideBy10],
            [112, 'voltage', tuya.valueConverter.divideBy10],
            [113, 'current_A', tuya.valueConverter.divideBy1000],
            [114, 'current_B', tuya.valueConverter.divideBy1000],
            [110, 'power_factor_A', tuya.valueConverter.raw],
            [121, 'power_factor_B', tuya.valueConverter.raw],
            [102, 'power_direction A', tuya.valueConverterBasic.lookup({ 'consuming': 0, 'producing': 1 })],
            [104, 'power_direction B', tuya.valueConverterBasic.lookup({ 'consuming': 0, 'producing': 1 })],
            [106, 'energy_forward_A', tuya.valueConverter.divideBy100],
            [108, 'energy_forward_B', tuya.valueConverter.divideBy100],
            [107, 'energy_reverse_A', tuya.valueConverter.divideBy100],
            [109, 'energy_reverse_B', tuya.valueConverter.divideBy100],
            [129, 'update_rate', tuya.valueConverter.raw],
            [119, 'calibration_forward_energy_A', tuya.valueConverter.divideBy1000],
            [125, 'calibration_forward_energy_B', tuya.valueConverter.divideBy1000],
            [122, 'calibration_ac_freq', tuya.valueConverter.divideBy1000],
        ],
    },
};

module.exports = definition;

andregoncalvespires, 2024-03-08:
Would it be possible for them to make an official update? What is needed for this to happen?

schauveau, 2024-03-21:
I have this device but I noticed something odd or, should I say, annoying.  Each flow direction (i.e. dp 102 or 104) is reported before the other values for that channel but the direction is 'late' by one update.

For example, let's imagine that I am currently producing 100W on channel A:

     "energy_flow_a":"producing" , "power_a":100

and that I turn on a device that consumes 1000W. The next state will contain 

     "energy_flow_a":"producing" , "power_a":900

which is of course impossible since I only have one 400W solar panel. 

After 10s the flow is updated and the state becomes 

     "energy_flow_a":"consuming" , "power_a":900

So it seems that `power_a` represents that power at the current time while `energy_flow_a` is the flow direction at the beginning of the previous period (so 10s in the past)

This is quite annoying because that means that all automations have to memorize the state and delay the action until the next update. 

Does anyone else see that?

pyzimmer, 2024-03-21:
@schauveau Yes, same observation with my solar panels. I think there is a time lag. I use this information to route electricity and it's a problem.

schauveau, 2024-03-22:
It should be possible to fix either in z2m using an external converter. I will have a look.

An alternative could be to write a small MQTT client to read the data and re-emit it properly after the lag.

diegoflyer, 2024-03-22:
I noticed the device is sending One message for every parameter that change, every message is then transferred in a mqtt, this was locking domoticz because of too many messages and some error. I solved by adding 1 second denounce delay to collect all the message in a single mqtt message, give a try and check if improves your situation.

schauveau, 2024-03-23:
@diegoflyer I am already using `debounce 1` but the direction issue is different.  The proper direction is provided in the NEXT group of message so 10s too late (if you use the default update period). The output looks nice but when switching between `producing` and `consuming` the state contains an incorrect direction during 10s.  

This is not obvious at first but as mentioned by [pyzimmer](https://github.com/pyzimmer), this is problematic when trying to route electricity from solar panels. 

I also noticed that the direction datapoints (102 and 104) are missing when there is no current . This is annoying but fortunately it is detectable because `total_power_X` and `current_X` are both 0. 

I already have a modified version of [KonsGit](https://github.com/KonsGit) converter (see comment above) that solves these issue by delaying `total_power_X`, `current_X` and `power_factor_X` by the same delay. However, I am not fully satisfied by my current solution so I will probably redo it in a cleaner way. I also want something that is more robust when some zigbee packets are lost.

schauveau, 2024-03-26:
My  own converter is in  https://github.com/schauveau/z2m_PJ-1203A
It is finally not based on [KonsGit](https://github.com/KonsGit) because I wanted to be backward compatible with the current version.

I started a discussion in https://github.com/Koenkk/zigbee2mqtt/discussions/21956

schauveau, 2024-04-01:
@smartandek @dronisuav  You both wrote that you obtained information from the manufacturer. Do you have any link?

I cannot even figure who is the manufacturer.  Tuya?

schauveau, 2024-04-08:
If you are wondering why you cannot use the statistic graphs on `power_a`, `power_b`, `power_ab`, `current_a` and `current_b` in Home Assistant, this is because the [zigbee2mqtt integration ](https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/extension/homeassistant.ts) cannot handle non-standard names. 

That can be solved by adding a `homeassistant:` section to the device in `configuration.yaml`

```
device:
  '0x1234567812345678'
    ...
    homeassistant:
      power_a:
        device_class: power
        entity_category: diagnostic
        state_class: measurement
      power_b:
        device_class: power
        entity_category: diagnostic
        state_class: measurement
      power_ab:
        device_class: power
        entity_category: diagnostic
        state_class: measurement
      current_a:
        device_class: current
   	entity_category: diagnostic
	state_class: measurement
	enabled_by_default: false
      current_b:
        device_class: current
        entity_category: diagnostic
        state_class: measurement
        enabled_by_default: false
```

**references**:
  - https://www.home-assistant.io/integrations/mqtt/#mqtt-discovery
  - https://www.zigbee2mqtt.io/guide/usage/integrations/home_assistant.html#overriding-discovery-properties
  - https://www.home-assistant.io/docs/configuration/customizing-devices/#device-class
  -