Koenkk/zigbee2mqtt #18420: [New device support]: Tuya TS0601 TZE204_k7mfgaen

dams74, 2023-07-24:
### Link

https://fr.aliexpress.com/item/1005005681411463.html

### Database entry

{     "ieee": "a4:c1:38:98:6c:13:5f:24",     "nwk": 4602,     "status": 2,     "lqi": "255",     "rssi": "-66",     "last_seen": "1689610321.859154",     "node_descriptor": "01:40:8e:41:11:42:42:00:00:2a:42:00:00",     "endpoints": [         {             "id": 1,             "status": 1,             "device_type": 81,             "profile_id": 260,             "manufacturer": "_TZE204_k7mfgaen",             "model": "TS0601",             "output_clusters": [                 {                     "id": 25,                     "name": "Ota",                     "attributes": []                 },                 {                     "id": 10,                     "name": "Time",                     "attributes": []                 }             ],             "input_clusters": [                 {                     "id": 4,                     "name": "Groups",                     "attributes": []                 },                 {                     "id": 5,                     "name": "Scenes",                     "attributes": []                 },                 {                     "id": 61184,                     "name": "Cluster",                     "attributes": []                 },                 {                     "id": 0,                     "name": "Basic",                     "attributes": [                         {                             "id": 0,                             "name": "zcl_version",                             "value": 3                         },                         {                             "id": 1,                             "name": "app_version",                             "value": 74                         },                         {                             "id": 2,                             "name": "stack_version",                             "value": 0                         },                         {                             "id": 3,                             "name": "hw_version",                             "value": 1                         },                         {                             "id": 4,                             "name": "manufacturer",                             "value": "_TZE204_k7mfgaen"                         },                         {                             "id": 5,                             "name": "model",                             "value": "TS0601"                         },                         {                             "id": 6,                             "name": "date_code",                             "value": ""                         },                         {                             "id": 7,                             "name": "power_source",                             "value": 1                         }                     ]                 }             ]         },         {             "id": 242,             "status": 1,             "device_type": 97,             "profile_id": 41440,             "manufacturer": null,             "model": null,             "output_clusters": [                 {                     "id": 33,                     "name": "GreenPowerProxy",                     "attributes": []                 }             ],             "input_clusters": []         }     ],     "signature": {         "manufacturer": "_TZE204_k7mfgaen",         "model": "TS0601",         "node_desc": {             "logical_type": 1,             "complex_descriptor_available": 0,             "user_descriptor_available": 0,             "reserved": 0,             "aps_flags": 0,             "frequency_band": 8,             "mac_capability_flags": 142,             "manufacturer_code": 4417,             "maximum_buffer_size": 66,             "maximum_incoming_transfer_size": 66,             "server_mask": 10752,             "maximum_outgoing_transfer_size": 66,             "descriptor_capability_field": 0         },         "endpoints": {             "1": {                 "profile_id": 260,                 "device_type": 81,                 "input_clusters": [                     4,                     5,                     61184,                     0                 ],                 "output_clusters": [                     25,                     10                 ]             },             "242": {                 "profile_id": 41440,                 "device_type": 97,                 "input_clusters": [],                 "output_clusters": [                     33                 ]             }         }     },     "class": "zigpy.device" }

### Comments

Can you help to support this device
I do not have the product
It's for another user
Thanks

tam-wh, 2023-08-11:
I have the same unit but when i control the volume and set to "medium", it failed. Apparently it only accepts "middle"

Publish 'set' 'volume' to 'Siren' failed: 'Error: Expected one of: low, middle, high, mute, got: 'medium''

dams74, 2023-08-11:
Hi
This device is supported now. 
I think you need to do an issue

Dams

closed 2023-09-19 by github-actions[bot]