Koenkk/zigbee2mqtt #18429: Schneider Electric - NHPB/SWITCH/1 - Errors due to incorrect endpoint/exposed options.

slackspace-io, 2023-07-25:
### What happened?

I believe was missed during implementation of this ticket; https://github.com/Koenkk/zigbee-herdsman-converters/pull/2676

My Schnieder devices complains about unable to configure after every restart. From debugging, it seems two issues:

1.) Wrong endpoint is being used, it should be endpoint 1 -- Not 21 
2.) Power On behaviour should be disabled as it throws errors regardless of endpoint you try to read or set.

One type of error upon startup;
`Zigbee2MQTT:error 2023-07-24 16:10:10: Failed to configure 'relay_light_master_bedroom', attempt 1 (Error: ConfigureReporting 0x70ac08fffe52095b/21 genOnOff([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":3600,"reportableChange":0}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
`

Unfortunately, I cannot confirm if the original person had this device, and all functionality worked as expected. I can only share that my current device does not. 

Pairing my device I see this;
```
2023-07-25T16:16:22.648Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":8,"manufacturerCode":null,"commandIdentifier":1},"Payload":[{"attrId":5,"status":0,"dataType":66,"**attrData":"NHPB/SWITCH/1**"}],"Command":{"ID":1,"name":"readRsp","parameters":[{"name":"attrId","type":33},{"name":"status","type":32},{"name":"dataType","type":32,"conditions":[{"type":"statusEquals","value":0}]},{"name":"attrData","type":1000,"conditions":[{"type":"statusEquals","value":0}]}]}},"address":13976,"endpoint":1,"linkquality":132,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}'
Zigbee2MQTT:debug 2023-07-25 18:16:22: Received Zigbee message from '0x70ac08fffe52095b', type 'readResponse', cluster 'genBasic', data '{"**modelId":"NHPB/SWITCH/1**"}' from endpoint 1 with groupID 0
```

I believe this is a mistake in the current configuration _not_ a need for an additional filtering of configuration ? I also do not see how I could filter further to create a new configuration for this device.

Reviewing the other configurations, I believe these devices would behave identical to the below.
```
        zigbeeModel: ['PUCK/SWITCH/1'],
        model: 'CCT5011-0001/CCT5011-0002/MEG5011-0001',
```

I'll be filing a PR in a few moments to change the endpoint to '1', and amend the extend to disable power on behavior functionality.

@roumano If you can comment/confirm whether your device behaves the same that be great :)

### What did you expect to happen?

No errrors upon 'configure' button, or restarts of zigbee2Mqtt.
For any exposed features to work, not throw errors.

### How to reproduce it (minimal and precise)

Pair Schnieder Wiser On/Off Light switch (10A Relay) to zigbee2mqtt
Will see errors on configure, lack of endpoints bound/reporting.
Errors on any restarts.
Error if try to set 'Power on Behaviour'
Error to try and manually read OnOff from endpoint 21 -- Success reading from Endpoint 1

### Zigbee2MQTT version

1.32.1

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Sonoff ezsp

### Debug log

[relay-debug.log](https://github.com/Koenkk/zigbee2mqtt/files/12163972/relay-debug.log)

roumano, 2023-07-25:
@slackspace-io ,
Create long time ago (and Power on Behaviour was not really existing) and i confirm i get also some error sometimes so you issue are confirmed i think.
i will be able to test your PR if any.
Regards

Koenkk, 2023-07-26:
Merged https://github.com/Koenkk/zigbee-herdsman-converters/pull/6011 , assuming this can be closed now.

Thanks @slackspace-io

closed 2023-07-26 by Koenkk