Koenkk/zigbee2mqtt #18432: [New device support]:  _TZE204_81yrt3lo  TS0601 Tuya Smart ZigBee Energy Meter Bidirection 1, 2 Channel with Current Transformer Clamp App Monitor Power 80A 110V/240V 50/60Hz

thibaultgrad, 2023-07-25:
### Link

https://www.aliexpress.com/item/1005005782430275.html?

### Database entry

{"id":915,"type":"Router","ieeeAddr":"0xa4c13839ca58402d","nwkAddr":2770,"manufId":4417,"manufName":"_TZE204_81yrt3lo","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\\ER,i]ER,i","65506":56,"65508":0,"65534":0,"modelId":"TS0601","manufacturerName":"_TZE204_81yrt3lo","powerSource":1,"zclVersion":3,"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-1440156621},"lastSeen":1690324782974,"defaultSendRequestWhen":"immediate"}

### Comments

I tried following the add new device tutotrial but have no idea how to separate arrays into multiple numbers. Moreover i don't have the tuya gateway so difficult to get which datapoint means what

dronisuav, 2023-07-26:
the same device https://github.com/Koenkk/zigbee2mqtt/issues/18419

yuriyb80, 2023-08-09:
same issue with the same TUYA device. 

const definition = {
fingerprint: tuya.fingerprint('TS0601', ['_TZE204_81yrt3lo']),
model: 'TS0601_clamp_meter',
vendor: 'TuYa',

tried different custom converters, not any success. 

do not have Zigbee TUYA hub and not possible to check...

Z2M logs ...
`, type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,0],"type":"Buffer"},"datatype":2,"dp":115}],"seq":54016}' from endpoint 1 with groupID 0
2023-08-09 
, type 'commandMcuSyncTime', cluster 'manuSpecificTuya', data '{"payloadSize":212}' from endpoint 1 with groupID 0`

thibaultgrad, 2023-08-09:
As @dronisuav said go to #18419 .
I tried the custom converter poste in comment and works for me

closed 2023-08-09 by thibaultgrad

ansantos, 2026-09-23:
Hi, is this normal with this device? 

[image]