Koenkk/zigbee2mqtt #18434: [New device VERSION! support]: Zigbee Wifi Mmwave Menselijke Aanwezigheid Motion Sensor Met Luminantie/Afstand Detectie 5/110/220V Tuya Smart leven Domotica

Doctor-JaKe, 2023-07-26:
### Link

https://nl.aliexpress.com/item/1005004704966668.html?spm=a2g0o.order_detail.order_detail_item.4.75bb6d76CNuiqz&gatewayAdapt=glo2nld

### Database entry

where in HAO / Home Assistant 2023.7.3 / ZijBee2MQTT 1.32.1-1 / can I 

### Comments

this is the 3rd device of this kind I am trying to use in Zigbee2MQTT. The 1st 2 are working fine, even they are not detected/recognised complety right (because of their (icon) picture. 

All 3 devices can be paierd (succesfully with inteview compledet) but only the 1st 2 are working /supoorted.  

All are these "supported" Tuya presence sensor : [](https://www.zigbee2mqtt.io/devices/TS0601_smart_human_presence_sensor_1.html )

The 1st 2 devices that work are :

---------------------------------------------------------------------------------------------------------------------------

Friendly name
    Tuya ZY-M100 by TS601 Human Presense Sensor Z2M Nr 1
Description

Last seen
    N/A
Availability
    [Disabled](https://www.zigbee2mqtt.io/guide/configuration/device-availability.html#availability-advanced-configuration)
Device type
    Router
Zigbee Model
    TS0601
Zigbee Manufacturer
    _TZE204_ztc6ggyl
Description
    Ceiling human breathe sensor
Support status

    Supported
IEEE Address
    0x70ac08fffef6176b
Network address
    0x3CC5
Manufacturer
    [TuYa](https://www.zigbee2mqtt.io/supported-devices/#v=TuYa)
Model
    [ZY-M100-L](https://www.zigbee2mqtt.io/devices/ZY-M100-L.html#tuya-zy-m100-l)
Power
Interview completed
    True
------------------------------------------
Friendly name
    Tuya ZY-M100 by TS601 Human Presense Sensor Z2M Nr 1
Description

Last seen
    N/A
Availability
    [Disabled](https://www.zigbee2mqtt.io/guide/configuration/device-availability.html#availability-advanced-configuration)
Device type
    Router
Zigbee Model
    TS0601
Zigbee Manufacturer
    _TZE204_ztc6ggyl
Description
    Ceiling human breathe sensor
Support status

    Supported
IEEE Address
    0x70ac08fffef6176b
Network address
    0x3CC5
Manufacturer
    [TuYa](https://www.zigbee2mqtt.io/supported-devices/#v=TuYa)
Model
    [ZY-M100-L](https://www.zigbee2mqtt.io/devices/ZY-M100-L.html#tuya-zy-m100-l)
Power
Interview completed
    True
------------------------------------------------------------------------------------------------------------------------------------
[image]

----------------------------------------------------------------------------------------------------------------------------

The 3rd device that is not suppoerted gives me this info m:

riendly name
    0xa4c1380679a8b7a8
Description

Last seen
    N/A
Availability
    [Disabled](https://www.zigbee2mqtt.io/guide/configuration/device-availability.html#availability-advanced-configuration)
Device type
    Router
Zigbee Model
    TS0601
Zigbee Manufacturer
    _TZE204_qasjif9e
Support status

    Not supported
IEEE Address
    0xa4c1380679a8b7a8
Network address
    0x8027
Power
Interview completed
    True
---------------------------------------------------------
[image]
[image]
-----------------------------------------------

DaanVT86, 2023-07-26:
I have the same (according to unsupported) _tze204_qasjif9e device. This seems to be the newer model that has the USB-C Connection instead of the micro-usb connector.

Bought it from: https://nl.aliexpress.com/item/1005005249827156.html?spm=a2g0o.productlist.main.1.1e2469feclbbqz&algo_pvid=7e0ec87b-0b90-4a79-92b3-b3dd38e12a18&algo_exp_id=7e0ec87b-0b90-4a79-92b3-b3dd38e12a18-0&pdp_npi=3%40dis%21EUR%2141.09%2120.95%21%21%2144.30%21%21%40210218bf16903767609661337d077b%2112000034515930057%21sea%21NL%21702494327&curPageLogUid=67xx5hexcUV7

DaanVT86, 2023-07-26:
This is the database entry

{"id":6,"type":"Router","ieeeAddr":"0xa4c138049b3a1a9f","nwkAddr":57618,"manufId":4417,"manufName":"_TZE204_qasjif9e","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1690381712647,"defaultSendRequestWhen":"immediate"}

Koenkk, 2023-07-26:
Thanks, added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-07-26 by Koenkk

Doctor-JaKe, 2023-07-26:
Ok.
Thanx everybody.

But how long will it take estimated until its supported by the normal/regular branch of Zigbee2MQTT (for Home Assistant) ??

Verzonden vanuit Outlook voor iOS<https://aka.ms/o0ukef>

Closed #18434<https://github.com/Koenkk/zigbee2mqtt/issues/18434> as completed.

DaanVT86, 2023-07-26:
@Koenkk Sorry if I`m not seeing everything (not a developer so not too familiar with github), but does your change take into account that the datapoints of this device are different then the other versions?

Doctor-JaKe, 2023-08-02:
After the latest update on 1st of august it seems to be working.

[image]

Thank You very much again !!

wizardofozzie, 2023-09-13:
Mine does not detect properly. How can I fix? Was above suggestion regarding dev brnach?

Doctor-JaKe, 2023-09-13:
Restart your Zigbee hub/device and try again. Maybe even power it off for at least 15 minutes so your Zigbee network will reset/reconfigure their mesh connection and then power on your main Zigbee hub/controller (running Zigbee2MQTT).

That worked for me.

Verzonden vanuit Outlook voor iOS<https://aka.ms/o0ukef>