Koenkk/zigbee2mqtt #18436: idinio Zigbee Dimmer-Switch-ZB3.0 Unavailable and Unsupported After Latest Update

Hefje97, 2023-07-26:
### What happened?

After the latest update, my Indinio Zigbee Dimmer has become unavailable and is no longer supported. The dimmer was functioning correctly before the update, but now it does not respond to commands, and its functionality seems to be entirely disabled.

### What did you expect to happen?

Expected Behavior:
The idinio Zigbee Dimmer should function as expected, responding to commands and providing dimming control over connected lights.

### How to reproduce it (minimal and precise)

Please note that this issue started occurring after the latest update, and no changes were made to the dimmer or the Zigbee network setup before the problem appeared. As a user, I would appreciate any assistance in resolving this issue or guidance on how to proceed with using the dimmer after the update.

### Zigbee2MQTT version

1.32.1-1

### Adapter firmware version

 20190608

### Adapter

CC2531

Koenkk, 2023-07-26:
can you provide the data/database.db entry of your device?

closed 2023-09-02 by github-actions[bot]

wkneppers, 2024-02-05:
> can you provide the data/database.db entry of your device?

Hi koenkk, I'm facing the same issues. Is it usefull to send in my database.db?