Koenkk/zigbee2mqtt #18440: [New device support]: Namron 1402768 dimmer

ottarlau, 2023-07-27:
### Link

https://www.namron.com/products/namron-zigbee-led-dimmer-tw-250w/

### Database entry

{"id":12,"type":"Router","ieeeAddr":"0x84b4dbfffe089b10","nwkAddr":47849,"manufId":4098,"manufName":"Namron AS","powerSource":"Mains (single phase)","modelId":"1402768","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[0,3,4,5,6,8,768,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"1402768","manufacturerName":"Namron AS","appVersion":0,"stackVersion":0,"hwVersion":0,"swBuildId":"1.02"}},"lightingColorCtrl":{"attributes":{"colorCapabilities":0,"colorTempPhysicalMin":250,"colorTempPhysicalMax":65279}},"genLevelCtrl":{"attributes":{"currentLevel":132}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0029e84858","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b0029e84858","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":0,"hwVersion":0,"dateCode":"210525","swBuildId":"1.02","zclVersion":8,"interviewCompleted":true,"meta":{"configured":1461352984},"lastSeen":1690454753514,"defaultSendRequestWhen":"immediate"}

### Comments

I copied a very similar device (1402769) from the Namron.ts file and adapted it to the style in the examples. It "works", dimming is fine, but on/off gives an error message, but lights go on/off, but sometimes with a delay of about 12 sek. Error messages appear typically after 10 sec or more (often device turns on at the same time as error message appears).

The difference between my device 1402768 and the one I copied 1402769, is that mine also supports color temperature. I assumed it could work without addressing that issue at first.

See my attempted external converter below (I called it 1402768.js). At first, I had compile errors and Z2M wouldn't start, but after reading the log file I found and fixed errors and it started to "work" (listed as OK in devices etc). Hence, I'm confident my setup is correct. The problem is the error messages and the delays.

Please excuse my lack of knowledge, I'm a newbie at this.

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const ota = require('zigbee-herdsman-converters/lib/ota');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');

//import {Definition, Fz, Tz, KeyValue} from '../lib/types';
//import {Zcl} from 'zigbee-herdsman';
//import * as constants from '../lib/constants';

const definition = {
        zigbeeModel: ['1402768'],
        model: '1402768',
        vendor: 'Namron',
        description: 'ZigBee LED dimmer TW',
        extend: extend.light_onoff_brightness({noConfigure: true}),
        configure: async (device, coordinatorEndpoint, logger) => {
            await extend.light_onoff_brightness().configure(device, coordinatorEndpoint, logger);
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
            await reporting.onOff(endpoint);
        },
};

module.exports = definition;
```

ottarlau, 2023-07-27:
Dimmer was off, I turn it on (using Z2M Dashboard). All good. Then I try to turn off again, but button on Dashboard not move and following error messages come in (first part of log is successful turn on as described):
Info 2023-07-27 13:50:13MQTT publish: topic 'zigbee2mqtt/0x84b4dbfffe089b10', payload '{"brightness":132,"linkquality":117,"power_on_behavior":null,"state":"OFF"}'
Info 2023-07-27 13:50:13MQTT publish: topic 'zigbee2mqtt/0x84b4dbfffe089b10', payload '{"brightness":132,"linkquality":117,"power_on_behavior":null,"state":"ON"}'
Error 2023-07-27 13:50:36Publish 'set' 'state' to '0x84b4dbfffe089b10' failed: 'Error: Command 0x84b4dbfffe089b10/1 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 47849 - 1 - 213 - 6 - 11 after 10000ms)'
Info 2023-07-27 13:50:36MQTT publish: topic 'zigbee2mqtt/0x84b4dbfffe089b10', payload '{"brightness":132,"linkquality":117,"power_on_behavior":null,"state":"OFF"}'
Error 2023-07-27 13:51:00Publish 'set' 'state' to '0x84b4dbfffe089b10' failed: 'Error: Command 0x84b4dbfffe089b10/1 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 47849 - 1 - 214 - 6 - 11 after 10000ms)'

closed 2023-09-23 by github-actions[bot]

uphillbattle, 2023-10-20:
Were you able to fix this? I see them same problem today, with the config show in the original post.

uphillbattle, 2023-10-20:
With the following external converter definition, I get the colortemp options (although I don't have the Tunable White bulbs, so I can't utilise it), but the slow response and zigbee timeout remains.

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const ota = require('zigbee-herdsman-converters/lib/ota');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');

const definition = {
        zigbeeModel: ['1402768'],
        model: '1402768',
        vendor: 'Namron AS',
        description: 'Namron ZigBee LED dimmer TW 250W',
        extend: extend.light_onoff_brightness_colortemp({noConfigure: true}),
        configure: async (device, coordinatorEndpoint, logger) => {
            await extend.light_onoff_brightness_colortemp().configure(device, coordinatorEndpoint, logger);
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl', 'lightingColorCtrl']);
            await reporting.onOff(endpoint);
            device.powerSource = 'Mains (single phase)';
            device.save;
        },
};

module.exports = definition;
```

GaliBatu, 2023-10-20:
I am still stuck with the same situation. The cousin 1402769 works beautifully, but I have several of the 1402768, with the lag. Not knowledgeable enough to find out what’s wrong with the communication to this unit.

uphillbattle, 2023-10-20:
Do both dimmers (-68 and -69) have the same firmware? Mine (-68) has firmware version 1.02.

GaliBatu, 2023-10-20:
No they have different firmware. -69 has firmware 3.09 with build date 20220531 whereas -68 has the 1.02 firmware with build date 210525

uphillbattle, 2023-10-21:
I’m suspecting this is a FW issue with the -68. My (little) experience so far is that response is quick the first time you send a command, but then, if you send a command within, say, 15-20 seconds, you get the latency/delay/timeout issue. However, if it takes longer, it seems that the first command is again snappy. Is that your experience as well. If so, I suspect the FW.

I just replaced the a Namron Z-Wave dimmer. It dropped from the Z-Wave network a couple of days after being energised. Had to de-energise it (flip the circuit breaker) and re-energise it to get it back on the Z-Wave network. It was a well-known issue with those dimmers. Got tired of it and replaced it by this Zigbee dimmer, but it would seem that Namron is not particularly good at making reliable stuff… 😠

GaliBatu, 2023-10-21:
Action is as you describe for me as well, first command snappy and normal, but then the lag/delay comes on subsequent commands. 

I sent question to Namron to ask if they plan any firmware updates on this product.

I think best and quickest option would be to just change to the 1402769 type if possible. Unfortunately I have quite a few of the -68’s so it would cost me. So I’m holding out a bit more in hope to find a solution.

oyvindhauge, 2023-10-28:
I have had the same issue both with Namron 1402768 and the Namron 1402767 dimmer (https://www.namron.com/products/namron-zigbee-led-dimmer-2-pol-250w/). The Namron 1402767 has already an adapter here: https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/namron.ts.

The following adapter works for me now for both models:

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['1402767'],
    model: '1402767',
    vendor: 'Namron',
    description: 'ZigBee LED dimmer',
    extend: extend.light_onoff_brightness({noConfigure: true, disableEffect: true}),
    meta: {disableDefaultResponse: true},
    configure: async (device, coordinatorEndpoint, logger) => {
        await extend.light_onoff_brightness().configure(device, coordinatorEndpoint, logger);
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
        await reporting.onOff(endpoint);
    },
};

module.exports = definition;
```

GaliBatu, 2023-10-29:
Thanks Øyvind,
This seems to work. Suddenly the 1402768 behaves as snappy as the others  👍

Koenkk, 2023-10-29:
@oyvindhauge thanks for the fix!

- Integrated this fix for the 1402767
- Added the 1402768 with the same fix

Changes will be included in the 1 November release.

uphillbattle, 2023-10-29:
Great @oyvindhauge!

I modified your code as follows:
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['1402768'],
    model: '1402768',
    vendor: 'Namron AS',
    description: 'Namron ZigBee LED dimmer TW 250W',
    extend: extend.light_onoff_brightness_colortemp({noConfigure: true, disableEffect: true}),
    meta: {disableDefaultResponse: true},
    configure: async (device, coordinatorEndpoint, logger) => {
        await extend.light_onoff_brightness_colortemp().configure(device, coordinatorEndpoint, logger);
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
        await reporting.onOff(endpoint);
    },
};

module.exports = definition;
```

With this modification `(light_onoff_brightness_colortemp)` I get a snappy device that even has the "tunable white" options in HA. (I don't have tunable white bulbs, though, so I can't verify the effect...)

uphillbattle, 2023-10-29:
Great @Koenkk, thanks!

I tested, the addition of `_colortemp` for 1402768 also seems to work (it's a "Tunable White" dimmer), but as I said, I haven't got the tunable white bulbs, so I can't be sure of its effects.

Koenkk, 2023-10-29:
Please add the color temperature range as described on https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_3-adding-converter-s-for-your-device

uphillbattle, 2023-10-29:
Update with color temp range:

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['1402768'],
    model: '1402768',
    vendor: 'Namron AS',
    description: 'Namron ZigBee LED dimmer TW 250W',
    extend: extend.light_onoff_brightness_colortemp({noConfigure: true, disableEffect: true, colorTempRange: [250, 65279]}),
    meta: {disableDefaultResponse: true},
    configure: async (device, coordinatorEndpoint, logger) => {
        await extend.light_onoff_brightness_colortemp().configure(device, coordinatorEndpoint, logger);
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
        await reporting.onOff(endpoint);
    },
};

module.exports = definition;
```

uphillbattle, 2023-10-29:
Here's a picture of the dimmer (1402768):

[image]

Koenkk, 2023-10-30:
@uphillbattle thanks, added!

Konstabel, 2024-10-23:
Hi, I have similar problem with a Namron 4512751 dimmer. Is there a similar fix for this?