Koenkk/zigbee2mqtt #18442: Devices unavailable after poweroutage

Drealine, 2023-07-27:
### What happened?

Hi,
I've got a small poweroutage today and all devices are unavalaible until restart Z2M.

### What did you expect to happen?

Devices restore state after poweroutage.

### How to reproduce it (minimal and precise)

My server is connected with an UPS.
The small poweroutage keep Z2M up with this test. So it's not a corrupted database or etc..

### Zigbee2MQTT version

 1.32.1

### Adapter firmware version

20221226

### Adapter

Sonoff ZB Plus

### Debug log

Nothing happen. Just some timeout regarding availibilty of devices.

Xeevis, 2023-08-03:
Hi, what MQTT broker are you using?

Drealine, 2023-08-04:
MQTT Addon of HA @Xeevis

Drealine, 2023-09-04:
Nope. Not stale

closed 2023-10-12 by github-actions[bot]