Koenkk/zigbee2mqtt #18445: Firmware update entity warning logs in Home Assistant 2023.8

RobertD502, 2023-07-27:
### What happened?

A breaking change was [introduced in Home Assistant 2023.8](https://github.com/home-assistant/core/pull/95159) involving naming of MQTT entities which affects update entities exposed by Zigbee2MQTT.

The following warning is logged by Home Assistant:
```
Logger: homeassistant.components.mqtt.mixins
Source: components/mqtt/mixins.py:294
Integration: MQTT (documentation, issues)
First occurred: 7:19:13 PM (10 occurrences)
Last logged: 7:19:14 PM

MQTT device name is equal to entity name in your config {'availability': [{'topic': 'zigbee/bridge/state', 'payload_not_available': 'offline', 'payload_available': 'online'}, {'topic': 'zigbee/Eagle Outlet/availability', 'payload_not_available': 'offline', 'payload_available': 'online'}], 'availability_mode': 'all', 'command_topic': 'zigbee/bridge/request/device/ota_update/update', 'device': {'identifiers': ['zigbee2mqtt_0x00158d00028a402f'], 'manufacturer': 'Xiaomi', 'model': 'Mi power plug ZigBee US (ZNCZ12LM)', 'name': 'Eagle Outlet', 'connections': []}, 'device_class': <UpdateDeviceClass.FIRMWARE: 'firmware'>, 'entity_picture': 'https://github.com/Koenkk/zigbee2mqtt/raw/master/images/logo.png', 'json_attributes_topic': 'zigbee/Eagle Outlet', 'latest_version_template': Template<template=({{ value_json['update']['latest_version'] }}) renders=0>, 'latest_version_topic': 'zigbee/Eagle Outlet', 'name': 'Eagle Outlet', 'payload_install': '{"id": "0x00158d00028a402f"}', 'state_topic': 'zigbee/Eagle Outlet', 'unique_id': '0x00158d00028a402f_update_zigbee', 'value_template': Template<template=({{ value_json['update']['installed_version'] }}) renders=0>, 'payload_available': 'online', 'enabled_by_default': True, 'payload_not_available': 'offline', 'qos': 0, 'encoding': 'utf-8', 'retain': False}, this is not expected. Please correct your configuration. The entity name will be set to `null`
MQTT device name is equal to entity name in your config {'availability': [{'topic': 'zigbee/bridge/state', 'payload_not_available': 'offline', 'payload_available': 'online'}, {'topic': 'zigbee/Kitchen Smart Plug/availability', 'payload_not_available': 'offline', 'payload_available': 'online'}], 'availability_mode': 'all', 'command_topic': 'zigbee/bridge/request/device/ota_update/update', 'device': {'identifiers': ['zigbee2mqtt_0x90fd9ffffe667d69'], 'manufacturer': 'Innr', 'model': 'Smart plug (SP 224)', 'name': 'Kitchen Smart Plug', 'sw_version': '2.7', 'connections': []}, 'device_class': <UpdateDeviceClass.FIRMWARE: 'firmware'>, 'entity_picture': 'https://github.com/Koenkk/zigbee2mqtt/raw/master/images/logo.png', 'json_attributes_topic': 'zigbee/Kitchen Smart Plug', 'latest_version_template': Template<template=({{ value_json['update']['latest_version'] }}) renders=0>, 'latest_version_topic': 'zigbee/Kitchen Smart Plug', 'name': 'Kitchen Smart Plug', 'payload_install': '{"id": "0x90fd9ffffe667d69"}', 'state_topic': 'zigbee/Kitchen Smart Plug', 'unique_id': '0x90fd9ffffe667d69_update_zigbee', 'value_template': Template<template=({{ value_json['update']['installed_version'] }}) renders=0>, 'payload_available': 'online', 'enabled_by_default': True, 'payload_not_available': 'offline', 'qos': 0, 'encoding': 'utf-8', 'retain': False}, this is not expected. Please correct your configuration. The entity name will be set to `null`
MQTT device name is equal to entity name in your config {'availability': [{'topic': 'zigbee/bridge/state', 'payload_not_available': 'offline', 'payload_available': 'online'}, {'topic': 'zigbee/Cellar Heater - Kitchen/availability', 'payload_not_available': 'offline', 'payload_available': 'online'}], 'availability_mode': 'all', 'command_topic': 'zigbee/bridge/request/device/ota_update/update', 'device': {'identifiers': ['zigbee2mqtt_0xb0ce18140003ef99'], 'manufacturer': 'Sengled', 'model': 'Smart plug (E1C-NB6)', 'name': 'Cellar Heater - Kitchen', 'connections': []}, 'device_class': <UpdateDeviceClass.FIRMWARE: 'firmware'>, 'entity_picture': 'https://github.com/Koenkk/zigbee2mqtt/raw/master/images/logo.png', 'json_attributes_topic': 'zigbee/Cellar Heater - Kitchen', 'latest_version_template': Template<template=({{ value_json['update']['latest_version'] }}) renders=0>, 'latest_version_topic': 'zigbee/Cellar Heater - Kitchen', 'name': 'Cellar Heater - Kitchen', 'payload_install': '{"id": "0xb0ce18140003ef99"}', 'state_topic': 'zigbee/Cellar Heater - Kitchen', 'unique_id': '0xb0ce18140003ef99_update_zigbee', 'value_template': Template<template=({{ value_json['update']['installed_version'] }}) renders=0>, 'payload_available': 'online', 'enabled_by_default': True, 'payload_not_available': 'offline', 'qos': 0, 'encoding': 'utf-8', 'retain': False}, this is not expected. Please correct your configuration. The entity name will be set to `null`
MQTT device name is equal to entity name in your config {'availability': [{'topic': 'zigbee/bridge/state', 'payload_not_available': 'offline', 'payload_available': 'online'}, {'topic': 'zigbee/Cellar Heater - 1st/availability', 'payload_not_available': 'offline', 'payload_available': 'online'}], 'availability_mode': 'all', 'command_topic': 'zigbee/bridge/request/device/ota_update/update', 'device': {'identifiers': ['zigbee2mqtt_0xb0ce18140003f855'], 'manufacturer': 'Sengled', 'model': 'Smart plug (E1C-NB6)', 'name': 'Cellar Heater - 1st', 'connections': []}, 'device_class': <UpdateDeviceClass.FIRMWARE: 'firmware'>, 'entity_picture': 'https://github.com/Koenkk/zigbee2mqtt/raw/master/images/logo.png', 'json_attributes_topic': 'zigbee/Cellar Heater - 1st', 'latest_version_template': Template<template=({{ value_json['update']['latest_version'] }}) renders=0>, 'latest_version_topic': 'zigbee/Cellar Heater - 1st', 'name': 'Cellar Heater - 1st', 'payload_install': '{"id": "0xb0ce18140003f855"}', 'state_topic': 'zigbee/Cellar Heater - 1st', 'unique_id': '0xb0ce18140003f855_update_zigbee', 'value_template': Template<template=({{ value_json['update']['installed_version'] }}) renders=0>, 'payload_available': 'online', 'enabled_by_default': True, 'payload_not_available': 'offline', 'qos': 0, 'encoding': 'utf-8', 'retain': False}, this is not expected. Please correct your configuration. The entity name will be set to `null`
MQTT device name is equal to entity name in your config {'availability': [{'topic': 'zigbee/bridge/state', 'payload_not_available': 'offline', 'payload_available': 'online'}, {'topic': 'zigbee/Cellar Heater - 2nd/availability', 'payload_not_available': 'offline', 'payload_available': 'online'}], 'availability_mode': 'all', 'command_topic': 'zigbee/bridge/request/device/ota_update/update', 'device': {'identifiers': ['zigbee2mqtt_0xb0ce18140003f6ee'], 'manufacturer': 'Sengled', 'model': 'Smart plug (E1C-NB6)', 'name': 'Cellar Heater - 2nd', 'connections': []}, 'device_class': <UpdateDeviceClass.FIRMWARE: 'firmware'>, 'entity_picture': 'https://github.com/Koenkk/zigbee2mqtt/raw/master/images/logo.png', 'json_attributes_topic': 'zigbee/Cellar Heater - 2nd', 'latest_version_template': Template<template=({{ value_json['update']['latest_version'] }}) renders=0>, 'latest_version_topic': 'zigbee/Cellar Heater - 2nd', 'name': 'Cellar Heater - 2nd', 'payload_install': '{"id": "0xb0ce18140003f6ee"}', 'state_topic': 'zigbee/Cellar Heater - 2nd', 'unique_id': '0xb0ce18140003f6ee_update_zigbee', 'value_template': Template<template=({{ value_json['update']['installed_version'] }}) renders=0>, 'payload_available': 'online', 'enabled_by_default': True, 'payload_not_available': 'offline', 'qos': 0, 'encoding': 'utf-8', 'retain': False}, this is not expected. Please correct your configuration. The entity name will be set to `null`
```

As you can see, the update entity name is the exact same as the name provided for the device name (for the device registry). As a result, this warning is being logged for all update entities.

By the looks of it, the update entity is the only one that currently isn't setting the name of the entity in the form of  {Device Name} {entity descriptor}.

For example, the energy sensor entity for my outlet has an entity name of `Eagle Outlet energy` with a device registry name of `Eagle Outlet`:

```
availability:
  - topic: zigbee/bridge/state
  - topic: zigbee/Eagle Outlet/availability
availability_mode: all
device:
  identifiers:
    - zigbee2mqtt_0x00158d00028a402f
  manufacturer: Xiaomi
  model: Mi power plug ZigBee US (ZNCZ12LM)
  name: Eagle Outlet
device_class: energy
enabled_by_default: true
[... cut ...]
entity_picture: https://github.com/Koenkk/zigbee2mqtt/raw/master/images/logo.png
json_attributes_topic: zigbee/Eagle Outlet
latest_version_template: '{{ value_json[''update''][''latest_version''] }}'
latest_version_topic: zigbee/Eagle Outlet
name: Eagle Outlet
payload_install: '{"id": "0x00158d00028a402f"}'
state_topic: zigbee/Eagle Outlet
unique_id: 0x00158d00028a402f_update_zigbee
value_template: '{{ value_json[''update''][''installed_version''] }}'
platform: mqtt
```

### Zigbee2MQTT version

1.32.1

### Adapter firmware version

N/A

### Adapter

N/A

hmmbob, 2023-07-28:
It's not just the `update` entities though....

```
2023-07-28 10:59:10.353 WARNING (MainThread) [homeassistant.components.mqtt.mixins] MQTT device name is equal to entity name in your config {'availability': [{'topic': 'zigbee2mqtt/bridge/state', 'payload_not_available': 'offline', 'payload_available': 'online'}], 'command_topic': 'zigbee2mqtt/Leeslampje zolder/set', 'device': {'identifiers': ['zigbee2mqtt_0x00158d000389270a'], 'manufacturer': 'Innr', 'model': 'Smart plug (SP 120)', 'name': 'Leeslampje zolder', 'sw_version': '2.0', 'connections': []}, 'name': 'Leeslampje zolder', 'payload_off': 'OFF', 'payload_on': 'ON', 'state_topic': 'zigbee2mqtt/Leeslampje zolder', 'unique_id': '0x00158d000389270a_switch_zigbee2mqtt', 'value_template': Template<template=({{ value_json.state }}) renders=0>, 'encoding': 'utf-8', 'payload_not_available': 'offline', 'enabled_by_default': True, 'payload_available': 'online', 'qos': 0, 'availability_mode': 'latest', 'retain': False, 'optimistic': False}, this is not expected. Please correct your configuration. The entity name will be set to `null`
2023-07-28 10:59:10.361 WARNING (MainThread) [homeassistant.components.mqtt.mixins] MQTT device name is equal to entity name in your config {'availability': [{'topic': 'zigbee2mqtt/bridge/state', 'payload_not_available': 'offline', 'payload_available': 'online'}], 'command_topic': 'zigbee2mqtt/Grijze lamp/set', 'device': {'identifiers': ['zigbee2mqtt_0x588e81fffedb88cc'], 'manufacturer': 'TuYa', 'model': '10A UK or 16A EU smart plug (TS0121_plug)', 'name': 'Grijze lamp', 'connections': []}, 'name': 'Grijze lamp', 'payload_off': 'OFF', 'payload_on': 'ON', 'state_topic': 'zigbee2mqtt/Grijze lamp', 'unique_id': '0x588e81fffedb88cc_switch_zigbee2mqtt', 'value_template': Template<template=({{ value_json.state }}) renders=0>, 'encoding': 'utf-8', 'payload_not_available': 'offline', 'enabled_by_default': True, 'payload_available': 'online', 'qos': 0, 'availability_mode': 'latest', 'retain': False, 'optimistic': False}, this is not expected. Please correct your configuration. The entity name will be set to `null`
2023-07-28 10:59:10.367 WARNING (MainThread) [homeassistant.components.mqtt.mixins] MQTT device name is equal to entity name in your config {'availability': [{'topic': 'zigbee2mqtt/bridge/state', 'payload_not_available': 'offline', 'payload_available': 'online'}], 'command_topic': 'zigbee2mqtt/Pubercave/set', 'device': {'identifiers': ['zigbee2mqtt_0x842e14fffe359c8b'], 'manufacturer': 'TuYa', 'model': '10A UK or 16A EU smart plug (TS0121_plug)', 'name': 'Pubercave', 'connections': []}, 'name': 'Pubercave', 'payload_off': 'OFF', 'payload_on': 'ON', 'state_topic': 'zigbee2mqtt/Pubercave', 'unique_id': '0x842e14fffe359c8b_switch_zigbee2mqtt', 'value_template': Template<template=({{ value_json.state }}) renders=0>, 'encoding': 'utf-8', 'payload_not_available': 'offline', 'enabled_by_default': True, 'payload_available': 'online', 'qos': 0, 'availability_mode': 'latest', 'retain': False, 'optimistic': False}, this is not expected. Please correct your configuration. The entity name will be set to `null`
2023-07-28 10:59:10.378 WARNING (MainThread) [homeassistant.components.mqtt.mixins] MQTT device name is equal to entity name in your config {'availability': [{'topic': 'zigbee2mqtt/bridge/state', 'payload_not_available': 'offline', 'payload_available': 'online'}], 'command_topic': 'zigbee2mqtt/TV Kastje/set', 'device': {'identifiers': ['zigbee2mqtt_0x847127fffe7e037c'], 'manufacturer': 'Innr', 'model': 'Smart plug (SP 220)', 'name': 'TV Kastje', 'sw_version': '2.3', 'connections': []}, 'name': 'TV Kastje', 'payload_off': 'OFF', 'payload_on': 'ON', 'state_topic': 'zigbee2mqtt/TV Kastje', 'unique_id': '0x847127fffe7e037c_switch_zigbee2mqtt', 'value_template': Template<template=({{ value_json.state }}) renders=0>, 'encoding': 'utf-8', 'payload_not_available': 'offline', 'enabled_by_default': True, 'payload_available': 'online', 'qos': 0, 'availability_mode': 'latest', 'retain': False, 'optimistic': False}, this is not expected. Please correct your configuration. The entity name will be set to `null`
2023-07-28 10:59:10.381 WARNING (MainThread) [homeassistant.components.mqtt.mixins] MQTT device name is equal to entity name in your config {'availability': [{'topic': 'zigbee2mqtt/bridge/state', 'payload_not_available': 'offline', 'payload_available': 'online'}], 'command_topic': 'zigbee2mqtt/aquarium/set', 'device': {'identifiers': ['zigbee2mqtt_0x847127fffe7f0994'], 'manufacturer': 'Innr', 'model': 'Smart plug (SP 220)', 'name': 'aquarium', 'sw_version': '2.3', 'connections': []}, 'name': 'aquarium', 'payload_off': 'OFF', 'payload_on': 'ON', 'state_topic': 'zigbee2mqtt/aquarium', 'unique_id': '0x847127fffe7f0994_switch_zigbee2mqtt', 'value_template': Template<template=({{ value_json.state }}) renders=0>, 'encoding': 'utf-8', 'payload_not_available': 'offline', 'enabled_by_default': True, 'payload_available': 'online', 'qos': 0, 'availability_mode': 'latest', 'retain': False, 'optimistic': False}, this is not expected. Please correct your configuration. The entity name will be set to `null`
2023-07-28 10:59:10.382 WARNING (MainThread) [homeassistant.components.mqtt.mixins] MQTT device name is equal to entity name in your config {'availability': [{'topic': 'zigbee2mqtt/bridge/state', 'payload_not_available': 'offline', 'payload_available': 'online'}], 'command_topic': 'zigbee2mqtt/mini_cave/set', 'device': {'identifiers': ['zigbee2mqtt [... cut ...]  '0x8cf681fffe9d549f_light_zigbee2mqtt', 'color_temp': False, 'xy': False, 'encoding': 'utf-8', 'flash_time_long': 10, 'white_scale': 255, 'payload_not_available': 'offline', 'color_mode': False, 'hs': False, 'effect': False, 'enabled_by_default': True, 'payload_available': 'online', 'qos': 0, 'flash_time_short': 2, 'availability_mode': 'latest', 'retain': False, 'rgb': False, 'optimistic': False}, this is not expected. Please correct your configuration. The entity name will be set to `null`
```

RobertD502, 2023-07-28:
> It's not just the `update` entities though....
>
>
>
> ```
>
> 2023-07-28 10:59:10.353 WARNING (MainThread) [homeassistant.components.mqtt.mixins] MQTT device name is equal to entity name in your config [...]

Yup, you're right! Looked deeper into it and I see that the name is also repeated for switch entities without appending the descriptor.

So, based on my setup I can confirm that update and switch entities are affected. From your log it appears that light entities are also affected.

francisp2, 2023-07-28:
I had not seen this, but my HA log does not shows warnings, only errors.

tjorim, 2023-07-29:
Not only update entities indeed, all names seem duplicate now:
[image]
Related change in HA core: https://github.com/home-assistant/core/pull/95159

RobertD502, 2023-07-30:
> Not only update entities indeed, all names seem duplicate now: [image] Related change in HA core: [home-assistant/core#95159](https://github.com/home-assistant/core/pull/95159)

@tjorim 
This is due to [how Home Assistant handles naming entities](https://developers.home-assistant.io/blog/2022/07/10/entity_naming/) with the `has_entity_name` property set to true. In Zigbee2MQTT's case, for the device that you showed in your example, the device name is set to "Ice cube machine" and the entity name is set to "Ice cube machine energy". What home assistant does is combines the device name with the entity name to produce the "Friendly name" that you are seeing. In order for this to not happen, Zigbee2MQTT should be setting the entity name to just "Energy" which would result in a friendly name of "Ice cube machine Energy" (name seen when the entity is clicked on in the UI), but the entity name within the device will appear as "Energy".

For example, for my Lavviebot litter box integration, the device name is Litterbox and the Humidity sensor entity name is "Humidity". On the device page, the entity is listed as "Humidity":

[image]

However, when I click on the entity, the friendly name is shown as "Litterbox Humidity" due to how naming is handled in Home Assistant:

[image]

If you want to customize the entity name from the Zigbee2MQTT side of things, although it requires some manual work, you can do so by checking out the documentation [here](https://www.zigbee2mqtt.io/guide/usage/integrations/home_assistant.html#using-a-custom-name-for-the-device-and-entities).

Hope this clears up some things.

Rockman18, 2023-08-01:
[image]

[image]

I updated my home assistant to 2023.8.0b3 this afternoon and those 2 warning messages appeared.

Koenkk, 2023-08-01:
Could you check if this is fixed by https://github.com/Koenkk/zigbee2mqtt/pull/18208 ? This is available in the dev branch now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

RobertD502, 2023-08-01:
@Koenkk Nope, still logging the warning. Example:

```
2023-08-01 13:59:44.714 WARNING (MainThread) [homeassistant.components.mqtt.mixins] MQTT device name is equal to entity name in your config {'availability': [{'topic': 'zigbee/bridge/state', 'payload_available': 'online', 'payload_not_available': 'offline'}, {'topic': 'zigbee/Cellar Heater - Kitchen/availability', 'payload_available': 'online', 'payload_not_available': 'offline'}], 'availability_mode': 'all', 'command_topic': 'zigbee/Cellar Heater - Kitchen/set', 'device': {'identifiers': ['zigbee2mqtt_0xb0ce18140003ef99'], 'manufacturer': 'Sengled', 'model': 'Smart plug (E1C-NB6)', 'name': 'Cellar Heater - Kitchen', 'connections': []}, 'json_attributes_topic': 'zigbee/Cellar Heater - Kitchen', 'name': 'Cellar Heater - Kitchen', 'payload_off': 'OFF', 'payload_on': 'ON', 'state_topic': 'zigbee/Cellar Heater - Kitchen', 'unique_id': '0xb0ce18140003ef99_switch_zigbee', 'value_template': Template<template=({{ value_json.state }}) renders=0>, 'retain': False, 'qos': 0, 'payload_available': 'online', 'optimistic': False, 'encoding': 'utf-8', 'enabled_by_default': True, 'payload_not_available': 'offline'}, this is not expected. Please correct your configuration. The entity name will be set to `null`
```

With their temporary fix of stripping the device name from the entities that had a name of `{device name} {descriptor}` a new warning is also now logged about that. For example, for a device named "Server Rack" with a battery sensor entity, they want the entity name to be "Battery" and not "Server Rack battery" as it currently is.:

```
2023-08-01 13:59:44.861 WARNING (MainThread) [homeassistant.components.mqtt.mixins] MQTT entity name starts with the device name in your config {'availability': [{'topic': 'zigbee/bridge/state', 'payload_available': 'online', 'payload_not_available': 'offline'}, {'topic': 'zigbee/Server Rack/availability', 'payload_available': 'online', 'payload_not_available': 'offline'}], 'availability_mode': 'all', 'device': {'identifiers': ['zigbee2mqtt_0x00158d000582ac5d'], 'manufacturer': 'Xiaomi', 'model': 'Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)', 'name': 'Server Rack', 'sw_version': '3000-0001', 'connections': []}, 'device_class': <SensorDeviceClass.BATTERY: 'battery'>, 'enabled_by_default': True, 'entity_category': <EntityCategory.DIAGNOSTIC: 'diagnostic'>, 'json_attributes_topic': 'zigbee/Server Rack', 'name': 'Server Rack battery', 'state_class': <SensorStateClass.MEASUREMENT: 'measurement'>, 'state_topic': 'zigbee/Server Rack', 'unique_id': '0x00158d000582ac5d_battery_zigbee', 'unit_of_measurement': '%', 'value_template': Template<template=({{ value_json.battery }}) renders=0>, 'qos': 0, 'payload_available': 'online', 'force_update': False, 'encoding': 'utf-8', 'payload_not_available': 'offline'}, this is not expected. Please correct your configuration. The device name prefix will be stripped off the entity name and becomes 'Battery'
```

Koenkk, 2023-08-01:
Merged another fix, could you check if it works? (Wait 1 hour before updating to make changes are build and published)

RobertD502, 2023-08-01:
Out of curiosity, after taking a quick look at the changes, any reason for choosing not to capitalize the entity name? For example, using Battery instead of battery. For entities with multiple words, only the first is capitalized (e.g., `power on behavior` would be `Power on behavior`) - This is based on Home Assistant naming conventions. I don't believe using lower-case entity names will be a problem though.

hmmbob, 2023-08-01:
Yep, fixed on 1.32.2-dev commit: [f619610](https://github.com/Koenkk/zigbee2mqtt/commit/f619610)

RobertD502, 2023-08-01:
@Koenkk Running `1.32.2-dev commit: [f619610](https://github.com/Koenkk/zigbee2mqtt/commit/f619610)`, but the issue still persists with both entity name equaling device name and entity name starts with device name:

```
2023-08-01 14:57:37.749 WARNING (MainThread) [homeassistant.components.mqtt.mixins] MQTT device name is equal to entity name in your config {'availability': [{'topic': 'zigbee/bridge/state', 'payload_not_available': 'offline', 'payload_available': 'online'}, {'topic': 'zigbee/Kitchen Smart Plug/availability', 'payload_not_available': 'offline', 'payload_available': 'online'}], 'availability_mode': 'all', 'command_topic': 'zigbee/Kitchen Smart Plug/set', 'device': {'identifiers': ['zigbee2mqtt_0x90fd9ffffe667d69'], 'manufacturer': 'Innr', 'model': 'Smart plug (SP 224)', 'name': 'Kitchen Smart Plug', 'sw_version': '2.7', 'connections': []}, 'json_attributes_topic': 'zigbee/Kitchen Smart Plug', 'name': 'Kitchen Smart Plug', 'payload_off': 'OFF', 'payload_on': 'ON', 'state_topic': 'zigbee/Kitchen Smart Plug', 'unique_id': '0x90fd9ffffe667d69_switch_zigbee', 'value_template': Template<template=({{ value_json.state }}) renders=0>, 'retain': False, 'encoding': 'utf-8', 'qos': 0, 'optimistic': False, 'payload_available': 'online', 'payload_not_available': 'offline', 'enabled_by_default': True}, this is not expected. Please correct your configuration. The entity name will be set to `null`
```

```
2023-08-01 14:57:37.894 WARNING (MainThread) [homeassistant.components.mqtt.mixins] MQTT entity name starts with the device name in your config {'availability': [{'topic': 'zigbee/bridge/state', 'payload_not_available': 'offline', 'payload_available': 'online'}, {'topic': 'zigbee/Kitchen Motion Sensor/availability', 'payload_not_available': 'offline', 'payload_available': 'online'}], 'availability_mode': 'all', 'device': {'identifiers': ['zigbee2mqtt_0x286d97000104be4d'], 'manufacturer': 'SmartThings', 'model': 'Motion sensor (2018 model) (IM6001-MTP01)', 'name': 'Kitchen Motion Sensor', 'sw_version': '', 'connections': []}, 'device_class': <SensorDeviceClass.TIMESTAMP: 'timestamp'>, 'enabled_by_default': False, 'entity_category': <EntityCategory.DIAGNOSTIC: 'diagnostic'>, 'icon': 'mdi:clock', 'json_attributes_topic': 'zigbee/Kitchen Motion Sensor', 'name': 'Kitchen Motion Sensor last seen', 'state_topic': 'zigbee/Kitchen Motion Sensor', 'unique_id': '0x286d97000104be4d_last_seen_zigbee', 'value_template': Template<template=({{ value_json.last_seen }}) renders=0>, 'encoding': 'utf-8', 'qos': 0, 'force_update': False, 'payload_available': 'online', 'payload_not_available': 'offline'}, this is not expected. Please correct your configuration. The device name prefix will be stripped off the entity name and becomes 'Last seen'
```

hmmbob, 2023-08-01:
Can you confirm those timestamps are close to your actual time?

Asking because they're hours of my timezone, minutes from now (if you disregard timezones) and I actually was looking at old logs just a few minutes ago as well. It appears that the system log page isn't fully cleared during reboot with the HA window open.

RobertD502, 2023-08-01:
> Can you confirm those timestamps are close to your actual time?
>
> Asking because they're hours of my timezone, minutes from now (if you disregard timezones) and I actually was looking at old logs [...]

Yes these are up to date logs. Fresh log from the time I restarted Home Assistant (after zigbe2mqtt was pulled and up and running).

RobertD502, 2023-08-01:
@Koenkk Indeed fixed! 

@hmmbob's discussion regarding Home Assistant cache made me remember there are caching settings within Zigbee2MQTT. By disabling `Persist cache state` and `Send cached state on startup`, restarting Zigbee2MQTT, and then restarting Home Assistant, I can confirm that the warning logs are gone! So, it was a Zigbee2MQTT caching setting that was preventing the use of the new naming scheme.

Drafteed, 2023-08-02:
> Out of curiosity, after taking a quick look at the changes, any reason for choosing not to capitalize the entity name? For example, using Battery instead of battery. For entities with multiple [...]

We tried to do this, but then reverted the change. https://github.com/Koenkk/zigbee2mqtt/pull/16702

@Koenkk Is this a breaking change for HA users lower 2022.08.0? https://github.com/Koenkk/zigbee2mqtt/commit/f6196107a75f9257366505a3ad5ef7b29ff5bc1c

RobertD502, 2023-08-02:
> We tried to do this, but then reverted the change. https://github.com/Koenkk/zigbee2mqtt/pull/16702
>
>
>
> @Koenkk Is this a breaking change for HA users lower 2022.08.0? [...]

It looks like it was shot down because the way of handling the naming in that PR was too uniform in the sense that all names will only have the first letter capitalized which isn't always the case (as seen in the VOC example which would incorrectly be turned into Voc).

The capitalization or not of the first word isn't a breaking change / not what this issue is about - I was just raising the capitalization discussion as a secondary question since the naming is being worked on at the moment. You could have an entity named `BaTtErY` and it won't raise an issue - it just isn't following the recommended naming convention. The breaking change came from the fact that entity names were either exactly the same as the device name, or were prefixed by the device name (Device name: Aqara, Entity name: Aqara humidity). However, @Koenkk has already fixed this with a PR introduced earlier today and it only applies starting with 2023.8.

Koenkk, 2023-08-02:
@Drafteed this is indeed somewhat breaking but there is no way we can prevent it (we don't have control over what HA users are running). I will put it as a note in the next release notes.

Assuming this can be closed now.

closed 2023-08-02 by Koenkk

Drafteed, 2023-08-02:
@Koenkk what do think about my suggestion in https://github.com/Koenkk/zigbee2mqtt/pull/16702?

I propose to introduce a new property `title` in [base Z2M expose class](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/lib/exposes.ts), containing the name of the exposed entity in the correct case (e.g. `Device Temperature`, `VOC`, `Power Outage Count`) and without using the underscore separator. I think the next release would be great for this change along with https://github.com/Koenkk/zigbee2mqtt/commit/f6196107a75f9257366505a3ad5ef7b29ff5bc1c

nohn, 2023-08-03:
It seems like the change breaks a lot of sensors. I have many https://www.zigbee2mqtt.io/devices/MCCGQ11LM.html#xiaomi-mccgq11lm installed. Those provided a balbalba_contact entity that now seems renamed to blabla_door:

[image]

[image]

[image]

nohn, 2023-08-03:
A downgrade to latest does not fix this. Entity stays named _door.

Koenkk, 2023-08-03:
@Drafteed sounds good, I propose to name it `label` and by default infer it from `name` (replace `_` with space and capitalise first letter) 

@nohn did you also update HA to 2023.8? For me it works fine:

[image]

nohn, 2023-08-03:
@Koenkk sure. I only switched to the dev build because I upgraded to Home Assistant 2022.8. There were a few more entities that have been renamed:

* _contact to _door
* _water_leak to _moisture
* _occupancy to _motion
* _particle_pm2.5 to _pm2.5

At least for my installation it seems that was all. Fortunately sed on Home Assistant's config/*.yaml and config/.storage/ was sufficient to fix this.

Drafteed, 2023-08-04:
> @Koenkk sure. I only switched to the dev build because I upgraded to Home Assistant 2022.8 before and saw this was already fixed in dev branch. There were a few more entities that have been renamed:
> 
> * _contact to _door
> * _water_leak to _moisture
> * _occupancy to _motion
> * _particle_pm2.5 to _pm2.5

This caused by #18208. Downgrade z2m to latest (not dev) release, delete all > `*_door`, `*_moisture`, `*_motion`,  `*_pm2.5` entites (**not devices**), reboot HA.

Koenkk, 2023-08-04:
@Drafteed did you experience the same issue? I'm confused why it did not happen in my setup.

Jhutjens92, 2023-08-06:
> @Koenkk Indeed fixed!
>
> @hmmbob's discussion regarding Home Assistant cache made me remember there are caching settings within Zigbee2MQTT. By disabling `Persist cache state` and `Send cached state [...]

Not sure what these two settings do (following your steps fixed the warning message for me  aswell) however did you enable those two settings again after the reboot?

RobertD502, 2023-08-06:
I did.

marvinweber, 2023-08-06:
> @Koenkk Indeed fixed!
>
> @hmmbob's discussion regarding Home Assistant cache made me remember there are caching settings within Zigbee2MQTT. By disabling `Persist cache state` and `Send cached state [...]

For me this (as well as the upgrade to version 1.32.2 / 1ec1e57) did not solve the issue with those warnings. I still get the "MQTT device name is equal to entity name in your config" for a lot of entities and the one aka "MQTT entity name starts with device name" for even more entities.

Don't know if this could be related or the cause: for many devices I set a custom name in zigbee2mqtt (frontend) with the toggle "Update Home Assistant entity ID" enabled.  
I also tested what happens when i actively rename a device in zigbee2mqtt to check if then the issue is resolved: it's not.

tjorim, 2023-08-06:
> For me this (as well as the upgrade to version 1.32.2 / 1ec1e57) did not solve the issue with those warnings. I still get the "MQTT device name is equal to entity name in your config" for a lot of entities and the one aka "MQTT entity name starts with device name" for even more entities.

Fix will only be included in the release of next month.

RobertD502, 2023-08-06:
> For me this (as well as the upgrade to version 1.32.2 / 1ec1e57) did not solve the issue with those warnings. I still get the "MQTT device name is equal to entity name in your config" for a lot of [...]

It is fixed in the image tagged "latest-dev". Changing the name in Home Assistant doesn't fix the issue because the naming is caught upstream - although you change the name in Home Assistant, this doesn't change the entity name that is sent in the MQTT message, which is what triggers the exception in Home Assistant.

marvinweber, 2023-08-06:
> Changing the name in Home Assistant doesn't fix the issue because the naming is caught upstream

I didn't change the name in Home Assistant, but I. Zigbee2Mqtt directly. And because this actually reduced the amount of warnings in Home Assistant I was confused and thought that 1.32.2 should already contain the fix.

Nevertheless, I'll try it with the dev version or with the next release. Thx!

RobertD502, 2023-08-06:
> I didn't change the name in Home Assistant, but in Zigbee2Mqtt directly.

Ah. If you changed the name via the UI, then you either changed the friendly name or device name. With 32.2, the MQTT messages would still prefix the device name to the entity name. Only way, that I know of, to manually circumvent the errors in 32.2 is to manually name each and every **entity** for a given device within the configuration file.

Drafteed, 2023-08-11:
> @Drafteed did you experience the same issue? I'm confused why it did not happen in my setup.

No, but if you remove a device from HA and add it again, it will be added with a new HA object_id, since it is generated based on the name and is not fixed in the discovery payload.

Koenkk, 2023-08-12:
> but if you remove a device or entity from HA and add it again

Z2M sets the `object_id` via the topic:
- HA docs: https://www.home-assistant.io/integrations/mqtt/#discovery-topic
- Z2M test: https://github.com/Koenkk/zigbee2mqtt/blob/1ec1e57244834cc6b9dea6dbd3612b01cd21fa65/test/homeassistant.test.js#L445 (`object_id` = `humidity` here)

However if the name just changes when you remove the device from HA an add it again, I think we should not consider this as a blocking issue.

bplatypus, 2023-08-13:
Hi, just switched to z2m (was using deconz before), should I use dev branch to configure my lights or can I just use the release?
Not really sure what impact this will have... I would like to avoid having to rename everything in my scenes, automations & node-red :-D.

I named all my lights in z2m with following pattern:
'room.deviceType.number'

ie: kitchen.light.1 or kitchen.switch.1

outlying, 2023-08-15:
It is still a bit confusing if it's OK to wait for the fix or take action now and what to do exactly, I have 320 entities reported with warnings across 30+ devices. 

Since all is working now and 2024.02 is a few months away should we assume it's likely to be fixed by the end of the year?

netweaver1970, 2023-08-15:
as stated numerous times before, it's already fixed in the next release: 1.33.x. You can use it already as part of the Edge version of the z2m add-on if you want but there's absolutely no real NEED for that, apart from cleaner logs (I temporarily switched over just for this), as the issue will be automatically solved at the next z2m go-live, end of this month. 

There was no need for me to change any scripts/automations etc, as HA is gentle enough to make the current issue invisible, up to release 2028.2. 

And after moving to the Edge version, all kept on working as before as well. I did have to do  the z2m cache disable/enable actions as described above but that was a once-off. Maybe it would be good to have this actions build it into the next update, to prevent all 1000s of installations to have to do it manually. If not, it's not a problem either, it just would be nicer/cleaner.

robert2s, 2023-08-27:
For Dutch users who search for clues to this problem: Home Assistant reports this problem as "_Ontdekte MQTT entiteiten met een naam gelijk aan de apparaatnaam_".

ASNNetworks, 2023-09-11:
So after updating everything and restarting etc etc (no error logs) I now end up with Zigbee devices that use the sensor type in the friendly name, like: `Kantoor Beweging`, became `Kantoor Beweging Beweging`. The only way to fix this is by manually changing the Home Assistant name in Zigbee2Mqtt for every sensor I have, and will connect in the future? Changing it to 'Kantoor' in Home Assistant name field, fixes the friendly name and allows me to still use `Kantoor Beweging` as Z2M name.

Is this really the solution, or am I missing something very obvious? 

[image]

bikeymouse, 2023-11-08:
While this is solved now largely with the updates from Home Assistant for all Zigbee devices, I still get the warning below, which is about not adhering to the naming conventions for the Z2MQT bridge states, version and health.

As I can not change the name, how can I get rid of these warnings?
`
MQTT entity name starts with the device name in your config {'name': 'Zigbee2MQTT Bridge state', 'unique_id': 'zigbee2mqtt_bridge_state', 'state_topic': 'zigbee2slzb/bridge/state', 'value_template': Template<template=({{ value_json.state }}) renders=0>, 'icon': 'mdi:router-wireless', 'entity_category': <EntityCategory.DIAGNOSTIC: 'diagnostic'>, 'device': {'identifiers': ['zigbee2mqtt'], 'name': 'Zigbee2MQTT', 'model': 'Zigbee2MQTT', 'manufacturer': 'Zigbee2MQTT', 'connections': []}, 'force_update': False, 'payload_not_available': 'offline', 'qos': 0, 'enabled_by_default': True, 'payload_available': 'online', 'availability_mode': 'latest', 'encoding': 'utf-8'}, this is not expected. Please correct your configuration. The device name prefix will be stripped off the entity name and becomes 'Bridge state'

MQTT entity name starts with the device name in your config {'name': 'Zigbee2MQTT Version', 'unique_id': 'zigbee2mqtt_version', 'state_topic': 'zigbee2slzb/bridge/info', 'value_template': Template<template=({{ value_json.version }}) renders=0>, 'icon': 'mdi:zigbee', 'entity_category': <EntityCategory.DIAGNOSTIC: 'diagnostic'>, 'device': {'identifiers': ['zigbee2mqtt'], 'name': 'Zigbee2MQTT', 'model': 'Zigbee2MQTT', 'manufacturer': 'Zigbee2MQTT', 'connections': []}, 'force_update': False, 'payload_not_available': 'offline', 'qos': 0, 'enabled_by_default': True, 'payload_available': 'online', 'availability_mode': 'latest', 'encoding': 'utf-8'}, this is not expected. Please correct your configuration. The device name prefix will be stripped off the entity name and becomes 'Version'

MQTT entity name starts with the device name in your config {'name': 'Zigbee2MQTT health', 'unique_id': 'zigbee2mqtt_healty', 'state_topic': 'zigbee2slzb/bridge/response/health_check', 'value_template': Template<template=({{ value_json['data']['healthy'] }}) renders=0>, 'icon': 'mdi:zigbee', 'entity_category': <EntityCategory.DIAGNOSTIC: 'diagnostic'>, 'device': {'identifiers': ['zigbee2mqtt'], 'name': 'Zigbee2MQTT', 'model': 'Zigbee2MQTT', 'manufacturer': 'Zigbee2MQTT', 'connections': []}, 'force_update': False, 'payload_not_available': 'offline', 'qos': 0, 'enabled_by_default': True, 'payload_available': 'online', 'availability_mode': 'latest', 'encoding': 'utf-8'}, this is not expected. Please correct your configuration. The device name prefix will be stripped off the entity name and becomes 'Health'
`

quadcom, 2023-12-08:
Running 1.34.0-1 with HA 2023-12 and still seeing these messages, hundreds of them. I found them after troubleshooting a hue bulb going unvail in HA after 2 minutes, but controllable in Z2M. As well as an Aqara TVOC sensor not updating values in HA.  I figured this error message could be attrib to the problems I was having.

Interestingly, all of those reports only mentioned devices that worked reliably in HA and Z2M.

I re-paired the TVOC to Z2M a couple of times and it seemed to clear up both that devices update lag and the hue bulb going unavail. At least for the last few hours now.

RobertD502, 2023-12-08:
> Running 1.34.0-1 with HA 2023-12 and still seeing these messages, hundreds of them. I found them after troubleshooting a hue bulb going unvail in HA after 2 minutes, but controllable in Z2M. As well [...]

If you have `Persist cache state` and/or `Send cached state on startup` enabled in the Z2M settings, disable them then restart the container. When the container is up you can re-enable those settings.

quadcom, 2023-12-09:
Well, I'm back to the disappearing Hue bulb. It was working fine for a while, but then I went and did the 2023.12.1 update, and both the Hue and the Aqara TVCO are awol in HA. I can see them and control them in Z2M but HA shows them unavailable.

I've deleted them in HA, shut down Z2M and restarted Z2M. They pop back up again but immediately go unavail.

I've made sure that all the cache in Z2M is disabled. I checked each device and made sure the cache was also unchecked. 

I considered my wifi could be causing a problem and changed AP's channels and power output. 

I've deleted the bulb again, renamed it in Z2M and then cold restarted Z2M.

Still, it's the same thing every time. HA finds it and adds it and immediately it's offline.

I have no idea what I need to check or do next to troubleshoot this!!!

Any ideas?

_PS, All my other devices are working perfectly while this is all going on. I have door sensors, motion sensors and AC plugs. All working without an issue. It's just these two items that are not playing nicely._

RobertD502, 2023-12-09:
I recommend opening a new issue or seeking guidance on the Home Assistant community forums since the issue you are experiencing is not related to the one this issue thread is about.

1mfaasj, 2023-12-15:
What steps do I need to do to fix the issue, its still present here.
I'm already using the latest versions of HA and Zigbee2mqtt.

Also disabled and restarted: "Persist cache state and/or Send cached state on startup" already

Do I need to change the mqtt device names or zigbee home automation device names?
Or the names / home assistant names in the zigbee2mqtt GUI?

Thx

RobertD502, 2023-12-16:
> What steps do I need to do to fix the issue, its still present here.
> 
> I'm using the latest versions of HA and Zigbee2mqtt.
> 
> 
> 
> Also disabled and restarted: "Persist cache state and/or Send cached state on startup" already but that didnt solve the issue.
> 
> 
> 
> Do I need to change the mqtt device names or zigbee home automation device names?
> 
> Or the names / home assistant names in the zigbee2mqtt GUI? Or something else?
> 
> 
> 
> Thx

Use a program such as MQTT explorer and delete all topics created by Z2M. There could be some things retained there that are causing your issue.

* Disable the cache options and stop Z2M before doing this since you don't want Z2M recreating the topics while you're trying to clear them out.

Once all topics are cleared out, restart Z2M and re-enable the options if you want to.

PR #18678 (2023-08-22): feat: Use labels in Home Assistant entity naming
Based on: https://github.com/Koenkk/zigbee-herdsman-converters/pull/6066, https://github.com/Koenkk/zigbee2mqtt/issues/18445#issuecomment-1662899667, and https://github.com/Koenkk/zigbee2mqtt/pull/16702#issuecomment-1438906709.

Finally fixes: #18445

This PR introduces using of a new `label` property from `exposes` in Home Assistant entity naming and brings entity naming to the requirements from https://github.com/home-assistant/core/pull/95159 and https://developers.home-assistant.io/docs/core/entity/#entity-naming.

This change also affects # [... cut]