Koenkk/zigbee2mqtt #18450: [New device support]: TuYa TS0601_smart_human_presence_sensor_1 from another manufacturer

87insane, 2023-07-28:
### Link

https://www.zigbee2mqtt.io/devices/TS0601_smart_human_presence_sensor_1.html

### Database entry

{"id":11,"type":"Router","ieeeAddr":"0xa4c138169ab47edf","nwkAddr":48004,"manufId":4417,"manufName":"_TZE204_qasjif9e","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"��V,i��V,i��V,i��V,i��V,i��V,i��V,i��V,i","65506":56,"65508":0,"stackVersion":0,"dateCode":"","appVersion":74}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1690547323147,"defaultSendRequestWhen":"immediate"}

### Comments

z2m already has the TuYa TS0601_smart_human_presence_sensor_1. But I got one now from a different manufacturer. This is not supported. Please add.

[image]

vincentBesseau, 2023-07-28:
I receive the same device since 1 hour.
I keep an eye on subject :-) !

cooldome, 2023-07-28:
+1. I have reported the same issue minutes ago

cooldome, 2023-07-28:
Closed my issue as duplicate of this one, moving my comments here:

I have bought another tuya ceiling presence sensor from this shop https://www.aliexpress.com/item/1005005709438307.html?pdp_npi=3%40dis%21%21US%20%2433.21%21US%20%2416.27%21%21%21%21%21%402101eb4f16905861679471675d8cca%2112000034097049715%21sh%21UK%211823063661&spm=a2g0o.store_pc_aeChoiceSimplified.allitems_choice_2005835956649.4

The device came in different packaging and  the plastic was slightly better than original, but otherwise the identical device. I have 14 tuya ceiling presence sensors I can tell the difference. This device wasn't recognised by zigbee2mqtt to my surprise. For example this one is recognized: https://www.aliexpress.com/item/1005004724689720.html?spm=a2g0o.order_list.order_list_main.44.16991802iulwjh

I believe it wasn't recognized because manufacturer name is different, for this new device it is:

model: 'TS0601'
manufacturer name: '_TZE204_qasjif9e'

87insane, 2023-07-29:
In my z2m this device works now. In the herdsmann converters the device is included.... I have Copy / Paste the Part for this in my z2m on the Raspberry. I dont know why this device is not included in the official z2m Release. But as temp workaround its okay for me.

cooldome, 2023-07-29:
It was added on 26th of July. Not released yet.

Koenkk, 2023-07-31:
`_TZE204_qasjif9e` will be supported in tomorrows release.

closed 2023-07-31 by Koenkk

Sch8er, 2023-10-13:
@Koenkk 

seemed the `TZE204_qasjif9e`  worked at my side but the related picture was the ceiling mount - in front of me - was a wall mount 

[image]

Koenkk, 2023-10-13:
@Sch8er please provide a good picture (150x150, jpg with a white background)

mrnoisytiger, 2024-12-29:
I purchased one recently as well with TuYa manufacturer `_TZE204_laokfqwu`. Would be great to add that in as well.

Koenkk, 2024-12-30:
`_TZE204_laokfqwu` will be supported in the 3 January z2m release (2.0.0)