Koenkk/zigbee2mqtt #18452: All Devices offline

eatoff8, 2023-07-29:
### What happened?

All devices appear offline all at once.

### What did you expect to happen?

Should continue working

### How to reproduce it (minimal and precise)

Happens irregularly. happened this morning, happened 4 days ago. Previously it happened after a week and a half.

### Zigbee2MQTT version

1.32.1-1

### Adapter firmware version

20230507

### Adapter

Sonoff ZB Dongle E

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/12205979/log.txt)

eatoff8, 2023-07-29:
So, some more info:

This appears to happen randomly. It can be stable for 2 weeks, or 2 days. the add-on is still running, its like the USB stick stops responding. A restart of the add-on brings everything back up.
[add on log.txt](https://github.com/Koenkk/zigbee2mqtt/files/12206001/add.on.log.txt)

I have also attached a copy of the home assistant add-on logs here

[add on log.txt](https://github.com/Koenkk/zigbee2mqtt/files/12206001/add.on.log.txt)

avvalex, 2023-07-30:
It's happening to me too. Suddenly the coordinator (Gio-Dot version, so same chip, and same adapter firmware version) stops working and the only thing that resuscitates it is a plug out and plug in. Restarting Z2M doesn't work, rebooting system doesn't work (maybe a shutdown does) and checking USB port shows that the port remains the same (only stops responding).
Someone has any idea on how to investigate the reason why the coordinator dies?? Maybe some system log....

DubhAd, 2023-08-01:
The Sonoff P is known to sometimes crash, and also there's some models with NVRAM issues, if that's the case a restart of Z2M will usually clear both of those.

eatoff8, 2023-08-03:
> The Sonoff P is known to sometimes crash, and also there's some models with NVRAM issues, if that's the case a restart of Z2M will usually clear both of those.

I thought the Sonoff P was the recommended one, hence my purchase. I moved house so was also migrating ZB from an ethernet based TubesZB to the USB based Sonoff P.

I am waiting for it to happen again so I can catch the logs a bit better before restarting the Z2M add-on.

eatoff8, 2023-08-10:
[log (2).txt](https://github.com/Koenkk/zigbee2mqtt/files/12308540/log.2.txt)

So, happenned again the other night, logs attached. The device we were trying to swtich off when we noticed it was the light - sarah bedside

francisp2, 2023-08-10:
>  I moved house so was also migrating ZB from an ethernet based TubesZB to the USB based Sonoff P.
You don't have the TubesZB anymore ?

eatoff8, 2023-08-10:
> You don't have the TubesZB anymore ? if you still have it,  see if it happens with that one too.

No, I sold it just about straight away after migrating to the Sonoff. Seems like a pretty stupid idea now...

cloudharps, 2023-08-15:
Same Problem. I am using Sonoff dongle p. After intermittent stops, no signal is received or sent. Only a z2m restart solves the problem. It happens once or twice a day.

eatoff8, 2023-08-16:
> Same Problem. I am using Sonoff dongle p.

Have you tried re-flashing your stick? That's what I'm going to try next.

I removed a plug with power monitoring from my network to see if that is what was causing it, will see how that has helped soon

eatoff8, 2023-08-28:
@Koenkk I'm having issues with this still, how can I provide more meaningful logs? Keen to get a solution, having some regrets moving to the Sonoff

Koenkk, 2023-08-29:
Can you try with the 20221226 firmware? https://github.com/Koenkk/Z-Stack-firmware/tree/Z-Stack_3.x.0_coordinator_20221226/coordinator/Z-Stack_3.x.0/bin

eatoff8, 2023-08-29:
> Can you try with the 20221226 firmware? https://github.com/Koenkk/Z-Stack-firmware/tree/Z-Stack_3.x.0_coordinator_20221226/coordinator/Z-Stack_3.x.0/bin

I just re-flashed the latest firmware this morning, so see if wiping the NVRAM will improve things. If it goes down again, i'll try that firmware and them monitor it again.

eatoff8, 2023-09-01:
My stick crashed again this morning with the 2023 firmware, so just loaded the 20221226 firmware Koenkk linked above. Will update on stability of the stick soon

eatoff8, 2023-10-02:
So I just got notification from the stalebot

@Koenkk since downgrading the firmware on the 1st September I have not had a single issue. Thanks for the advice

closed 2023-10-02 by eatoff8