Koenkk/zigbee2mqtt #18455: [New device support]: Philips 9290024406A / LWU002

RPiNut, 2023-07-29:
### Link

https://www.philips-hue.com/nl-nl/p/hue-white-kogellamp---e14-slimme-lamp----2-pack-/8719514356771#specifications

### Database entry

{"id":12,"type":"Router","ieeeAddr":"0x001788010d851c5a","nwkAddr":40091,"manufId":4107,"manufName":"Signify Netherlands B.V.","powerSource":"Mains (single phase)","modelId":"LWU002","epList":[11,242],"endpoints":{"11":{"profId":260,"epId":11,"devId":257,"inClusterList":[0,3,4,5,6,8,4096,64515,64514],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"LWU002","manufacturerName":"Signify Netherlands B.V.","powerSource":1,"zclVersion":8,"appVersion":2,"stackVersion":1,"hwVersion":1,"dateCode":"20230116","swBuildId":"1.101.10"}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":1,"hwVersion":1,"dateCode":"20230116","swBuildId":"1.101.10","zclVersion":8,"interviewCompleted":true,"meta":{},"lastSeen":1690616379907,"defaultSendRequestWhen":"immediate"}

### Comments

This is my first comment on this GitHub page. I tried to use the Zigbee2MQTT dev version for docker to make this bulbs working. But unfortunately it didn't work. It looks like it's the same bulb on the [Zigbee2MQTT-website.](https://www.zigbee2mqtt.io/devices/9290024406.html) I do have this bulb also, and the only diffrence on the product it self, is instead of no. 9290024406 there is an A after it.

Some research take me to the website of Philips it self, and there where 3 different model numbers on the website, all starting with 9290024406 and anding with 02, 03 and 04. The bulb that I haven't working at the moment is the 9290024406**04** and the one that is working with Zigbee2MQTT is the 9290024406**02**.

### Supported color modes

None. Just dimming only

popadiys, 2023-07-29:
I want to add the same bulb, but in color variant. It seems that the new manufacturer code is Signify:

{"id":5,"type":"Router","ieeeAddr":"0x001788010d84a9b1","nwkAddr":23204,"manufId":4107,"manufName":"Signify Netherlands B.V.","powerSource":"Mains (single phase)","modelId":"LCU001","epList":[11,242],"endpoints":{"11":{"profId":260,"epId":11,"devId":269,"inClusterList":[0,3,4,5,6,8,4096,64515,768,64513,64516],"outClusterList":[25],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":1,"hwVersion":1,"dateCode":"20230403","swBuildId":"1.104.2","zclVersion":8,"interviewCompleted":true,"meta":{},"lastSeen":1690641575733,"defaultSendRequestWhen":"immediate"}

RPiNut, 2023-07-29:
Is there a simple sollution for this issue? I've tried to follow the guide to create a new sensor, but for the couple of minutes I didn't get it to work. If I add the JS file to the configuration Zigbee2MQTT failed to start.

popadiys, 2023-07-29:
proobably you have some syntax error. I have added signify.js with the following content:

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters//lib/ota');
const philips = require('zigbee-herdsman-converters//lib/philips');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
        zigbeeModel: ['LCU001'],
        model: 'LCU001',
        vendor: 'Signify Netherlands B.V.',
        description: 'Hue White and Color Ambiance E14',
        extend: philips.extend.light_onoff_brightness_colortemp_color({colorTempRange: [153, 500]}),
};

module.exports = definition;

and configuration.yaml:

advanced:
  log_level: debug
external_converters:
  - signify.js

RPiNut, 2023-07-30:
Thanks for sharing your configuration. I made two tiny mistakes. First adding advanced twice in the configuration.yaml. Second, I didn't add the row `extend: extend.light_onoff_brightness()` to the js-file, cause I thought it is only a simple lamp.

The only thing I didn't have working at the moment, is the picture. That is missing right now, but I can fully control it at the moment.

This is my working js-file now for this bulb;

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters//lib/ota');
const philips = require('zigbee-herdsman-converters//lib/philips');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
zigbeeModel: ['LWU002'],
model: '9290024406A',
vendor: 'Signify Netherlands B.V.',
description: 'Hue P45 light bulb',
extend: extend.light_onoff_brightness(),
};

module.exports = definition;

Koenkk, 2023-07-31:
Added, thanks!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-07-31 by Koenkk